evcc-io/evcc #21566: Zaptec control is not working any more: unexpected status: 500 (Internal Server Error)

TimoD87, 2025-06-01:
### Describe the bug

The Zaptec integration isn’t working anymore for now I would say 2 to 3 weeks.

EVCC isn’t able to control the charge itself and also not the power.

 I tried the nightly built, older the versions, but it’s not working anymore. I think the problem comes from the back end of Zaptec. I also tried to increase the interval to 60 which didn’t help or at the charger ID.

My evcc.yaml:

If there is a way, I can help you to fix this problem. Please tell me what I can do. I’m looking forward that the problem hopefully is getting solved in the near future.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
chargers:
- type: template
template: zaptec
id: 42144850a-9ab5-4601-b560-654646516198
user: user@user.de  
password: xxxxxx  
name: wallbox4
```

### Log details

```text
`
[lp-1 ] DEBUG 2025/06/01 11:39:36 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/06/01 11:39:36 vehicle range: 302km
[lp-1 ] WARN 2025/06/01 11:39:36 charger logic error: disabled but charging
[lp-1 ] DEBUG 2025/06/01 11:39:36 charger enable
[lp-1 ] ERROR 2025/06/01 11:39:36 charger enable: unexpected status: 500 (Internal Server Error)
[lp-1 ] DEBUG 2025/06/01 11:40:17 set smart cost limit: 0.2
[lp-1 ] DEBUG 2025/06/01 11:40:19 set smart cost limit: 0.2
[site ] DEBUG 2025/06/01 11:40:36 ----
[lp-1 ] DEBUG 2025/06/01 11:40:36 charge power: 3592W
[lp-1 ] DEBUG 2025/06/01 11:40:36 charge currents: [5.81 5.59 5.84]A
[site ] DEBUG 2025/06/01 11:40:36 grid power: -103W
[site ] DEBUG 2025/06/01 11:40:36 pv 1 power: 4131W
[site ] DEBUG 2025/06/01 11:40:36 battery 1 power: 0W
[site ] DEBUG 2025/06/01 11:40:36 battery 1 soc: 81%
[site ] DEBUG 2025/06/01 11:40:36 site power: -103W
[lp-1 ] DEBUG 2025/06/01 11:40:37 vehicle odometer: 13850km
[lp-1 ] DEBUG 2025/06/01 11:40:37 detected active phases: 3p
[lp-1 ] DEBUG 2025/06/01 11:40:37 charger status: C
[lp-1 ] DEBUG 2025/06/01 11:40:37 vehicle soc: 58%
[lp-1 ] DEBUG 2025/06/01 11:40:37 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/06/01 11:40:37 vehicle range: 302km
[lp-1 ] WARN 2025/06/01 11:40:37 charger logic error: disabled but charging
[lp-1 ] ERROR 2025/06/01 11:40:37 charger enable: unexpected status: 500 (Internal Server Error)
 `
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.204.1

Flipper81, 2025-06-01:
The same problem for me since today.

TimoD87, 2025-06-01:
I tested some more things and in my opinion the problem is on Zaptec side because it is also not possible if I try to set power in the Zaptec App.

iamHow4rd, 2025-06-02:
I also have this problem now since a few days.

tbaehler, 2025-06-02:
I think this should solve the issue

https://github.com/evcc-io/evcc/pull/21594

@TimoD87 
Can you please give this evcc version a try ? 
https://www.dropbox.com/scl/fi/cu46szf1sqms388ltw7fv/evcc_linux_armrlkey=5c3si7x6egwuwtccd37sbcjya&st=0876ztj6&dl=0

Its a arm build and the newest evcc version with the fix in the PR

andig, 2025-06-02:
> I think the problem is, that the Web Api is generating a 500 error when we call enable and the wallbox is already enabled.

We could just ignore the error result in that case.

@TimoD87 could you provide a trace log of the error and of last working version? Must be a difference somewhere.

TimoD87, 2025-06-02:
Sure, will test it in about 2 hours and give feedback

TimoD87, 2025-06-02:
> @TimoD87 could you provide a trace log of the error and of last working version? Must be a difference somewhere.

Ich bin mir da nicht sicher, b das Problem nicht vom Zaptec Backend kommt. Ich habe auch Probleme die Leistung über die Zaptec App einzustellen, wenn ich evcc deaktiviert habe.

TimoD87, 2025-06-02:
> > I think the problem is, that the Web Api is generating a 500 error when we call enable and the wallbox is already enabled.
> 
> We could just ignore the error result in that case.
> 
> @TimoD87 could you provide a trace log of the error and of last working version? Must be a difference somewhere.

Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Jun 2 19:00:48 2025 from [REDACTED_IP_ADDRESS]
pi@raspberrypi:~ $ sudo journalctl -fau evcc
Jun 02 19:15:30 raspberrypi evcc[2195092]: [lp-1  ] WARN 2025/06/02 19:15:30 charger logic error: disabled but charging
Jun 02 19:15:30 raspberrypi evcc[2195092]: [zaptec] TRACE 2025/06/02 19:15:30 POST https://api.zaptec.com/api/chargers/[REDACTED_CHARGER_ID]/sendCommand/507
Jun 02 19:15:30 raspberrypi evcc[2195092]: [zaptec] TRACE 2025/06/02 19:15:30 {"Code":528,"Details":"Charging is not Paused nor Scheduled; Resume command cannot be sent","StackTrace":null}
Jun 02 19:15:30 raspberrypi evcc[2195092]: [lp-1  ] ERROR 2025/06/02 19:15:30 charger enable: unexpected status: 500 (Internal Server Error)
Jun 02 19:15:30 raspberrypi evcc[2195092]: [site  ] TRACE 2025/06/02 19:15:30 telemetry: charge: Δ0/[REDACTED_NUMERIC_VALUE]Wh @ [REDACTED_NUMERIC_VALUE]W
Jun 02 19:15:30 raspberrypi evcc[2195092]: [site  ] TRACE 2025/06/02 19:15:30 POST https://api.evcc.io/v1/charge
Jun 02 19:15:30 raspberrypi evcc[2195092]: [site  ] TRACE 2025/06/02 19:15:30 {"instanceId":"[REDACTED_INSTANCE_ID]","chargePower":[REDACTED_NUMERIC_VALUE],"greenPower":[REDACTED_NUMERIC_VALUE],"chargeEnergy":[REDACTED_NUMERIC_VALUE],"greenEnergy":[REDACTED_NUMERIC_VALUE]}
Jun 02 19:15:30 raspberrypi evcc[2195092]: --
Jun 02 19:15:30 raspberrypi evcc[2195092]: {"status":"ok"}
Jun 02 19:16:23 raspberrypi evcc[2195092]: [db    ] TRACE 2025/06/02 19:16:23 INSERT INTO `settings` (`key`,`value`) VALUES ("telemetry","true"),("lp1.title","[REDACTED_GARAGE_NAME]"),("solarAccYield","{""pv2"":{""accumulated"":[REDACTED_NUMERIC_VALUE]}}"),("lp1.mode","off"),("adminPassword","[REDACTED_PASSWORD_HASH]"),("jwtSecretKey","[REDACTED_JWT_SECRET]"),("lp1.smartCostLimit","[REDACTED_NUMERIC_VALUE]"),("lp1.limitSoc","0") ON CONFLICT (`key`) DO UPDATE SET `value`=`excluded`.`value` 8 <nil>
Jun 02 19:16:30 raspberrypi evcc[2195092]: [site  ] DEBUG 2025/06/02 19:16:30 ----
Jun 02 19:16:30 raspberrypi evcc[2195092]: [zaptec] TRACE 2025/06/02 19:16:30 GET https://api.zaptec.com/api/chargers/[REDACTED_CHARGER_ID]/state
Jun 02 19:16:30 raspberrypi evcc[2195092]: [zaptec] TRACE 2025/06/02 19:16:30 [{"ChargerId":"[REDACTED_CHARGER_ID]","StateId":-2,"Timestamp":"2025-05-13T08:16:15.453","ValueAsString":"1"},{"ChargerId":"[REDACTED_CHARGER_ID]","StateId":-1,"Timestamp":"2025-06-02T17:15:37.957"},{"ChargerId":"[REDACTED_CHARGER_ID]","StateId":1,"Timestamp":"2023-05-13T17:49:51.78","ValueAsString":"0"},{"ChargerId":"[REDACTED_CHARGER_ID]","StateId":100,"Timestamp":"2025-05-25T15:57:36.89","ValueAsString":"{\"CommunicationModes\":[\"LTE\",\"WiFi\"],\"DeviceType\":\"Go\",\"MeterCalibrated\":false,\"OcppVersions\":[\"V1_6\"],\"SchemaVersion\":\"[REDACTED_VERSION]\"}"},{"ChargerId":"[REDACTED_CHARGER_ID]","StateId":120,"Timestamp":"2023-05-13T10:59:11.65","ValueAsString":"0"},{"ChargerId":"[REDACTED_CHARGER_ID]","StateId":145,"Timestamp":"2023-05-13T10:59:11.607","ValueAsString":"[REDACTED_NUMERIC_VALUE]"},{"ChargerId":"[REDACTED_CHARGER_ID]","StateId":147,"Timestamp":"2025-06-02T17:15:37.253","ValueAsString":"[REDACTED_NUMERIC_VALUE]"},{"ChargerId":"[REDACTED_CHARGER_ID]","StateId":150,"Timestamp":"2025-06-01T10:53:45.507","ValueAsString":"LTE"},{"ChargerId":"[REDACTED_CHARGER_ID]","StateId":151,"Timestamp":"2025-06-01T10:54:58.79","ValueAsString":"0"},{"ChargerId":"[REDACTED_CHARGER_ID]","StateId":153,"Timestamp":"2025-06-01T09:58:00.613","ValueAsString":"1.000"},{"ChargerId":"[REDACTED_CHARGER_ID]","StateId":201,"Timestamp":"2025-06-02T17:11:59.783","ValueAsString":"[REDACTED_NUMERIC_VALUE]"},{"ChargerId":"[REDACTED_CHARGER_ID]","StateId":202,"Timestamp":"2025-06-02T17:11:59.787","ValueAsString":"[REDACTED_NUMERIC_VALUE]"},{"ChargerId":"[REDACTED_CHARGER_ID]","StateId":204,"Timestamp":"2025-06-02T17:11:59.787","ValueAsString":"[REDACTED_NUMERIC_VALUE]"},{"ChargerId":"[REDACTED_CHARGER_ID]","StateId":205,"Timestamp":"2025-06-02T17:11:59.787","ValueAsString":"[REDACTED_NUMERIC_VALUE]"},{"ChargerId":"[REDACTED_CHARGER_ID]","StateId":206,"Timestamp":"2025-06-02T17:11:59.787","ValueAsString":"[REDACTED_NUMERIC_VALUE]"},{"ChargerId":"[REDACTED_CHARGER_ID]","StateId":207,"Timestamp":"2025-06-02T17:11:59.79","ValueAsString":"[REDACTED_NUMERIC_VALUE]"},{"ChargerId":"[REDACTED_CHARGER_ID]","StateId":220,"Timestamp":"2025-06-02T17:15:36.867","ValueAsString":"[REDACTED_NUMERIC_VALUE]"},{"ChargerId":"[REDACTED_CHARGER_ID]","StateId":221,"Timestamp":"2025-06-02T17:15:36.867","ValueAsString":"[REDACTED_NUMERIC_VALUE]"},{"ChargerId":"[REDACTED_CHARGER_ID]","StateId":270,"Timestamp":"2025-06-02T17:11:59.783","ValueAsString":"[REDACTED_NUMERIC_VALUE]"},{"ChargerId":"[REDACTED_CHARGER_ID]","StateId":501,"Timestamp":"2025-06-02T17:15:36.867","ValueAsString":"[REDACTED_NUMERIC_VALUE]"},{"ChargerId":"[REDACTED_CHARGER_ID]","State [... cut ...]
Jun 02 19:16:30 raspberrypi evcc[2195092]: [site  ] DEBUG 2025/06/02 19:16:30 pv 1 power: [REDACTED_NUMERIC_VALUE]W
Jun 02 19:16:30 raspberrypi evcc[2195092]: [powerwall] TRACE 2025/06/02 19:16:30 {"percentage":[REDACTED_PERCENTAGE]}
Jun 02 19:16:30 raspberrypi evcc[2195092]: [site  ] DEBUG 2025/06/02 19:16:30 battery 1 soc: [REDACTED_PERCENTAGE]%
Jun 02 19:16:30 raspberrypi evcc[2195092]: [site  ] DEBUG 2025/06/02 19:16:30 site power: [REDACTED_NUMERIC_VALUE]W
Jun 02 19:16:30 raspberrypi evcc[2195092]: [lp-1  ] DEBUG 2025/06/02 19:16:30 charger status: A

tbaehler, 2025-06-02:
As I thought, due to the 

charger logic error: disabled but charging

evcc calls enable(true), again and zaptec answers with 500 

Jun 02 19:15:30 raspberrypi evcc[2195092]: [zaptec] TRACE 2025/06/02 19:15:30 {"Code":528,"Details":"Charging is not Paused nor Scheduled; Resume command cannot be sent","StackTrace":null}

so the fix should help here.

andig, 2025-06-03:
@tbaehler made a suggestion in your PR, please give it a try

TimoD87, 2025-06-03:
In which nightly will it appear?

gimmel, 2025-06-03:
BTW: here is the official description from Zaptec:

Charger Command Validation Update

Zaptec has updated the sendCommand endpoint to prevent commands that would result in no operational change. Commands are now rejected if the charger is in a state where the command would have no effect.

This change enhances API efficiency, reduces unnecessary operations, and aligns with best practices for charger state validation.
https://docs.zaptec.com/page/charger-command-validation-update#/