evcc-io/evcc #21569: Config UI: Error using heating related attributes when adding a user-defined charger

Hoffmann77, 2025-06-01:
### Describe the bug

When using the config UI to add a user-defined charger it's not possible to use heating related attributes: 
"cannot create charger type 'custom': decoding failed due to the following error(s): '' has invalid keys: getmode, temp, templimit".

### Steps to reproduce

1. Click on add Charger/Loadpoint (Ladepunkt hinzufügen).
2. Select user-defined device (Benutzerdefiniertes Gerät).
3. Uncomment heating related attributes like 'getmode', 'temp', 'templimit' or 'setmode'.
4. Try to save the configuration.

### Configuration details

```yaml
Config UI
```

### Log details

```text
No Log entries.
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.1

naltatis, 2025-06-01:
Any chance we can simplify the `sgready` configuration? In many cases it's an embedded switch socket / charger (shelly, ...) configuration for which we have a proper template that we could show as a form.

What do you mean with `switchsocket`? Currently I see templates with type `heatpump` and `sgready` that we need to handle.

In general we had multiple requests make configuring heating more approachable. Currently you have to add a "Loadpoint" and a "Charger". I'm thinking of introducing a dedicated area/list for "Heating" which uses more appropriate wording and offers a different set of device options (group heating, custom, heat pump, sgready). Similar to the grid/pv/aux-meter distinction.

andig, 2025-06-02:
Aktuell gibts keinen, möglich wäre es aber:

```go
var cc struct {
	embed        `mapstructure:",squash"`
	Enabled      plugin.Config
	Enable       plugin.Config
	Power        plugin.Config
	Energy       *plugin.Config
	Soc          *plugin.Config
	StandbyPower float64
}
```

andig, 2025-06-02:
> Any chance we can simplify the sgready configuration? In many cases it's an embedded switch socket / charger (shelly, ...) configuration for which we have a proper template that we could show as a form.

`sgready` is it's own custom type made up of plugins.

andig, 2025-06-15:
Den gibts schon: https://github.com/evcc-io/evcc/issues/21569#issuecomment-2930015770

naltatis, 2025-06-15:
Der Charger selbst hat keinen Titel. Im UI wird immer der Titel das Ladepunktes verwendet.

closed 2025-07-07 by andig