evcc-io/evcc #21572: rct: missing second inverter production and missing consumption starting with 0.204.0

alexandermuch, 2025-06-01:
### Describe the bug

Dear evcc developers,
since the update to 0.204.0 (and now verified with 0.204.1) the integration of rct power systems is faulty. Two symptoms (visible from screenshot):
1. I have two inverters. It seems that only the pv production of the 'master' inverter is now used. Previously to 0.204.0 the total pv production was used.
2. The consumption of the house is not shown anymore. Only battery charging, charger and grid export. The consumption is always displayed as 0.0 kW.

[image]

There are now warnings or strange in the log, even when set to debug level.
Thank you,
Alex

### Steps to reproduce

1. Simply start evcc with the same configuration as has been used previously
2.
3.
...

### Configuration details

```yaml
meters:
  - name: my_grid
    type: template
    template: rct-power
    usage: grid
    host: xx.xx.xx.xx # IP-Adresse oder Hostname
  - name: my_pv
    type: template
    template: rct-power
    usage: pv
    host: xx.xx.xx.xx # IP-Adresse oder Hostname 
  - name: my_battery
    type: template
    template: rct-power
    usage: battery
    host: xx.xx.xx.xx # IP-Adresse oder Hostname
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/06/01 13:39:09 evcc 0.204.1
[main  ] INFO 2025/06/01 13:39:09 using config file: /config/evcc.yaml
[main  ] INFO 2025/06/01 13:39:09 listening at :7070
[site  ] INFO 2025/06/01 13:39:14 site config:
[site  ] INFO 2025/06/01 13:39:14   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/06/01 13:39:14     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/06/01 13:39:14     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/06/01 13:39:14     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2025/06/01 13:39:14   vehicles:
[site  ] INFO 2025/06/01 13:39:14     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2025/06/01 13:39:14   tariffs:
[site  ] INFO 2025/06/01 13:39:14     grid:      priceforecast
[site  ] INFO 2025/06/01 13:39:14     feed-in:   pricestatic
[site  ] INFO 2025/06/01 13:39:14     co2:       ✓
[site  ] INFO 2025/06/01 13:39:14     solar:     ✓
[lp-1  ] INFO 2025/06/01 13:39:14 loadpoint 1:
[lp-1  ] INFO 2025/06/01 13:39:14   mode:        off
[lp-1  ] INFO 2025/06/01 13:39:14   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2025/06/01 13:39:14   meters:      charge ✓
[lp-1  ] INFO 2025/06/01 13:39:14     charge:    power ✓ energy ✓ currents ✓
[site  ] DEBUG 2025/06/01 13:39:14 set buffer soc: 80
[site  ] DEBUG 2025/06/01 13:39:14 set priority soc: 10
[site  ] DEBUG 2025/06/01 13:39:14 ----
[site  ] DEBUG 2025/06/01 13:39:14 battery 1 power: 3W
[site  ] DEBUG 2025/06/01 13:39:14 pv 1 power: 2463W
[site  ] DEBUG 2025/06/01 13:39:14 grid power: -817W
[site  ] DEBUG 2025/06/01 13:39:15 battery 1 soc: 98%
[site  ] DEBUG 2025/06/01 13:39:15 site power: -714W
[lp-1  ] INFO 2025/06/01 13:39:15 car connected
[lp-1  ] INFO 2025/06/01 13:39:15 start charging ->
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.0 and 0.204.1

alexandermuch, 2025-06-01:
Hallo @Maschga 
Wenn ich den Parameter `externalpower ` hinzufüge und auf `true ` setze, dann scheint sich nichts zum Positiven zu ändern.

Maschga, 2025-06-01:
Derzeit ist es ein Breaking Change, der die Benutzer verwirrt.

andig, 2025-06-01:
Ja, müssten wir als BC markieren. Findest du den PR noch?

andig, 2025-06-01:
Zumindest ist es dokumentiert, daher würde ich hier schliessen.

closed 2025-06-01 by andig