evcc-io/evcc #21593: Config UI: Ladestand Auto immer (wenn bekannt) im Dashboard anzeigen

jori0001, 2025-06-02:
Ich nutze EVCC viel mehr als die Tesla-App und die Huawei-App.
In der EVCC Konfiguration sieht man, dass EVCC den Ladestand des Autos immer kennt (auch wenn es nicht angeschlossen ist):

[image]

Von daher wäre es toll, wenn der Ladestand auch im Dashboard angezeigt werden würde:

[image]

Könnt/wollt ihr das einbauen?

Grüße...

jori0001, 2025-06-02:
Ah ok. In dem Fall könnte man ja einbauen, dass der Ladestand alle x Minuten automatisch abgefragt wird.

0x3d13f, 2025-06-02:
Das kann man so konfigurieren, es ist aber dringend von zu häufigem Abfragen des Fahrzeugs abzuraten:
https://docs.evcc.io/docs/reference/configuration/loadpoints#pollmode

jori0001, 2025-06-02:
Sehr gut, dann fehlt ja nur noch die Anzeige aufm Dashboard, auch wenn das Auto nicht angeschlossen ist...