evcc-io/evcc #21603: FATAL: ocpp timeout, after charge point connects

scotthernandez, 2025-06-02:
### Describe the bug

I've configured my ocpp template and have my charger setup with the correct server address but sometimes I get this error, and sometimes it works. I've enabled debug logging but I don't see what the cause could be:

> [main ] INFO 2025/06/02 22:55:16 evcc 0.203.6
> [main ] INFO 2025/06/02 22:55:16 using config file: /etc/evcc.yaml
> [db ] INFO 2025/06/02 22:55:16 using sqlite database: /root/.evcc/evcc.db
> [mqtt ] INFO 2025/06/02 22:55:17 connecting evcc-1488174973 at tcp://192.168.1.5:1883
> [mqtt ] DEBUG 2025/06/02 22:55:17 tcp://192.168.1.5:1883 connected
> [main ] INFO 2025/06/02 22:55:17 listening at :7070
> [40252204784191-1] DEBUG 2025/06/02 22:55:17 waiting for chargepoint: 5m0s
> [ocpp ] DEBUG 2025/06/02 22:55:43 charge point connected: 40252204784191
> [main ] FATAL 2025/06/02 22:56:13 charger [vista-garage] cannot create charger 'vista-garage': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
> [main ] FATAL 2025/06/02 22:56:13 will attempt restart in: 15m0s

Here is the config:
```
chargers:
  - name: vista-garage
    type: template
    template: ocpp
    idtag: 1
    stationid: 40252204784191
```

I have not done a nightly build as this is deployed with docker/latest.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
see above
```

### Log details

```text
see above
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.6

mfuchs1984, 2025-06-03:
Needs cop trace.

romestylez, 2025-06-04:
Im having the exact same error ! Its all working fine the whole day today and after some docker restarts (changing config) i just have this

```
[ocpp ] INFO 2025/06/04 12:35:41 charge point connected, registering: EVB-P22046467
[ocpp ] DEBUG 2025/06/04 12:37:33 charge point disconnected: EVB-P22046467
[ocpp ] DEBUG 2025/06/04 12:37:50 charge point connected: EVB-P22046467
[ocpp ] DEBUG 2025/06/04 12:57:27 charge point disconnected: EVB-P22046467
[ocpp ] DEBUG 2025/06/04 12:57:50 charge point connected: EVB-P22046467
```

I restarted the Wallbox (EVbox Elvi) also did recreated the container but the error is still the same. If someone can tell me how to create a ocpp trace i would be down to do it :)

scotthernandez, 2025-06-10:
Here is the trace:

> [main ] INFO 2025/06/10 21:03:53 evcc 0.204.1
> [main ] INFO 2025/06/10 21:03:53 using config file: /etc/evcc.yaml
> [main ] INFO 2025/06/10 21:03:53 listening at :7070
> [ocpp ] DEBUG 2025/06/10 21:04:20 charge point connected: 40252204784191
> [ocpp ] TRACE 2025/06/10 21:04:20 send 40252204784191: [2,"79302239","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
> [ocpp ] TRACE 2025/06/10 21:04:20 recv 40252204784191: [3,"79302239",{"status":"Accepted"}]
> [ocpp ] TRACE 2025/06/10 21:04:20 send 40252204784191: [2,"2232284784","GetConfiguration",{}]
> [ocpp ] TRACE 2025/06/10 21:04:21 recv 40252204784191: [2,"119041904","BootNotification",{"chargePointVendor":"WWWW","chargePointModel":"SingleSocketCharger","chargePointSerialNumber":"40252204784191","firmwareVersion":"AC_DUL_2.19GS"}]
> [ocpp ] TRACE 2025/06/10 21:04:21 send 40252204784191: [3,"119041904",{"currentTime":"2025-06-10T19:04:21Z","interval":60,"status":"Accepted"}]
> [ocpp ] TRACE 2025/06/10 21:04:21 recv 40252204784191: [2,"119042105","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available"}]
> [ocpp ] TRACE 2025/06/10 21:04:21 send 40252204784191: [3,"119042105",{}]
> [ocpp ] TRACE 2025/06/10 21:04:41 recv 40252204784191: [2,"119044006","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing"}]
> [ocpp ] TRACE 2025/06/10 21:04:41 send 40252204784191: [3,"119044006",{}]
> [main ] FATAL 2025/06/10 21:04:50 charger [vista-garage] cannot create charger 'vista-garage': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
> [main ] FATAL 2025/06/10 21:04:50 will attempt restart in: 15m0s
> 
> [ocpp ] TRACE 2025/06/10 21:05:22 recv 40252204784191: [2,"119052007","Heartbeat",{}]
> [ocpp ] TRACE 2025/06/10 21:05:22 send 40252204784191: [3,"119052007",{"currentTime":"2025-06-10T19:05:22Z"}]
> [ocpp ] TRACE 2025/06/10 21:06:23 recv 40252204784191: [2,"119062108","Heartbeat",{}]
> [ocpp ] TRACE 2025/06/10 21:06:23 send 40252204784191: [3,"119062108",{"currentTime":"2025-06-10T19:06:23Z"}]
> [ocpp ] TRACE 2025/06/10 21:06:40 recv 40252204784191: [2,"119063909","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing"}]
> [ocpp ] TRACE 2025/06/10 21:06:40 send 40252204784191: [3,"119063909",{}]
> [ocpp ] TRACE 2025/06/10 21:07:24 recv 40252204784191: [2,"119072210","Heartbeat",{}]
> [ocpp ] TRACE 2025/06/10 21:07:24 send 40252204784191: [3,"119072210",{"currentTime":"2025-06-10T19:07:24Z"}]
> [ocpp ] TRACE 2025/06/10 21:08:25 recv 40252204784191: [2,"119082311","Heartbeat",{}]
> [ocpp ] TRACE 2025/06/10 21:08:25 send 40252204784191: [3,"119082311",{"currentTime":"2025-06-10T19:08:25Z"}]
> [ocpp ] TRACE 2025/06/10 21:08:40 recv 40252204784191: [2,"119083912","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Preparing"}]
> [ocpp ] TRACE 2025/06/10 21:08:40 send 40252204784191: [3,"119083912",{}]

mfuchs1984, 2025-06-10:
There is no response to the GetConfiguration request.