evcc-io/evcc #21629: WAHLWEISE Darstellung bei MEHREREN CHARGERN UNTER EINANDER ?

typxxi, 2025-06-04:
[image]

IST: Hier werden im Web 3 in einer Darstellung 2 pro Zeile dargestellt und zusammengerückt.

Wünschenswert wäre eine einfache Option, alle 3 "CHARGER" untereinander anzuzeigen, so dass jeder breiter wird und besser erkennbar gerade für Sehbehinderte. 

Eine Alternative habe ich  nicht, weil der Platz nun mal mit 2 zu klein wird. 

P.S.: Ich habe danach in der Doku gesucht und leider nix dazu gefunden bzw. ggf. übersehen, wenn es das schon geben sollte. 
DANKE SCHÖN

StefanSchoof, 2025-06-04:
Siehe #7235

andig, 2025-06-04:
/cc @naltatis wir haben kein Trackingissue dazu?

typxxi, 2025-06-04:
Ich habe mir den Ansatz hier angeschaut
> Siehe [#7235](https://github.com/evcc-io/evcc/discussions/7235)

Ich finde das erst einmal OK und von daher kann das hier geschlossen werden, wenn es unter 7235 weitergeht.
Bis dahin kann ich damit leben, wie es ist und auch, dass UI hohe Priorität genießt, wenn ich auch bei meinen 2 Seh Behinderten ungern Abstriche mache wie bei den Leistungsbalken und deren Höhe. Ich betreue deren Installation nur und muss dann immer erklären, was noch geht und was nicht bzw. warum nicht. Wie gesagt, das sind Menschen, die mit 49 Zoll Bildschirmen arbeiten müssen, um was zu erkennen, während es seltsamerweise mit der Distanz gut geht. 

Wenn was bei der Visualisierung passiert, dann warten wir das ab 

DANKE

typxxi, 2025-06-04:
> mal mit dem Zoom Level des Browser spielen. Kann sein, dass du gerade vor dieser Grenze bist.

Ja, Versuch macht klug , aber das hatte ich schon gemacht und brachte nix, weil dann alles vorbei ist, was man lesen nennt.

[image]

ab 33% bin ich in dem Club dabei, wo der browser umbricht auf das Zielformat und dann alles auf 1 Seite.

Am Ende hast Du ja das in dem verlinkten Topic sicher irgendwo drin und wir warten einfach, wie die Reise weiter geht.
Wichtiger war, dass es jetzt zu laufen beginnt nach 8 Monaten Stress mit am Ende den Hardwareherstellern bzw. deren Firmware bzw. WALLOX und deren PULSAR PLUS Varianten der letzten Jahre. 

Von daher bestellen bzw. kaufen wir morgen die Lizenz und warten, bis das von Dir genannte Thema in ein Release kommt. 
Nach 8 Monaten ist das mit den Wundern und morgen (seitens der Firmware für die Wallboxen) gegessen. 

DANKE - schönen Abend noch

naltatis, 2025-06-04:
Quickfix Möglichkeiten: Inzwischen gibts ja die Möglichkeit User-CSS https://github.com/evcc-io/evcc/pull/21263 mit einzubinden. Damit kannst du dir das gewünschte Layout (oben im Link siehst du ja was geändert werden muss) auch fix selbst basteln.

typxxi, 2025-06-07:
> Quickfix Möglichkeiten: Inzwischen gibts ja die Möglichkeit User-CSS [#21263](https://github.com/evcc-io/evcc/pull/21263) mit einzubinden. Damit kannst du dir das gewünschte Layout (oben im Link siehst du ja was geändert werden muss) auch fix selbst basteln.

Danke,
hab es gelesen und absolut nix beim Beispiel ohne Kopf und Fuß verstanden, außer dass die beiden verschwanden und sich die Farben änderten. 
Aber wie ich da 3 "Wallboxen" untereinander anordnen soll, geht über mein Verständnis.

Das LIZENZ kaufen ging auch daneben, denn die Eigentümer, für die ich das machen darf, haben keine und auch der Ratschlag, dass das jede EC Karte heute könne , half auch nix, denn die erforderliche Prüfziffer "cvv" oder so fehlt.

Und damit war mal eben evcc Lizenz kaufen auch nicht drin. Gibt es noch eine andere Zahlmöglichkeit ?
Ist ja eine Katastrophe, wenn es nicht 1 Alternative gibt - also ein eigener Store oder was auch immer. Die Gebühren a la PayPal kann ich ja zahlen, aber wie ? 

DANKE