evcc-io/evcc #21642: Problem withs eebus heartbeat between EVCC and controlbox

JKoetzing, 2025-06-04:
### Describe the bug

1. The timeout for monitoring the heartbeats is 2 minutes. Nevertheless, the connection to the control box is considered interrupted if no heartbeat is received from the control box in the first 10 seconds.

2. EVCC sends a heartbeat every 2 seconds. The interval is too short and spams the log file. The heartbeats should be sent every 60 seconds at the same interval as the control box.

### Steps to reproduce

1. In EVCC configure eebus and hems
2. Use eebus-cbsim for simulation of the controlbox
3. 10 seconds after the connection between EVCC and the contolbox was established, the eebus system enters the failsafe mode

Please refer to the logs below. Failsafe mode is entered 10 seconds after connection was established.
[eebus ] DEBUG 2025/06/03 18:27:52 ski 60dea5b2aa072d0baac7a1e48b8115308ca5f91e connected
[eebus ] WARN 2025/06/03 18:28:02 missing heartbeat- entering failsafe mode

### Configuration details

```yaml
eebus:
  interfaces:
    - lo
  certificate:
    public: |
      -----BEGIN CERTIFICATE-----
      ...
      -----END CERTIFICATE-----
    private: |
      -----BEGIN EC PRIVATE KEY-----
      ...
      -----END EC PRIVATE KEY-----

hems:
  type: eebus
  ski: ...
```

### Log details

```text
[eebus ] INFO 2025/06/03 18:27:50 Local SKI: 352263edcfdf39b95ac0f749880d92a26f230680
[eebus ] DEBUG 2025/06/03 18:27:50 starting websocket server on :4712
[eebus ] DEBUG 2025/06/03 18:27:50 mdns: announce
[eebus ] DEBUG 2025/06/03 18:27:50 mdns: using avahi
[eebus ] DEBUG 2025/06/03 18:27:50 mdns: avahi - ignoring service as its interface is not in the allowed list: EVCC_HEMS_01
[eebus ] DEBUG 2025/06/03 18:27:52 mdns: new - ski: 60dea5b2aa072d0baac7a1e48b8115308ca5f91e name: Demo-ControlBox-123456789 brand: Demo model: ControlBox typ: ElectricitySupplySystem identifier: Demo-ControlBox-123456789 register: false host: joachim-Virtual-Machine.local port: 4711 addresses: [127.0.0.1]
[eebus ] DEBUG 2025/06/03 18:27:52 mdns: avahi - ignoring service as its interface is not in the allowed list: Demo-ControlBox-123456789
[eebus ] DEBUG 2025/06/03 18:27:52 trying to connect to 60dea5b2aa072d0baac7a1e48b8115308ca5f91e at joachim-Virtual-Machine.local
[eebus ] DEBUG 2025/06/03 18:27:52 initiating connection to 60dea5b2aa072d0baac7a1e48b8115308ca5f91e at joachim-Virtual-Machine.local:4711/ship/
[eebus ] DEBUG 2025/06/03 18:27:52 ski 60dea5b2aa072d0baac7a1e48b8115308ca5f91e connected
[eebus ] DEBUG 2025/06/03 18:27:53 Send:  read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2025/06/03 18:27:53 Recv: d:_n:Demo_ControlBox-123456789:NodeManagement to NodeManagement read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2025/06/03 18:27:53 Send: d:_n:Demo_ControlBox-123456789 reply 2 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2025/06/03 18:27:53 Recv: d:_n:Demo_ControlBox-123456789:NodeManagement to NodeManagement reply 2 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2025/06/03 18:27:53 Send: d:_n:Demo_ControlBox-123456789 call 3 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2025/06/03 18:27:53 Send: d:_n:Demo_ControlBox-123456789 read 4 NodeManagementUseCaseData
[eebus ] DEBUG 2025/06/03 18:27:53 cs-lpc: 1 DeviceDiagnosis Server found
[eebus ] DEBUG 2025/06/03 18:27:53 cs-lpp: 1 DeviceDiagnosis Server found
[eebus ] DEBUG 2025/06/03 18:27:53 Send: d:_n:Demo_ControlBox-123456789 call 5 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2025/06/03 18:27:53 Send: d:_n:Demo_ControlBox-123456789 call 6 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2025/06/03 18:27:53 Send: d:_n:Demo_ControlBox-123456789 read 7 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2025/06/03 18:27:53 Recv: d:_n:Demo_ControlBox-123456789:NodeManagement to NodeManagement call 3 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2025/06/03 18:27:53 Send: d:_n:Demo_ControlBox-123456789 result 8 3 ResultData 0
[eebus ] DEBUG 2025/06/03 18:27:53 Recv: d:_n:Demo_ControlBox-123456789:NodeManagement to NodeManagement read 4 NodeManagementUseCaseData
[eebus ] DEBUG 2025/06/03 18:27:53 Send: d:_n:Demo_ControlBox-123456789 reply 9 4 NodeManagementUseCaseData
[eebus ] DEBUG 2025/06/03 18:27:53 Recv: d:_n:Demo_ControlBox-123456789:NodeManagement to NodeManagement result 5 3 ResultData 0
[eebus ] DEBUG 2025/06/03 18:27:53 Recv: d:_n:Demo_ControlBox-123456789:NodeManagement to NodeManagement call 6 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2025/06/03 18:27:53 Send: d:_n:Demo_ControlBox-123456789 result 10 6 ResultData 0
[eebus ] DEBUG 2025/06/03 18:27:53 Recv: d:_n:Demo_ControlBox-123456789:NodeManagement to NodeManagement reply 7 4 NodeManagementUseCaseData
[eebus ] DEBUG 2025/06/03 18:27:53 Recv: d:_n:Demo_ControlBox-123456789:NodeManagement to NodeManagement call 8 NodeManagementBindingRequestCall
[... 90 lines cut ...]
[eebus ] DEBUG 2025/06/03 18:29:22 Send: d:_n:Demo_ControlBox-123456789 notify 70 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2025/06/03 18:29:24 Send: d:_n:Demo_ControlBox-123456789 notify 71 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2025/06/03 18:29:26 Send: d:_n:Demo_ControlBox-123456789 notify 72 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2025/06/03 18:29:28 Send: d:_n:Demo_ControlBox-123456789 notify 73 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2025/06/03 18:29:30 Send: d:_n:Demo_ControlBox-123456789 notify 74 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2025/06/03 18:29:32 Send: d:_n:Demo_ControlBox-123456789 notify 75 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2025/06/03 18:29:34 Send: d:_n:Demo_ControlBox-123456789 notify 76 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2025/06/03 18:29:36 Send: d:_n:Demo_ControlBox-123456789 notify 77 DeviceDiagnosisHeartbeatData
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

Note that i run the current trunk system using go run.

andig, 2025-06-04:
> The timeout for monitoring the heartbeats is 2 minutes. Nevertheless, the connection to the control box is considered interrupted if no heartbeat is received from the control box in the first 10 seconds.

Not sure why- this will need analysis with debugger. Help welcome!

> EVCC sends a heartbeat every 2 seconds. The interval is too short and spams the log file. The heartbeats should be sent every 60 seconds at the same interval as the control box.

evcc as such doesn't send anything afaia. Seems this is part of the eebus library and probably adheres to spec. Maybe @DerAndereAndi can advise.

DerAndereAndi, 2025-06-04:
The spec says to send an heartbeat every <= 60s. And if the entity is also supporting other use cases like EMobility, then it shall use the minimal required heartbeat for all.

There will be optimizations in the next version of the stack, but nevertheless, this is all fine and according to the spec.

An heartbeat should be interpreted as missing in LPC, if for 120s there is NO heartbeat received.

I haven't looked into the usage of the stack in evcc, but only from the data being sent, this is all fine and valid.

JKoetzing, 2025-06-05:
> An heartbeat should be interpreted as missing in LPC, if for 120s there is NO heartbeat received.

This should be the case, but is not. This is not a question but an error. You can see this clearly from the logs. 10 seconds after the connection between EVCC and the contolbox was established, the eebus system enters the failsafe mode. You can trust the logs or test it for yourself.
I have a solution for this error.

>The spec says to send an heartbeat every <= 60s. And if the entity is also supporting other use cases like EMobility, then it shall use the minimal required heartbeat for all.

Perhaps the heartbeat frequency shall be configurable. I could also implement this.

closed 2025-06-06 by andig