evcc-io/evcc #21667: Sungrow SH Batterieladestand auf 100% eingefroren

hudri4, 2025-06-06:
### Describe the bug

Vermutlich seit Update ist der Batterieladezustand über eine Sungrow Hybrid Wechselrichter SH auf 100% obwohl tatsächlich 29% eingefroren, damit funktioniert die bevorzugte Hausbatterieladung nicht mehr. Alle anderen Werte des Wechselrichters auch Lade und Entladeleistung der Batterie funktionieren.
Zur Info: Die Abfrage der Batterie über Homeassistant funktioniert noch richtig.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *

interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
database:
  type: sqlite
  dsn: /.evcc/evcc.db

# token, 15000 days
sponsortoken: *

meters:
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.168.*
  port: 502  
  usage: grid  
  modbus: tcpip  
  timeout: 20s  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.168.*
  port: 502  
  usage: pv  
  modbus: tcpip  
  timeout: 20s  
  name: pv2
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.168.*
  usage: battery  
  modbus: tcpip  
  timeout: 20s  
  name: battery3

chargers:
- type: template
  template: ocpp-fronius-wattpilot
  timeout: 2m # optional
  connector: 1
  idtag: no-card
  stationid: wattpilot2
  meterinterval: 10s 
  name: wallbox4

- type: template
  template: ocpp-fronius-wattpilot
  timeout: 2m # optional
  connector: 1
  idtag: no-card
  stationid: wattpilot1
  meterinterval: 10s 
  name: wallbox1	  

loadpoints:
- title: Garage vorne
  charger: wallbox4
  mode: pv
  soc:
    poll:
      mode: always
      interval: 60m
    estimate: true
  phases: 0

- title: Garage hinten
  charger: wallbox1
  mode: pv
  soc:
    poll:
      mode: always
      interval: 60m
    estimate: true
  phases: 0

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3

vehicles:
  - name: SonjasModelY
    type: template
    template: tesla
    title: Sonjas Model Y
    clientId: *    # von developer.tesla.com
    accessToken:  *
    refreshToken: *
    capacity: 57.5 # Akkukapazität in kWh (optional) 
    vin: * # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)

  - name: ClausModelY
    type: template
    template: tesla
    title: Claus Model Y
    clientId: *     # von developer.tesla.com
    accessToken: *   # von myteslamate.com
    refreshToken: * # von myteslamate.com)
    capacity: 75 # Akkukapazität in kWh (optional) 
    vin: * # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)

#tarrifs:
#  currency: EUR # (default EUR)
#  grid:
#    # static grid price
#    type: fixed
#    price: 0.1164 # [currency]/kWh

#  feedin:
#    # rate for feeding excess (pv) energy to the grid
#    type: custom
#    price: 
#      source: mqtt
#      topic: HA/OEMAG
#      scale: 0.01
#      timeout: 20s

# mqtt message broker
mqtt:
  broker: 192.168.168.*
  topic: evcc # root topic for publishing, set empty to disable
  user: *
  password: *
  

# Influx Anbindung
influx:
  url: http://localhost:8086
  database: evcc_log
  token: *
  org:  *
```

### Log details

```text
kein Fehler gemeldet
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

V0.204.1

StefanSchoof, 2025-06-06:
> Vermutlich seit Update

Update von evcc oder der Sungrow Firmware?

hudri4, 2025-06-06:
EVCC, habe am Wechselrichter kein Update durchgeführt, kann aber auch nicht zu  100% ausschließen, dass Sungrow ein Remoteupdate ausgeführt hat. Mein Bruder hat bei gleicher Hardware WR das gleiche Problem

hudri4, 2025-06-06:
Neustartversuche evcc haben nichts gebracht. Ich gabe auch versucht über die GUI eine neue Battery hinzuzufügen und mit prüfen getestet, meldet auch 100% zurück

StefanSchoof, 2025-06-06:
Dann als ersten Schritt einmal auf die funktionierende evcc Version zurück und überprüfen, ob es wirklich an einer Änderung in evcc liegt.

hudri4, 2025-06-06:
Es dürfte doch an einem Firmwareupdate des Sungrow Wechselrichters liegen, laut Log wurde am 05.06.25 um 15:24 ein Firmwareupdate ohne mein Wissen installiert.

Type WR:  SH10RT-V112

Firmware: 
SDSP | SUBCTL-S_04011.01.01
MDSP | SAPPHIRE-H_03011.51.03
LCD | SAPPHIRE-H_01011.51.04

Geräteversion: A2292760807

andig, 2025-06-06:
Wir nutzen:

```yaml
  soc:
    source: modbus
    {{- include "modbus" . | indent 2 }}
    register:
      address: 13022 # Battery level
      type: input
      decode: int16
    scale: 0.1
```

> Evcc fragt über den WinetS Dongle ab (letztes Upgrade WinetS fand am 23.05 statt, 

Dann änder das doch mal

> kann als Fehlerursache ausgeschlossen werden)

anscheinend nicht- irgendwo muss der Unterschied ja sein. Bei evcc hat sich nichts geändert.