evcc-io/evcc #21673: Zaptec unexpected status: 500 (Internal Server Error) and not charging

spasdev, 2025-06-06:
### Describe the bug

EVCC is throwing an error message 

```
[zaptec] TRACE 2025/06/06 08:41:36 POST https://api.zaptec.com/api/chargers/XXX/sendCommand/507
[zaptec] TRACE 2025/06/06 08:41:36 {"Code":528,"Details":"Charging is not Paused nor Scheduled; Resume command cannot be sent","StackTrace":null}
[lp-1 ] ERROR 2025/06/06 08:41:36 charger enable: unexpected status: 500 (Internal Server Error)
[httpd ] TRACE 2025/06/06 08:42:05 GET /api/system/log
```
with Zaptec Go 2 (firmware 3.0.2.1) and not starting the charging as it should.

The error is not happening in every cases but on a regular basis. 

### Steps to reproduce

1. start the evcc and wait till the charger is triggered for charging or force charging with clicking "fast"

[evcc-20250606-091522-trace.log](https://github.com/user-attachments/files/20623533/evcc-20250606-091522-trace.log)

[evcc.yaml.txt](https://github.com/user-attachments/files/20623306/evcc.yaml.txt)

note: latest nightly was throwing other errors with vehicle. thus could not test it properly.

### Configuration details

```yaml
Zaptec Go 2 (firmware 3.0.2.1) and attached evcc configuration
```

### Log details

```text
see attached logs (500 (Internal Server Error))
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.1

andig, 2025-06-06:
>I have verified that the issue is reproducible with the latest nightly build

Did you?

andig, 2025-06-06:
>Same issue here

Bitte trace log

ttomasini, 2025-06-06:
Thanks for your quick refly @andig! I'm happy to provide the requested trace log snippet 

It seems that there are some internal issues with Zaptec's API at the moment.

Context: The Chargers, as well as the Chargers State Call, were successful beforehand and provided the necessary information.

```log
[zaptec] TRACE 2025/06/06 17:33:11 GET https://api.zaptec.com/api/installation/1cafdcc7-195e-4a7e-b52a-52adbb53c59a
[zaptec] TRACE 2025/06/06 17:33:11 {"Code":512,"Details":"bb07c1c0-9669-4a2b-86ca-609c75226209","StackTrace":null}
[main  ] FATAL 2025/06/06 17:33:11 charger [wallbox1] cannot create charger 'wallbox1': cannot create charger type 'template': cannot create charger type 'zaptec': unexpected status: 500 (Internal Server Error)
[main  ] FATAL 2025/06/06 17:33:11 will attempt restart in: 15m0s
```

andig, 2025-06-07:
/cc @tbaehler

tbaehler, 2025-06-08:
strange..
no clue what 512 means here?
the call should work with normal user permission.

Do you get more details if you do the call on the zaptec api directly?

you can try every call here 
https://api.zaptec.com/

click on api documentation (bottom of the page)
click authorize...

then go to installations and use the try out button & execuse do issue a calls.

ttomasini, 2025-06-08:
One more thing: I looked into the trace log of @spasdev and I've seen that this issue has another root cause. The common factor is that the error handling for unexpected responses is not sufficient:

```log
[zaptec] TRACE 2025/06/06 09:13:25 POST https://api.zaptec.com/api/installation/f10e5305-8fd8-4b1c-88a8-2152ebca0eca/update
[zaptec] TRACE 2025/06/06 09:13:25 {"availableCurrentPhase1":16,"availableCurrentPhase2":16,"availableCurrentPhase3":16}
[lp-1  ] DEBUG 2025/06/06 09:13:25 switched phases: 3p
[zaptec] TRACE 2025/06/06 09:13:25 POST https://api.zaptec.com/api/chargers/caae7c1a-1a64-4a57-a45f-19b6bbeb0207/update
[zaptec] TRACE 2025/06/06 09:13:26 {"maxChargeCurrent":16}
[lp-1  ] DEBUG 2025/06/06 09:13:26 set charge current limit: 16A
[zaptec] TRACE 2025/06/06 09:13:26 POST https://api.zaptec.com/api/chargers/caae7c1a-1a64-4a57-a45f-19b6bbeb0207/sendCommand/507
[zaptec] TRACE 2025/06/06 09:13:26 {"Code":528,"Details":"Charging is not Paused nor Scheduled; Resume command cannot be sent","StackTrace":null}
[lp-1  ] ERROR 2025/06/06 09:13:26 charger enable: unexpected status: 500 (Internal Server Error)
```

I am not sure if we should split these two issues into separate tickets or keep them in this thread anyway?

tbaehler, 2025-06-08:
Thanks for testing. 

One approach here would be to ignore the error (512). We did this already for the enable call. 

The Installation call here is just needed to detect, if the charges is a zaptec go 2, so we can just assume that it is a zaptec1 or pro if the call fails. 
I think you are right,  the problem is related to a permission issue. As you do not see any installations with your user.

tbaehler, 2025-06-08:
> Code":528,

this is fixed with https://github.com/evcc-io/evcc/issues/21566

tbaehler, 2025-06-08:
I will come up with a PR to ignore the error this evening....

andig, 2025-06-08:
Installation update is only required for phase switching. If getting installation details fails we can hide phase switching (which wouldn't work in load balancing setups anyway).

closed 2025-06-09 by andig