evcc-io/evcc #21689: Keba: Bug in enabled logic

premultiply, 2025-06-07:
I think something is not quite right with the enabled status. When charging to a limit, everything works fine. 

Then charging stops when car has reached the limit - I have a fixed limit in my car for 90%. In evcc I also selected 90%.
When this happens, evcc goes into a endless loop of trying to disable the charger. The charger is already disabled but still the wallbox reports enabled. For then on I get the error **charger out of sync: expected disabled, got enabled** every few minutes.

The app shows no charging session, but somehow the "stop charging" symbol is visible on the bottom. Seems like the charger is not fully disabled.
[image]

The car has no active charging session.
[image]

When I press the stop button in the Keba app, the error stops.

Log:
```
[lp-1 ] DEBUG 2025/06/06 12:44:44 charge power: 1W
[lp-1 ] DEBUG 2025/06/06 12:44:44 charge currents: [0.311 0.235 0.004]A
[site ] DEBUG 2025/06/06 12:44:44 grid power: -6156W
[site ] DEBUG 2025/06/06 12:44:44 pv 1 power: 7747W
[site ] DEBUG 2025/06/06 12:44:44 grid currents: [5.44 10.4 10.3]A
[site ] DEBUG 2025/06/06 12:44:44 site power: -5956W
[lp-1 ] DEBUG 2025/06/06 12:44:44 charge total import: 89.144kWh
[lp-1 ] DEBUG 2025/06/06 12:44:44 charger status: C
[lp-1 ] DEBUG 2025/06/06 12:44:45 soc estimated: 86.61% (vehicle: 85.00%)
[lp-1 ] DEBUG 2025/06/06 12:44:45 vehicle soc: 87%
[lp-1 ] DEBUG 2025/06/06 12:44:45 vehicle range: 217km
[lp-1 ] DEBUG 2025/06/06 12:44:45 pv charge current: 15.3A = 2.31A + 12.9A (-5956W @ 2p)
[lp-1 ] DEBUG 2025/06/06 12:44:45 set charge current limit: 15.3A
[site ] DEBUG 2025/06/06 12:44:45 solar forecast: accumulated 119.781kWh, produced 230.170kWh, scale 1.922
[site ] DEBUG 2025/06/06 12:45:14 ----
[lp-1 ] DEBUG 2025/06/06 12:45:14 charge power: 1W
[lp-1 ] DEBUG 2025/06/06 12:45:14 charge currents: [0.31 0.235 0.004]A
[site ] DEBUG 2025/06/06 12:45:14 grid power: -6145W
[site ] DEBUG 2025/06/06 12:45:14 pv 1 power: 7762W
[site ] DEBUG 2025/06/06 12:45:14 grid currents: [5.47 10.4 10.3]A
[site ] DEBUG 2025/06/06 12:45:14 site power: -5945W
[lp-1 ] DEBUG 2025/06/06 12:45:14 charge total import: 89.144kWh
[lp-1 ] DEBUG 2025/06/06 12:45:14 charger status: C
[lp-1 ] DEBUG 2025/06/06 12:45:14 soc estimated: 86.61% (vehicle: 85.00%)
[lp-1 ] DEBUG 2025/06/06 12:45:14 vehicle soc: 87%
[lp-1 ] DEBUG 2025/06/06 12:45:14 vehicle range: 217km
[lp-1 ] DEBUG 2025/06/06 12:45:14 pv charge current: 15.2A = 2.31A + 12.9A (-5945W @ 2p)
[lp-1 ] DEBUG 2025/06/06 12:45:15 set charge current limit: 15.2A
[site ] DEBUG 2025/06/06 12:45:15 solar forecast: accumulated 119.816kWh, produced 230.230kWh, scale 1.922
[site ] DEBUG 2025/06/06 12:45:44 ----
[lp-1 ] DEBUG 2025/06/06 12:45:44 charge power: 0W
[lp-1 ] DEBUG 2025/06/06 12:45:44 charge currents: [0.265 0.199 0.004]A
[site ] DEBUG 2025/06/06 12:45:44 grid power: -6023W
[site ] DEBUG 2025/06/06 12:45:44 pv 1 power: 7758W
[site ] DEBUG 2025/06/06 12:45:44 grid currents: [5.55 10.4 9.88]A
[site ] DEBUG 2025/06/06 12:45:44 site power: -5823W
[lp-1 ] DEBUG 2025/06/06 12:45:44 charge total import: 89.144kWh
[lp-1 ] DEBUG 2025/06/06 12:45:44 charger status: B
[lp-1 ] DEBUG 2025/06/06 12:45:44 charge final energy: 4.806kWh
[lp-1 ] INFO 2025/06/06 12:45:44 stop charging <-
[lp-1 ] DEBUG 2025/06/06 12:45:44 charge total import: 89.144kWh
[lp-1 ] DEBUG 2025/06/06 12:45:46 vehicle status: B
[lp-1 ] DEBUG 2025/06/06 12:45:46 vehicle soc: 90%
[lp-1 ] DEBUG 2025/06/06 12:45:46 vehicle range: 222km
[lp-1 ] DEBUG 2025/06/06 12:45:46 limitSoc reached: 90.0% > 90%
[lp-1 ] DEBUG 2025/06/06 12:45:46 charger disable
[lp-1 ] DEBUG 2025/06/06 12:45:46 wake-up timer: stop
[site ] DEBUG 2025/06/06 12:45:46 solar forecast: accumulated 119.852kWh, produced 230.300kWh, scale 1.922
[site ] DEBUG 2025/06/06 12:46:14 ----
[lp-1 ] DEBUG 2025/06/06 12:46:14 charge power: 0W
[lp-1 ] DEBUG 2025/06/06 12:46:14 charge currents: [0.005 0.004 0.004]A
[site ] DEBUG 2025/06/06 12:46:14 grid power: -6642W
[site ] DEBUG 2025/06/06 12:46:14 pv 1 power: 7756W
[site ] DEBUG 2025/06/06 12:46:15 grid currents: [8.37 9.87 9.84]A
[site ] DEBUG 2025/06/06 12:46:15 site power: -6442W
[... 129 lines cut ...]
[site ] DEBUG 2025/06/06 12:51:15 solar forecast: accumulated 120.234kWh, produced 230.990kWh, scale 1.921
[site ] DEBUG 2025/06/06 12:51:44 ----
[lp-1 ] DEBUG 2025/06/06 12:51:44 charge power: 0W
[lp-1 ] DEBUG 2025/06/06 12:51:44 charge currents: [0.004 0.004 0.004]A
[site ] DEBUG 2025/06/06 12:51:44 grid power: -7432W
[site ] DEBUG 2025/06/06 12:51:44 pv 1 power: 7783W
[site ] DEBUG 2025/06/06 12:51:44 grid currents: [10.2 10.4 10.6]A
[site ] DEBUG 2025/06/06 12:51:44 site power: -7232W
[lp-1 ] DEBUG 2025/06/06 12:51:44 charge total import: 89.144kWh
[lp-1 ] DEBUG 2025/06/06 12:51:44 charger status: B
[lp-1 ] DEBUG 2025/06/06 12:51:45 limitSoc reached: 90.0% > 90%
[site ] DEBUG 2025/06/06 12:51:45 solar forecast: accumulated 120.269kWh, produced 231.050kWh, scale 1.921
[site ] DEBUG 2025/06/06 12:52:14 ----
[lp-1 ] DEBUG 2025/06/06 12:52:15 charge power: 0W
[lp-1 ] DEBUG 2025/06/06 12:52:15 charge currents: [0.004 0.004 0.004]A
[site ] DEBUG 2025/06/06 12:52:15 grid power: -7419W
[site ] DEBUG 2025/06/06 12:52:15 pv 1 power: 7778W
[site ] DEBUG 2025/06/06 12:52:15 grid currents: [10.2 10.4 10.6]A
[site ] DEBUG 2025/06/06 12:52:15 site power: -7219W
[lp-1 ] DEBUG 2025/06/06 12:52:15 charge total import: 89.144kWh
[lp-1 ] DEBUG 2025/06/06 12:52:15 charger status: B
[lp-1 ] WARN 2025/06/06 12:52:15 charger out of sync: expected disabled, got enabled
```

_Originally posted by @crazyfx1 in https://github.com/evcc-io/evcc/discussions/14552#discussioncomment-13389251_

andig, 2025-06-07:
Missing config

andig, 2025-06-07:
Kann jemand bestätigen, dass sich der Freigabestatus der Box über Register 5014 auslesen lässt? So wie er vorher geschrieben wurde?

crazyfx1, 2025-06-07:
Eventuell gibt's hier auch Unterschiede zwischen der Keba P30 und P40.

crazyfx1, 2025-06-07:
Bei mir tritt das auf mit einer Keba P40 mit der Config

```JSON
{"host":"192.168.1.229","id":255,"modbus":"tcpip","template":"keba-modbus"}
```

crazyfx1, 2025-06-07:
Stimmt die Abfrage so?
Wenn ja, dann scheint es nicht zu funktionieren.
[image]

[image]

andig, 2025-06-07:
Die P40 verwendet das Stromregister für enable.

andig, 2025-06-07:
> 2: The charging station is ready for charging and waits for a reaction from the electric vehicle.

Also enabled=true, alles richtig

premultiply, 2025-06-07:
Könnte jemand bitte einfach mal den PR https://github.com/evcc-io/evcc/pull/21690 testen?

Idealerweise mit P30 und P40.

andig, 2025-06-07:
> Ja, evcc will aber die Wallbox disablen, weil der SOC erreicht ist. Leider bleibt die Wallbox im Status == 2 und liefert somit ewig enabled. Deshalb gibts alle paar Minuten den Fehler im Log.

Exakt. Und disabled ist sie nicht mehr „ready“, leider aber anscheinend schon. Warum?!

crazyfx1, 2025-06-08:
> Könnte jemand bitte einfach mal den PR https://github.com/evcc-io/evcc/pull/21690 testen?
> 
> Idealerweise mit P30 und P40.

Ich kann es morgen testen. Jedoch denke ich nicht, dass es funktioniert. Das Register kann man nicht auslesen. Siehe mein Kommentar oben.

andig, 2025-06-08:
...und das war der Grund für die krude Lösung. Die ist allerdings kaputt- da würde vielleicht mal ein Ticket bei KEBA helfen, denn der Status funktionioniert nicht wie dokumentiert!

crazyfx1, 2025-06-08:
Evtl ist das Enable auch noch nicht implementiert, in der Doku steht's so drin:
[image]

crazyfx1, 2025-06-08:
Hab mal reindebugged und jetzt erst bemerkt, dass bei der P40 alles rein über das 5004 Register gesteuert wird. Das 5014 wird gar nicht verwendet. 
Ich kann morgen testen, welchen Status die Wallbox liefert, wenn der Fehler auftritt.

crazyfx1, 2025-06-09:
@premultiply Beim Start der Ladung kommt einmalig dieser Fehler:
**WARN 2025/06/09 09:50:28 charger logic error: disabled but charging**
Sieht so aus als könnte man das Register erst lesen, sofern schon mal drauf geschrieben wurde.

```
[site ] DEBUG 2025/06/09 09:49:58 solar forecast: accumulated 0.052kWh, produced 0.060kWh, scale 1.150
[site ] DEBUG 2025/06/09 09:50:28 ----
[lp-1 ] DEBUG 2025/06/09 09:50:28 charge power: 7134W
[lp-1 ] DEBUG 2025/06/09 09:50:28 charge currents: [15.4 15.3 0.004]A
[site ] DEBUG 2025/06/09 09:50:28 grid power: -6265W
[site ] DEBUG 2025/06/09 09:50:28 pv 1 power: 7587W
[site ] DEBUG 2025/06/09 09:50:28 grid currents: [8.93 7.26 10.4]A
[site ] DEBUG 2025/06/09 09:50:28 site power: -6065W
[lp-1 ] DEBUG 2025/06/09 09:50:28 charge total import: 89.453kWh
[lp-1 ] DEBUG 2025/06/09 09:50:28 charger status: C
[lp-1 ] DEBUG 2025/06/09 09:50:28 charge start energy: 89.453kWh
[lp-1 ] INFO 2025/06/09 09:50:28 car connected
[lp-1 ] DEBUG 2025/06/09 09:50:28 pv timer elapse
[lp-1 ] DEBUG 2025/06/09 09:50:28 pv timer inactive
[lp-1 ] DEBUG 2025/06/09 09:50:28 charge total import: 89.453kWh
[lp-1 ] DEBUG 2025/06/09 09:50:28 charge start energy: 89.453kWh
[lp-1 ] INFO 2025/06/09 09:50:28 start charging ->
[lp-1 ] DEBUG 2025/06/09 09:50:28 wake-up timer: stop
[lp-1 ] DEBUG 2025/06/09 09:50:28 vehicle status: A
[lp-1 ] DEBUG 2025/06/09 09:50:28 vehicle soc: 25%
[lp-1 ] DEBUG 2025/06/09 09:50:28 vehicle range: 53km
[lp-1 ] WARN 2025/06/09 09:50:28 charger logic error: disabled but charging
[lp-1 ] DEBUG 2025/06/09 09:50:28 charger enable
[lp-1 ] DEBUG 2025/06/09 09:50:28 pv charge current: 13.2A = 0A + 13.2A (-6065W @ 2p)
[lp-1 ] DEBUG 2025/06/09 09:50:28 pv timer reset
[lp-1 ] DEBUG 2025/06/09 09:50:28 pv timer inactive
[lp-1 ] DEBUG 2025/06/09 09:50:28 set charge current limit: 13.2A
[site ] DEBUG 2025/06/09 09:50:28 solar forecast: accumulated 0.095kWh, produced 0.130kWh, scale 1.373
[site ] DEBUG 2025/06/09 09:50:59 ----
```

crazyfx1, 2025-06-09:
Hab sonst noch etwas getestet, scheint alles zu gehen. Auch der Fehler beim Erreichen des konfigurierten SOC ist weg. Wenn man von Aus auf PV schaltet, dann kommt auch nicht mehr der Fehler (charger logic error: disabled but charging). Scheint nur beim ersten Anstecken zu kommen. 

Hab eine Keba P40 mit MID-Zähler.

closed 2025-06-18 by andig