evcc-io/evcc #21705: Timeout for Nissan Ariya

geert-claes, 2025-06-07:
I tried to add the timeout parameter to the vehicles section in the evcc.yaml.   The parameter is not accepted.
I see the parameters in other vehicles.

Is it possible to add the paramater to the Nissan Ariya ?

andig, 2025-06-08:
For what purpose do you need it?

andig, 2025-06-09:
As usual, a trace log will bring more clarity where those 20s timeouts might be coming from. Adding a timeout parameter wont help since, as said, it‘s already 2min.

closed 2025-06-09 by andig

geert-claes, 2025-06-09:
supposing you are looking for this :
```
[main ] INFO 2025/06/09 11:31:30 evcc 0.132.0 (a558aa2d)
[main ] INFO 2025/06/09 11:31:30 using config file: /etc/evcc.yaml
[db ] INFO 2025/06/09 11:31:30 using sqlite database: /var/lib/evcc/evcc.db
[db ] TRACE 2025/06/09 11:31:30 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db ] TRACE 2025/06/09 11:31:30 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db ] TRACE 2025/06/09 11:31:30 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db ] TRACE 2025/06/09 11:31:30 SELECT * FROM `settings` 3 <nil>
[db ] TRACE 2025/06/09 11:31:30 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db ] TRACE 2025/06/09 11:31:30 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db ] TRACE 2025/06/09 11:31:30 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db ] TRACE 2025/06/09 11:31:30 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db ] TRACE 2025/06/09 11:31:30 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db ] TRACE 2025/06/09 11:31:30 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db ] TRACE 2025/06/09 11:31:30 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[main ] INFO 2025/06/09 11:31:30 listening at :7070
[db ] TRACE 2025/06/09 11:31:30 SELECT * FROM `configs` WHERE `configs`.`class` = 2 0 <nil>
[db ] TRACE 2025/06/09 11:31:30 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
[db ] TRACE 2025/06/09 11:31:30 SELECT * FROM `configs` WHERE `configs`.`class` = 3 0 <nil>
[nissan] TRACE 2025/06/09 11:31:30 POST https://prod.eu2.auth.kamereon.org/kauth/json/realms/root/realms/a-ncb-prod/authenticate
[main ] ERROR 2025/06/09 11:31:35 creating vehicle 2FAF failed: cannot create vehicle type 'template': cannot create vehicle type 'nissan': login failed: Post "https://prod.eu2.auth.kamereon.org/kauth/json/realms/root/realms/a-ncb-prod/authenticate": dial tcp: lookup prod.eu2.auth.kamereon.org: i/o timeout
[main ] FATAL 2025/06/09 11:31:35 failed configuring loadpoints: missing loadpoints
[main ] FATAL 2025/06/09 11:31:35 will attempt restart in: 15m0s
```

andig, 2025-06-09:
DNS lookup fails with timeout. It shouldn‘t. Nothing to do with configured HTTP request timeout.

andig, 2025-06-09:
Yes, establishing the tcp connection has its own timeouts in Go.

geert-claes, 2025-06-09:
I'll explan the detail.
(My) Prerequisites : 
- Need the SOC of the vehicle (and only that).
- AD is involved, fixed IP is not an option.
- Pi-hole is used to control/log traffic to the internet (src and dst).
therefore :
- First (and second) NS needs to be AD controllers -> Name resolution for local.
- Third NS is Pi-Hole -> Name resolution for non-local

Works like a charm for all (any host and any applications) except for EVCC (mostly because 2 cname lookups and 1 A lookup is needed to get the final ip address).

Chaning timeout and attemps in resolv.conf is a solution, but is overwritten when ip adres is renewed.

Any suggestion is welcome (not touching the prerequisites)..