evcc-io/evcc #21708: Phasenumschaltung von 1 auf 3 Phasen aufgrund des Ladevehaltens funktioniert nicht

MetaStack-Exchange, 2025-06-08:
### Describe the bug

Die Phasenumschaltung meiner EVC04 (Softwarestand: v3.187.0) funktioniert nicht. Die Umschaltung von 3 zu 1 Phase funktioniert. Das "hochschalten" aber nicht immer. Wurde erst einmal auf 1phasigen Betrieb gewechsel, schaltet die WB beim Schnellladen oder auch geplanten laden nicht wieder auf 3 Phasen. Ein manueller Wechsel auf 3phasigen Betrieb ist dann auch nicht mehr möglich. Wird dieser Wechsel häufiger manuell angesoßen, wird irgendwann der Betriebsmodus "automatisch" ausgeblendet und erscheint erst nach einem Neustart der Wallbox wieder. Ich konnte aber schon beobachte, dass das bei der automatischen Umschaltung aufgrund hoher Sonneneinstrahlung funktioniert hat. Ein manueller Wechsel auf 3phasigen Betrieb ist dann auch nicht mehr möglich. Wird dieser Wechsel häufiger manuell angesoßen, wird irgendwann der Betriebsmodus "automatisch" ausgeblendet und erscheint erst nach einem Neustart der Wallbox wieder.

### Steps to reproduce

1. automatischer Wechsel auf 1phasiges Laden
2. Wechsel zu schnellem Laden oder Beginn der Ladeplanung

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 192.168.1.9
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: ist Vorhanden

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true # use `deny` to raise modbus errors

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  - name: my_grid
[... 232 lines cut ...]
  #   priority: <priority>
  #   tags: <tags>
```

### Log details

```text
folgen
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.1

MetaStack-Exchange, 2025-06-08:
Hier nun noch die logs:

´´´
[vestel] WARN 2025/06/07 12:33:32 failed to parse firmware version "\x00v\x003\x00.\x001\x008\x007\x00.\x000\x00-\x001\x00.\x008\x00.\x001\x005\x006\x00.\x000\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00": Malformed version:  3 . 1 8 7 . 0 - 1 . 8 . 1 5 6 . 0                                                                
[lp-1  ] DEBUG 2025/06/07 12:33:36 charge total import: 2433.600kWh
[site  ] INFO 2025/06/07 12:33:37 site config:
[site  ] INFO 2025/06/07 12:33:37   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/06/07 12:33:37     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2025/06/07 12:33:37     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/06/07 12:33:37     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2025/06/07 12:33:37   vehicles:
[site  ] INFO 2025/06/07 12:33:37     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2025/06/07 12:33:37   tariffs:
[site  ] INFO 2025/06/07 12:33:37     grid:      pricestatic
[site  ] INFO 2025/06/07 12:33:37     feed-in:   pricestatic
[site  ] INFO 2025/06/07 12:33:37     co2:       ✗
[site  ] INFO 2025/06/07 12:33:37     solar:     ✓
[lp-1  ] INFO 2025/06/07 12:33:37 loadpoint 1:
[lp-1  ] INFO 2025/06/07 12:33:37   mode:        off
[lp-1  ] INFO 2025/06/07 12:33:37   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2025/06/07 12:33:37   meters:      charge ✓
[lp-1  ] INFO 2025/06/07 12:33:37     charge:    power ✓ energy ✓ currents ✓
[site  ] DEBUG 2025/06/07 12:33:37 set buffer soc: 90
[site  ] DEBUG 2025/06/07 12:33:37 set buffer start soc: 0
[site  ] DEBUG 2025/06/07 12:33:37 set priority soc: 85
[site  ] DEBUG 2025/06/07 12:33:37 set battery discharge control: true
[lp-1  ] DEBUG 2025/06/07 12:33:37 phase timer inactive
[lp-1  ] DEBUG 2025/06/07 12:33:37 pv timer inactive
[site  ] DEBUG 2025/06/07 12:33:38 ----
[lp-1  ] DEBUG 2025/06/07 12:33:38 charge power: 4W
[lp-1  ] DEBUG 2025/06/07 12:33:38 charge currents: [0.049 0.018 0.018]A
[site  ] DEBUG 2025/06/07 12:33:38 grid power: -8198W
[site  ] DEBUG 2025/06/07 12:33:38 grid powers: [-2780 -2800 -2619]W
[site  ] DEBUG 2025/06/07 12:33:38 pv 1 power: 8620W
[site  ] DEBUG 2025/06/07 12:33:38 battery 1 power: -7W
[site  ] DEBUG 2025/06/07 12:33:38 grid currents: [-11.9 -11.9 -11.1]A
[site  ] DEBUG 2025/06/07 12:33:38 battery 1 soc: 93%
[site  ] DEBUG 2025/06/07 12:33:38 site power: -8105W
[lp-1  ] DEBUG 2025/06/07 12:33:38 charge voltages: [235 236 237]V
[lp-1  ] DEBUG 2025/06/07 12:33:38 charge total import: 2433.600kWh
[lp-1  ] DEBUG 2025/06/07 12:33:38 charger status: A
[lp-1  ] INFO 2025/06/07 12:33:38 car disconnected
[lp-1  ] DEBUG 2025/06/07 12:33:38 set battery boost: false
[lp-1  ] DEBUG 2025/06/07 12:33:38 set session soc limit: 0
[lp-1  ] DEBUG 2025/06/07 12:33:38 set session energy limit: 0
[lp-1  ] DEBUG 2025/06/07 12:33:57 set phases: 0
[site  ] DEBUG 2025/06/07 12:33:57 ----
[lp-1  ] DEBUG 2025/06/07 12:33:57 charge power: 4W
[lp-1  ] DEBUG 2025/06/07 12:33:57 charge currents: [0.049 0.018 0.018]A
[site  ] DEBUG 2025/06/07 12:33:57 grid power: -8195W
[site  ] DEBUG 2025/06/07 12:33:57 grid powers: [-2775 -2797 -2623]W
[site  ] DEBUG 2025/06/07 12:33:57 pv 1 power: 8622W
[site  ] DEBUG 2025/06/07 12:33:57 battery 1 power: -7W
[site  ] DEBUG 2025/06/07 12:33:57 grid currents: [-11.9 -11.9 -11.2]A
[site  ] DEBUG 2025/06/07 12:33:57 battery 1 soc: 93%
[site  ] DEBUG 2025/06/07 12:33:57 site power: -8102W
[lp-1  ] DEBUG 2025/06/07 12:33:57 charge voltages: [235 236 237]V
[lp-1  ] DEBUG 2025/06/07 12:33:57 charge total import: 2433.600kWh
[lp-1  ] DEBUG 2025/06/07 12:33:57 charger status: A
[site  ] DEBUG 2025/06/07 12:34:08 ----
[lp-1  ] DEBUG 2025/06/07 12:34:08 charge power: 4W
[lp-1  ] DEBUG 2025/06/07 12:34:08 charge currents: [0.049 0.018 0.018]A
[site  ] DEBUG 2025/06/07 12:34:08 grid power: -8246W
[site  ] DEBUG 2025/06/07 12:34:08 grid powers: [-2797 -2812 -2637]W
[site  ] DEBUG 2025/06/07 12:34:08 pv 1 power: 8670W
[site  ] DEBUG 2025/06/07 12:34:08 battery 1 power: -1W
[site  ] DEBUG 2025/06/07 12:34:08 grid currents: [-12 -12 -11.2]A
[site  ] DEBUG 2025/06/07 12:34:08 battery 1 soc: 93%
[site  ] DEBUG 2025/06/07 12:34:08 site power: -8147W
[lp-1  ] DEBUG 2025/06/07 12:34:08 charge voltages: [235 236 237]V
[... 873 lines cut ...]
[lp-1  ] DEBUG 2025/06/07 14:19:47 wake-up timer: stop
[lp-1  ] DEBUG 2025/06/07 14:19:53 set charge mode: pv
[site  ] DEBUG 2025/06/07 14:19:53 ----
[lp-1  ] DEBUG 2025/06/07 14:19:53 charge power: 204W
[lp-1  ] DEBUG 2025/06/07 14:19:53 charge currents: [3.57 0.018 0.018]A
[site  ] DEBUG 2025/06/07 14:19:53 grid power: -3513W
[site  ] DEBUG 2025/06/07 14:19:53 grid powers: [-1215 -1222 -1076]W
[site  ] DEBUG 2025/06/07 14:19:53 pv 1 power: 1661W
[site  ] DEBUG 2025/06/07 14:19:53 battery 1 power: -370W
[site  ] DEBUG 2025/06/07 14:19:53 grid currents: [-5.25 -5.24 -4.63]A
[site  ] DEBUG 2025/06/07 14:19:53 battery 1 soc: 85%
[site  ] DEBUG 2025/06/07 14:19:53 battery has priority at soc 85% (< 85%)
[site  ] DEBUG 2025/06/07 14:19:53 site power: -3413W
[lp-1  ] DEBUG 2025/06/07 14:19:53 charge voltages: [234 235 235]V
[lp-1  ] DEBUG 2025/06/07 14:19:53 detected active phases: 1p
[... cut ...] port: 2438.700kWh
[lp-1  ] DEBUG 2025/06/07 14:19:53 charger status: B
[lp-1  ] INFO 2025/06/07 14:19:53 stop charging <-
[lp-1  ] DEBUG 2025/06/07 14:19:53 charge total import: 2438.700kWh
[lp-1  ] DEBUG 2025/06/07 14:19:56 vehicle status: B
[lp-1  ] DEBUG 2025/06/07 14:19:56 vehicle soc: 65%
[lp-1  ] DEBUG 2025/06/07 14:19:56 vehicle soc limit: 100%
[lp-1  ] DEBUG 2025/06/07 14:19:56 vehicle range: 218km
[lp-1  ] DEBUG 2025/06/07 14:19:56 pv charge current: 14.8A = 0A + 14.8A (-3413W @ 1p)
´´´

mfuchs1984, 2025-06-08:
Das Problem existiert schon von Anfang an, siehe bwsp. https://github.com/evcc-io/evcc/discussions/11701#discussioncomment-12372226

Entweder die Phasenumschaltung der Wallbox ist Buggy oder evcc muss die Umschaltung anders steuern. Die Modbus Spezifikation ist leider sehr oberflächlich.

andig, 2025-06-08:
Laut Beschreibung und Discussion scheint die Box ein Eigenleben zu haben. Da kann nur der Hersteller helfen :/

closed 2025-06-08 by andig

MetaStack-Exchange, 2025-06-08:
Ok, also lieber erstmal auf 3 Phasen lassen und nach jeden FW-Update Mal schauen, ob es vielleicht funktioniert... Schade aber OK

Vielen Dank für die Info!