evcc-io/evcc #21728: Loadpoint: split charging sessions um 0:00 Uhr

Tombra1889, 2025-06-08:
**Is your feature request related to a problem? Please describe.**
Sowohl für Wärmepumpen als auch für nachts ladende Fahrzeuge wäre es aus meiner Sicht sinnvoll, wenn ein Ladevorgang automatisch um 0:00 Uhr gesplittet würde und so besser in der Charging Sessions UI visualisiert werden.

**Describe the solution you'd like**
Bessere Ansicht von Tagesüberschreitenden Ladevorgängen.

**Describe alternatives you've considered**
Chargingsessions im Nachhinein splitten zu können.

naltatis, 2025-06-08:
Hi @Tombra1889,

die Ladevorgänge wollen wir bewusst nicht auftrennen. Für das Thema Auswertung (vor allem für Geräte ohne echte Ladevorgänge wie WPs) gibts bereits Pläne: https://github.com/evcc-io/evcc/issues/13050.

Daher schließe ich das hier mal als Duplikat.

naltatis, 2025-06-08:
@Tombra1889 sieh auch hier den Hinweis in der Doku für Schaltsteckdosen: https://docs.evcc.io/docs/features/sessions#schaltbare-steckdosen

closed 2025-06-08 by naltatis