evcc-io/evcc #21734: Evcc Sonnenanteil Berechnung falsch?

phL9r, 2025-06-09:
### Describe the bug

Hallo! Ich nutze evcc seit geraumer Zeit mit einer huawei sun 2000 Anlage sowie einem Luna 10kw Speicher und einer easee wallbox. Ich habe zusätzlich noch mittels Fritz DECT Steckdosen 2 Balkonanlagen als PV Meter konfiguriert..
Heute zum Beispiel zeigt evcc an, dass ich lediglich 81,7% des Ladestroms aus Sonnenenergie erwirtschaftet habe, was allerdings nicht sein kann, da mir die Anlage lediglich 0,39kwh Netzbezug ausgibt, was auch realistisch erscheint. Wenn ich den Sonnenanteil aus evcc dagegen halte müssten zum jetzigen Zeitpunkt bereits >2kWh aus dem Netz bezogen worden sein. Scheinbar regelt evcc sauber aber die Berechnung des Sonnenanteils wirkt falsch. Ich rechne damit mit meinen Vermietern ab, daher wäre es schön wenn jemand eine Lösung für das Problem hätte.

Vielen Dank im Voraus! 

### Steps to reproduce

1. E Auto mit easee WB laden
2. 
3.

[evcc-20250609-135812-debug.log](https://github.com/user-attachments/files/20654343/evcc-20250609-135812-debug.log)

...

### Configuration details

```yaml
Kommt sofort am pc
```

### Log details

```text
log liegt als Datei bei
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.2404.2

christianbaumann, 2025-06-12:
Ist mir auch schon augefallen - könnte das daran liegen, dass die Batterie mit zum Laden des Autos genutzt wird?

phL9r, 2025-06-14:
Nein eigentlich rechnet er den Ladestrom aus der Batterie als PV Strom, also nicht als Netzbezug, oder etwa nicht?

phL9r, 2025-06-17:
Bei mir klappt es jetzt übrigens. Ich habe das Update-Intervall auf 45 Sek. gestellt. Jetzt ist der PV Anteil bei annähernd 100%

naltatis, 2025-06-20:
Ok, dann mach ich hier erst einmal zu. Falls du reproduzierbar rausfindest in welchem Szenario falsch gezählt wird gerne ein neues Issue öffnen.

closed 2025-06-20 by naltatis