evcc-io/evcc #21735: Solax X3 Ultra // Battery ModBus register address vertauscht?

Pwnagetool, 2025-06-09:
### Describe the bug

Hallo Zusammen,

ich habe mein Problem bereits [hier ](https://github.com/evcc-io/evcc/discussions/21729) beschrieben.
Und ich glaube, dass hier im[ Solax Template](https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/solax.yaml) eine Modbus Register Adresse vertauscht ist

Ab Zeile 85

```
  maxacpower: {{ .maxacpower }} # W
  {{- end }}
  {{- if eq .usage "battery" }}
    source: modbus
    {{- include "modbus" . | indent 2 }}
    register:
      {{- if eq .storageunit "1" }}
      address: 22 # 0x0016 Batpower_Charge1
      {{- else }}
      address: 297 # 0x0129 Batpower_Charge2
      {{- end }}
      type: input
      decode: int16
    scale: -1
  soc:
    source: modbus
    {{- include "modbus" . | indent 2 }}
    register:
      {{- if eq .storageunit "1" }}
      address: 28 # 0x001C Battery 1 Capacity
      {{- else }}
      address: 301 # 0x012D Battery 2 Capacity
      {{- end }}
      type: input
      decode: uint16
```

Kann jemand mit Solax Wechselrichter das so bestätigen bzw. gibt es eine Möglichkeit das in meiner Konfiguration anzupassen? Meines Wissens nach sind die Templates in der Installation ja nicht anpassbar.

### Steps to reproduce

1. Im ursprünglichen Topic sieht man es ganz gut an den Screenshots

### Configuration details

```yaml
meters:
  # replace with your real grid meter
  - name: solax_inverter
    type: template
    template: solax
    usage: grid
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: x.x.x.x # Hostname
    port: 502 # Port 
    mppt3: true # Der Wechselrichter hat einen dritten PV-Eingang (MPPT3), optional
    storageunit: 0 # optional

  # replace with your real pv system
  - name: solax_pv
    type: template
    template: solax
    usage: pv
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: x.x.x.x # Hostname
    port: 502 # Port
    mppt3: true # Der Wechselrichter hat einen dritten PV-Eingang (MPPT3), optional
    storageunit: 0 # optional

  # replace with your real battery
  - name: solax_battery
    type: template
    template: solax
    usage: battery
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: x.x.x.x # Hostname
    port: 502 # Port
    mppt3: true # Der Wechselrichter hat einen dritten PV-Eingang (MPPT3), optional
```

### Log details

```text
Nicht im Log ersichtlich
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

Pwnagetool, 2025-06-09:
Nene ich habe nur eine Batterie an meinem Wechselrichter hängen :)
Wenn man dann nur eine Batterie konfiguriert, ist es klar, dass man nur einen der beiden Werte für die jeweilige Batterie bekommt:)

andig, 2025-06-09:
@Pwnagetool warum machst du es nicht einfach? Statt langer Diskussion, einfach Datei editieren und PR stellen ;) Sagt mehr als 1000 Worte...