evcc-io/evcc #21745: Zaptec (Pro) charging should not be started with maxChargeCurrent flag

ttomasini, 2025-06-09:
### Describe the bug

Following up on issue https://github.com/evcc-io/evcc/issues/21673 respectively pull request https://github.com/evcc-io/evcc/pull/21744 I was able to build and run the new evcc version. The old issue has indeed been fixed, but now I run into another issue with my Zaptec Pro charger _within the external managed load distribution system_ when I want to resume charging.

```log
[zaptec] TRACE 2025/06/09 20:03:16 POST https://api.zaptec.com/api/chargers/1cafdcc7-195e-4a7e-b52a-52adbb53c59a/update
[zaptec] TRACE 2025/06/09 20:03:16 {"maxChargeCurrent":16}
[lp-1 ] ERROR 2025/06/09 20:03:16 set charge current limit 16A: unexpected status: 403 (Forbidden)
```

Following the documentation at https://api.zaptec.com/help/index.html#/Charger/Charger_SendCommand_POST it is **NOT** recommended to use the `maxChargeCurrent` and `minChargeCurrent` properties. These are intended for special cases where an external party must override the Zaptec load balancer, which may cause unexpected behavior. It is recommended to control the charging system using the installation’s `AvailableCurrent` setting instead.

Apart from that, it looks like as part of a external managed Zaptec load distribution system I don't have access to set the maximum charging current via API. (therefore probably the HTTP 403 response)

Would a possible solution be to `request /api/chargers/{id}/sendCommand/{commandId}` (https://api.zaptec.com/help/index.html#/Charger/Charger_SendCommand_POST) with command `507` instead?

### Steps to reproduce

1. Own Zaptec Pro within a external managed load distribution system
2. Optional: Unlock Zaptec Pro charger with RFID card _or_ Zaptec App (I am not sure if this is actually necessary of if this is also something evcc could do? Please point me to the right direction if someone knows.
3. Pause charging (e.g. with intelligent charging based on current pricing)
4. Try to resume charging.

### Configuration details

```yaml
chargers:
- type: template
  template: zaptec
  id: 1cafdcc7-195e-4a7e-b52a-52adbb53c59a
  user: *****
  password: *****
  name: wallbox1
```

### Log details

```text
see above
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.2 (9761c0d3)

ttomasini, 2025-06-09:
> 2. Optional: Unlock Zaptec Pro charger with RFID card _or_ Zaptec App (I am not sure if this is actually necessary of if this is also something evcc could do? Please point me to the right direction if someone knows.

Slightly off-topic but I figured out how to authorize the Zaptec Pro charger (without starting the charging) avoid having to use the RFID tag or the app manually.

https://github.com/custom-components/zaptec/blob/master/custom_components/zaptec/api.py lists quite some undocumented API endpoints like

- `GET /api/dashboard`
- `GET /api/chargers/{chargerId}/live`
- `GET /api/chargers/{chargerId}/settings`
- `PUT /api/installation/{installationId}`
- _aand drum roll_ `POST /api/chargers/{chargerId}/authorizecharge`

As soon as the `authorizecharge` call has been executed using a valid bearer token, the charging station can be controlled until the next unplugging or `10001` (DeauthorizeAndStop) command. Unfortunately, I cannot operate the charging station in free mode because the RFID/app authorization is hard-set by my load management operator. Is this something evcc supports already? I've seen the yaml `authorize` key for Easee chargers but I am not sure if this reflects the same use case?

andig, 2025-06-10:
/cc @tbaehler

helijunky, 2025-06-10:
From my understanding the `availableCurrent` is affecting the whole installation (e.g. multiple chargers in a managed load distribution system). It is used to limit the power for all chargers on the same installation. There are [three modes](https://help.zaptec.com/hc/en-001/articles/30489134765329-How-do-charging-modes-work):

- default charging: available current can be manually controlled
- scheduled charging: available current is controlled by a time schedule
- automatic charging: available current is controlled by an associated energy sensor

On top of that, each user can control the currents of his own charger by setting `minChargeCurrent` and `maxChargeCurrent` and number of phases (however it seems this requires owner or service access rights). The `maxChargeCurrent` can only be reached if not limited by `availableCurrent`. 

see also https://docs.zaptec.com/docs/dynamic-load-balancing-with-the-zaptec-api

Does this make sense?

tbaehler, 2025-06-10:
authorizecharge very interesting....
will take a closer look

ttomasini, 2025-06-14:
@tbaehler thank you very much, please let me know if I can support you somehow.

Since my go skills are kind of rusty, I implemented a quick (but very dirty) local workaround to make my evcc setup work until there is a proper fix:

```diff
diff --git a/charger/zaptec.go b/charger/zaptec.go
index 42902132..91cfbdaa 100644
--- a/charger/zaptec.go
+++ b/charger/zaptec.go
@@ -265,7 +265,8 @@ func (c *Zaptec) MaxCurrent(current int64) error {
                MaxChargeCurrent: &curr,
        }

-       return c.chargerUpdate(data)
+       c.chargerUpdate(data)
+       return nil
 }

 var _ api.Meter = (*Zaptec)(nil)
```

andig, 2025-06-15:
@ttomasini this lopks exactly as I‘d expect- do you want to open a PR? Does it work?

ttomasini, 2025-06-15:
@andig thanks for your feedback, I really appreciate that. The mentioned solution works: The charging starts, resumes and stops as expected. But as I wrote, provided you unlock the station manually beforehand.

I am not familiar with the evcc codebase at all but initially I thought about the following additions to my mentioned patch. Unfortunately, I'm not sure if I can deliver sufficiently high code quality for a patch with my existing go know-how.

1. I was thinking of just ignoring the setting current error if `getInstallationMaxCurrent()` fails. This way the error handling will still take place if the setup allows it.
https://github.com/evcc-io/evcc/blob/9761c0d345060e649895ae55b4ec19a3c1404de6/charger/zaptec.go#L146-L152

2. I'm also not sure if it's common practice at evcc to still log via INFO for such errors?
3. Is it also common to hide the current settings for the charging station in the UI?
4. I also remember that you recommended hiding phase switching in the other issue (https://github.com/evcc-io/evcc/issues/21673#issuecomment-2953808273). Would it make sense to do this straight away?

ttomasini, 2025-06-15:
> > No, it is part of the installation settings as [@helijunky](https://github.com/helijunky) has pointed out so well. I don't think I would have the permissions to adjust this setting in my load balancing setup anyway.
> 
> Then that's no option. Seems we're done here, then?

You're absolutely right that the charging current probably can't be set in situations like my load balancing setup. Nevertheless, there would probably need to be a feasible solution so that the 403 error no longer occurs and possibly the settings in the UI are hidden, right?

Or am I misunderstanding your last question?

andig, 2025-06-15:
> Nevertheless, there would probably need to be a feasible solution so that the 403 error no longer occurs (and the charging starts without my dirty, manual fix) and possibly the settings in the UI are hidden, right?

Which solution would that be? Seems there is none. If we can't control charging current there's no PV mode.

ttomasini, 2025-06-15:
> Which solution would that be? Seems there is none. If we can't control charging current there's no PV mode.

Two reasons why it would be worth continuing to work on this issue:

- I have already opened a support ticket with Zaptec for possible solutions that we may have overlooked. I will report as soon as there is any new information.
- It would make the PV mode obsolete, but the smart mode (charging based on electricity price) would still be worth writing a fix for.

andig, 2025-07-06:
I'm  afraid reopening won't help if nobody wants to grab this for implementation.

ttomasini, 2025-07-07:
@andig I'll try to have a look at it this coming weekend and acquire the necessary go know-how and maybe @tbaehler also has time to point me to the right direction? The current hack I am using is no long term solution.