evcc-io/evcc #21753: EVCC seems always start with max. Power on Wallbe (maybe others as well)

AndyNew2, 2025-06-10:
### Describe the bug

I always observe following behavious.
The start and end criterias seemed to work quite well.
But once the start criteria is fullfilled, the loading starts with max. power of the charger to create a short peak also in Currnet power import from Grid.
This will be recognised quick by EVCC and the power is reduced. 
I would expect EVCC starts with the seens PC overpower instead starting always just with max. power.

It might be related to my Wallbox Wallbe, but would not expect, it is necessary to start at full power.
My Wallbe is set to 20A max. Power and just uses 1 phase due to the car I am charging (Nissan Leaf)

### Steps to reproduce

1. Connect Wallbox (Wallbe) to car in a situation not too much PV power is available.
2. Configure EVCC to PV mode
3. In my case, my start criteria need 1,4kW overpower
4. Once PV systems generates enough power, assuming in this case 2,1kW, it starts after timeout the charging. That is fine.
5. Once started, I always see a very short period it starts charging with 4,6kW (is max. supported by car and Wallbox).
6. That is seen by EVCC and the power is quickly reduced. 

However I do not understand, why EVCC starts with max. power. If a regulation cycle needs to be triggered a few rounds, I would propose to start with MIN power instead of MAX power. The best would be, if it straight aways started with the detected 2,1kW overpower, but I never see this on my installation.
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 127.0.0.1
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 6s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:
sponsortoken: eyJhbGXXX
# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

[... 633 lines cut ...]
[site  ] DEBUG 2025/06/10 07:58:27 grid currents: [1.32 2.16 1.3]A
[site  ] DEBUG 2025/06/10 07:58:27 battery has priority at soc 81% (< 100%)
[site  ] DEBUG 2025/06/10 07:58:27 site power: 161W
[lp-1  ] DEBUG 2025/06/10 07:58:27 charge voltages: [232 231 233]V
[lp-1  ] DEBUG 2025/06/10 07:58:27 detected connected phases: 3p
[lp-1  ] DEBUG 2025/06/10 07:58:27 charge total import: 124.961kWh
[lp-1  ] DEBUG 2025/06/10 07:58:27 charger status: B
[lp-1  ] DEBUG 2025/06/10 07:58:27 pv charge current: 0A = 0A + -0.7A (161W @ 1p)
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.2

andig, 2025-06-10:
Your interval is **MUCH** too low. That's why there is a warning. And why warnings should be observed.

closed 2025-06-10 by andig

AndyNew2, 2025-06-10:
That was quick, Thanks for the tip, never thought about that. I appreciate it still allows to set higher update rates. 
Do not change anything, great support.