evcc-io/evcc #21754: Add Slack as messaging plugin

marcelGoerentz, 2025-06-10:
**Is your feature request related to a problem? Please describe.**
I would like to implement  a Slack integration for evcc, so I can get messages from evcc via my mostly used app.

**Describe the solution you'd like**
I would implement a solution that will give the user the opportunity to use Slack without needing to set up a script.

**Describe alternatives you've considered**
It is possible to use a script for that or maybe the HTTP plugin. Nevertheless an integration of the service would make it easier for users to set it up correctly.

**Additional context**
None

andig, 2025-06-10:
Does Slack have an api? Please note that the core team won't be working on this feature request, but happy to take a PR.

marcelGoerentz, 2025-06-10:
They are using a webhook URL

bboehmke, 2025-06-10:
Am I wrong or does this already work with the [shoutrrr](https://containrrr.dev/shoutrrr) service:

https://docs.evcc.io/en/docs/reference/configuration/messaging#shout

See also the shoutrrr documentation for the Slack config:
https://containrrr.dev/shoutrrr/v0.8/services/slack/

marcelGoerentz, 2025-06-10:
You are right. Please close this issue.

andig, 2025-06-10:
You can click "close" any time ;)

closed 2025-06-10 by andig