evcc-io/evcc #21776: FritzDect mit Syntax error : "Ladeziegel: charge power: XML syntax error on line 34: element <link> closed by </head>

linuxstony, 2025-06-10:
### Describe the bug

Seit (ich meine) dem letzten Release funktioniert die Steuerung der FRITZ!Smart Energy 210 Steckdose nicht mehr mit o.g. Fehler

### Steps to reproduce

EVCC installieren
FritzDECT 210 konfigurieren. 

### Configuration details

```yaml
Über die UI eingerichtet
```

### Log details

```text
Keine weiteren Hinweise als die Meldung im Betreff.
Ich kann aber gerne noch einen Trace beifügen, wenn gewünscht.
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.1

VolkerK62, 2025-06-11:
meine FritzDect210 funktioniert ohne Probleme mit der aktuellsten Version.

linuxstony, 2025-06-12:
For future reference: I had to reconfigure the plug with http only (without https). Now it works like before.