evcc-io/evcc #21781: Zendure Hyber AB2000S Battery

na-ma-ste, 2025-06-11:
### Describe the bug

Hi everyone,

I wanted to integrate my Zendure Hyber battery into evcc and used the corresponding template in the documentation. However, I get the following error when I restart evcc:

meter: [zendure_battery] cannot create meter ‘zendure_battery’: cannot create meter type ‘template’: cannot create meter type ‘zendure’: unexpected status: 400 (Bad Request)

Adding the inverter (Hyber 2000) has worked via template under meters.

I have checked the type, but cannot find any errors compared to the documentation.

I look forward to the feedback.

Translated with www.DeepL.com/Translator (free version)

### Steps to reproduce

1.Enter Zendure Configuration into yaml
2. Restart evcc
3.
...

### Configuration details

```yaml
meters:
- name: zendure_battery
  type: template
  template: zendure
  usage: battery
  account: abc@mail.com
  serial: VU5D99F74021B04 # Zu finden in der Zendure App in den Einstellungen des Geräts
  region: EU # [EU, Global]
  timeout: 100m # Zeitüberschreitung, optional
#  capacity: 2 # Akkukapazität (kWh), optional
```

### Log details

```text
Jun 11 09:43:28 evcc-pi3b evcc[129382]: [main  ] INFO 2025/06/11 09:43:28 evcc 0.204.2
Jun 11 09:43:28 evcc-pi3b evcc[129382]: [main  ] INFO 2025/06/11 09:43:28 using config file: /etc/evcc.yaml
Jun 11 09:43:28 evcc-pi3b evcc[129382]: [db    ] INFO 2025/06/11 09:43:28 using sqlite database: /var/lib/evcc/evcc.db
Jun 11 09:43:28 evcc-pi3b evcc[129382]: [main  ] INFO 2025/06/11 09:43:28 listening at :7070
Jun 11 09:43:28 evcc-pi3b evcc[129382]: [zendure] INFO 2025/06/11 09:43:28 connecting  at tcp://mqtt-eu.zen-iot.com:1883
Jun 11 09:43:29 evcc-pi3b evcc[129382]: [zendure] DEBUG 2025/06/11 09:43:29 tcp://mqtt-eu.zen-iot.com:1883 connected
Jun 11 09:43:29 evcc-pi3b evcc[129382]: [zendure] DEBUG 2025/06/11 09:43:29 tcp://mqtt-eu.zen-iot.com:1883 subscribe f15ARcMj/#
Jun 11 09:43:29 evcc-pi3b evcc[129382]: [main  ] FATAL 2025/06/11 09:43:29 meter [zendure_battery] cannot create meter 'zendure_battery': cannot create meter type 'templa>
Jun 11 09:43:29 evcc-pi3b evcc[129382]: [main  ] FATAL 2025/06/11 09:43:29 will attempt restart in: 15m0s
Jun 11 09:47:02 evcc-pi3b systemd[1]: Stopping evcc.service - evcc...
Jun 11 09:47:02 evcc-pi3b systemd[1]: evcc.service: Main process exited, code=exited, status=1/FAILURE
Jun 11 09:47:02 evcc-pi3b systemd[1]: evcc.service: Failed with result 'exit-code'.
Jun 11 09:47:02 evcc-pi3b systemd[1]: Stopped evcc.service - evcc.
Jun 11 09:47:02 evcc-pi3b systemd[1]: evcc.service: Consumed 2.101s CPU time.
Jun 11 09:47:02 evcc-pi3b systemd[1]: Started evcc.service - evcc.
Jun 11 09:47:03 evcc-pi3b evcc[129402]: [main  ] INFO 2025/06/11 09:47:03 evcc 0.204.2
Jun 11 09:47:03 evcc-pi3b evcc[129402]: [main  ] INFO 2025/06/11 09:47:03 using config file: /etc/evcc.yaml
Jun 11 09:47:03 evcc-pi3b evcc[129402]: [db    ] INFO 2025/06/11 09:47:03 using sqlite database: /var/lib/evcc/evcc.db
Jun 11 09:47:03 evcc-pi3b evcc[129402]: [main  ] INFO 2025/06/11 09:47:03 listening at :7070
Jun 11 09:47:03 evcc-pi3b evcc[129402]: [zendure] INFO 2025/06/11 09:47:03 connecting  at tcp://mqtt-eu.zen-iot.com:1883
Jun 11 09:47:04 evcc-pi3b evcc[129402]: [zendure] DEBUG 2025/06/11 09:47:04 tcp://mqtt-eu.zen-iot.com:1883 connected
Jun 11 09:47:04 evcc-pi3b evcc[129402]: [zendure] DEBUG 2025/06/11 09:47:04 tcp://mqtt-eu.zen-iot.com:1883 subscribe f15ARcMj/#
Jun 11 09:47:04 evcc-pi3b evcc[129402]: [main  ] FATAL 2025/06/11 09:47:04 meter [zendure_battery] cannot create meter 'zendure_battery': cannot create meter type 'templa>
Jun 11 09:47:04 evcc-pi3b evcc[129402]: [main  ] FATAL 2025/06/11 09:47:04 will attempt restart in: 15m0s
Jun 11 09:57:52 evcc-pi3b systemd[1]: Stopping evcc.service - evcc...
Jun 11 09:57:52 evcc-pi3b systemd[1]: evcc.service: Main process exited, code=exited, status=1/FAILURE
Jun 11 09:57:52 evcc-pi3b systemd[1]: evcc.service: Failed with result 'exit-code'.
Jun 11 09:57:52 evcc-pi3b systemd[1]: Stopped evcc.service - evcc.
Jun 11 09:57:52 evcc-pi3b systemd[1]: evcc.service: Consumed 2.609s CPU time.
Jun 11 09:57:52 evcc-pi3b systemd[1]: Started evcc.service - evcc.
Jun 11 09:57:53 evcc-pi3b evcc[129430]: [main  ] INFO 2025/06/11 09:57:53 evcc 0.204.2
Jun 11 09:57:54 evcc-pi3b evcc[129430]: [main  ] INFO 2025/06/11 09:57:54 using config file: /etc/evcc.yaml
Jun 11 09:57:54 evcc-pi3b evcc[129430]: [db    ] INFO 2025/06/11 09:57:54 using sqlite database: /var/lib/evcc/evcc.db
Jun 11 09:57:54 evcc-pi3b evcc[129430]: [main  ] INFO 2025/06/11 09:57:54 listening at :7070
Jun 11 09:57:54 evcc-pi3b evcc[129430]: [zendure] INFO 2025/06/11 09:57:54 connecting  at tcp://mqtt-eu.zen-iot.com:1883
Jun 11 09:57:54 evcc-pi3b evcc[129430]: [zendure] DEBUG 2025/06/11 09:57:54 tcp://mqtt-eu.zen-iot.com:1883 connected
Jun 11 09:57:54 evcc-pi3b evcc[129430]: [zendure] DEBUG 2025/06/11 09:57:54 tcp://mqtt-eu.zen-iot.com:1883 subscribe f15ARcMj/#
Jun 11 09:57:54 evcc-pi3b evcc[129430]: [main  ] FATAL 2025/06/11 09:57:54 meter [zendure_battery] cannot create meter 'zendure_battery': cannot create meter type 'templa>
Jun 11 09:57:54 evcc-pi3b evcc[129430]: [main  ] FATAL 2025/06/11 09:57:54 will attempt restart in: 15m0s
Jun 11 09:58:38 evcc-pi3b systemd[1]: Stopping evcc.service - evcc...
Jun 11 09:58:38 evcc-pi3b systemd[1]: evcc.service: Main process exited, code=exited, status=1/FAILURE
Jun 11 09:58:38 evcc-pi3b systemd[1]: evcc.service: Failed with result 'exit-code'.
Jun 11 09:58:38 evcc-pi3b systemd[1]: Stopped evcc.service - evcc.
Jun 11 09:58:38 evcc-pi3b systemd[1]: evcc.service: Consumed 2.007s CPU time.
Jun 11 09:58:38 evcc-pi3b systemd[1]: Started evcc.service - evcc.
Jun 11 09:58:39 evcc-pi3b evcc[129452]: [main  ] INFO 2025/06/11 09:58:39 evcc 0.204.2
Jun 11 09:58:39 evcc-pi3b evcc[129452]: [main  ] INFO 2025/06/11 09:58:39 using config file: /etc/evcc.yaml
Jun 11 09:58:39 evcc-pi3b evcc[129452]: [db    ] INFO 2025/06/11 09:58:39 using sqlite database: /var/lib/evcc/evcc.db
Jun 11 09:58:39 evcc-pi3b evcc[129452]: [main  ] INFO 2025/06/11 09:58:39 listening at :7070
Jun 11 09:58:40 evcc-pi3b evcc[129452]: [zendure] INFO 2025/06/11 09:58:40 connecting  at tcp://mqtt-eu.zen-iot.com:1883
Jun 11 09:58:40 evcc-pi3b evcc[129452]: [zendure] DEBUG 2025/06/11 09:58:40 tcp://mqtt-eu.zen-iot.com:1883 connected
Jun 11 09:58:40 evcc-pi3b evcc[129452]: [zendure] DEBUG 2025/06/11 09:58:40 tcp://mqtt-eu.zen-iot.com:1883 subscribe f15ARcMj/#
Jun 11 09:58:40 evcc-pi3b evcc[129452]: [main  ] FATAL 2025/06/11 09:58:40 meter [zendure_battery] cannot create meter 'zendure_battery': cannot create meter type 'templa>
Jun 11 09:58:40 evcc-pi3b evcc[129452]: [main  ] FATAL 2025/06/11 09:58:40 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.2

andig, 2025-06-11:
> cannot create meter type 'templa>

Die relevante Fehlermeldung fehlt, bitte ergänzen. Bis dahin close.

StefanSchoof, 2025-06-11:
Siehe https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen und dort die Variante für das Speichern in einer Datei

na-ma-ste, 2025-06-11:
Hallo @StefanSchoof , danke dir. Das habe ich oben eigentlich "reinkopiert" unter Log Details. Genutzt hab ich  "sudo journalctl -u evcc -S "2023-03-21 07:00:00" -U "2023-03-21 08:00:00" -q > ~/evcc.log"

Hallo @andig, nun als Datei beigefügt.

[Logfile evcc.txt](https://github.com/user-attachments/files/20697780/Logfile.evcc.txt)

MaTr75, 2025-07-02:
Ich habe dasselbe Problem. Der Fehler ist noch nicht auf Reopen. Braucht es noch weitere Infos?

na-ma-ste, 2025-07-02:
HI @MaTr75 , welche Seriennummer nutzt du bei der Konfiguration? Du musst jeweils die Seriennummer des Hyber2000 nutzen (sowohl für PV als auch für Battery) und brauchst nicht die Seriennummer der Batterie selbst in die Konfiguration eintragen. Das hat bei mir das Problem gelöst -müsste ggf. in der Doku spezifiziert werden

na-ma-ste, 2025-07-02:
Hi @MaTr75 , versuch mal die "Experimental" zu aktivieren und es über die Config UI (ist dann unter dem Menüpunkt Konfiguration in der UI zu finden) den Speicher anzulegen. Ich hab das auch darüber am Ende gemacht und dann nahezu alle Element zur Config UI emigriert.

Ich beton es nur nochmal: In der Zendure App findest du 2 SNs: Einmal für den Wechselrichter (Hyber 2000) und dann jeweils immer eine SN für die Batteriemodule. Bitte immer (sowohl für den Typ PV als auch für den Typ Batterie bei evcc) die SN des Wechselrichters nehmen.

MaTr75, 2025-07-09:
Ja, das hat das Problem bei mir auch gelöst! Ich bin vom „Zweitaccount“ auf meinen echten Account gewechselt und jetzt läuft es wieder.

Mit dem Extra-Account hat es definitiv aber mal funktioniert 🤷‍♂️