evcc-io/evcc #21782: Meter SoC should take precedence over charger SoC for heating devices

rhuss, 2025-06-11:
**Is your feature request related to a problem? Please describe.**

When configuring a heating device (like AC-ELWA-2) with both a charger and a meter that provides SoC readings, the current implementation prioritizes the charger's SoC over the meter's SoC. For heating devices, this behavior is counterproductive because:

* Meter SoC are often more accurate for heating applications. In heating scenarios, the meter often provides more precise readings (e.g., temperature sensors via MQTT) compared to the charger's internal calculations. If a meter is configured, it is usually in addition to the charger, which typically displays readings for temperature and power.

* For heating devices like water heaters, the temperature (represented as SoC) from external sensors is more reliable than the charger's power-based estimates, and it also fluctuates less as they are farther away from the actual heating process. 

* When a user explicitly configures a meter with SoC capability for a heating device, they expect this to be used as the primary source.

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Current Behavior:

Currently, the system follows this priority order for all devices:
1. Charger SoC (if available)
2. Vehicle SoC (if available, but not applicable for heating devices)
3. Meter SoC (as fallback)

**Describe the solution you'd like**

For heating devices (identified by the `api.Heating` feature), the priority should be:
1. Meter SoC (if available)
2. Charger SoC (as fallback)

For vehicles and other non-heating devices, maintain the existing behavior.

Example Configuration:

```yaml
meters:
- name: elwa-2-power
  type: custom
  power:
    source: http
    uri: http://10.9.11.153/data.jsn
    jq: .power_elwa2
  soc:
    source: mqtt
    topic: "home/puffer"
    timeout: 5m
    jq: .temp_med
    scale: 0.1

loadpoints:
- title: Puffer
  charger: elwa-2  # AC-ELWA-2 heating device
  meter: elwa-2-power  # Meter with MQTT temperature reading
  mode: pv
```

In this case, the MQTT temperature reading (`soc:` from meter) should take precedence over any SoC calculation from the ELWA-2 charger.

Benefits:

1. Allow more accurate temperature monitoring for heating devices
2. Respects user configuration intent when they explicitly configure meter SoC
3. Maintains backward compatibility for vehicle charging use cases
4. Automatic detection using existing `api.Heating` feature classification

rhuss, 2025-06-12:
Thanks for considering this "override" possibility. I think it makes sense to allow this hook (i.e., allowing to overwrite some of the charger metrics) for some corner cases (like the AC ELWA-2 that has a very 'nervous' temperature measurement compared to other thermometers). You can always configure the meter only to give the measurements that you'd like to override.

On the other hand, I would also understand if this is too rare of a use case to add this extra overhead. (but maybe there are other use cases, too ?)

andig, 2025-06-15:
> I think it makes sense to allow this hook (i.e., allowing to overwrite some of the charger metrics) for some corner cases

As written in the PR. Then this measurement needs be part of the meter and must always be used if present. Not just for heating devices.

rhuss, 2025-06-16:
Exactly, this is what I meant: Having a new field `override:` in a _loadpoint_ that applied to any _charger_ (heating device or not). `override:` would take a reference to a _meter_ definition and all the measurements that this meter defines overrides the corresponding measurement of the _charger_ configured in the loadpont.

That would provide an easy way to customize a given charger definition for a single value only. However, as mentioned, it might still be overkill (although it's a pretty elegant solution for some corner cases), and it adds to the cognitive load of the configuration surface.

rhuss, 2025-06-16:
The reason why I would like an _additional_ field `override:` (in addition to `meter` which typically is also used for other purposes) is that it clearly states the **intention** what this definition is for. Having just a `meter:` just reflects the type of the field but not what the measurements are for. Meter is a general purpose type that can be used in different contexts, so already stating the intention as a config keyword has its benefits and make the config schema more self-describing and intuitive.

Also this additional field would be fully backwards compatible to the existing features (as it doesn't alter the current behaviour of `meter:`).

closed 2025-06-28 by github-actions[bot]