evcc-io/evcc #21796: GetMaxPower i.V.m. valid plugin gibt Warnung aus

VolkerK62, 2025-06-12:
aus GetMaxPower PR https://github.com/evcc-io/evcc/pull/16809#issue-2609192850
> If plugin fails, the configured value is acts as fallback.

Die Funktionalität ist gegeben. Wenn valid plugin "false" zurückgibt, wird der default Wert genommen.
Aber es wird dauerhaft eine Warnung ausgegeben 
`[circuit-main] WARN 2025/06/12 06:29:58 get max power: not available`

Könnte man diese Warnung unterdrücken?

VolkerK62, 2025-06-12:
Hmm ... gerade im Lastmanagement. Unter bestimmten Umständen soll der Wert reduziert werden.
Drauf gekommen bin auch aufgrund deines Kommentars im Doku PR  https://github.com/evcc-io/docs/pull/809#issuecomment-2921763724

Zu Testzwecken habe ich hier den Schalter der Batteriekontrolle zweckentfremdet
```
circuits:
- name: main
  title: load
  maxpower: 11000
  GetMaxPower:
    source: valid
    valid:
      source: http
      uri: http://localhost:7070/api/state
      jq: .result.batteryDischargeControl      
    value:
      source: http
      uri: http://localhost:7070/api/state
      jq: 4200 + .result.pvPower
```
Das Ganze geht natürlich auch ohne valid; mit "if .. then .. else".

andig, 2025-06-12:
Ich denke das ist "misuse" des valid Plugins. Mein Anwendungsfall war ioBroker: denn Meßwert gibts immer, wenn aber adapter.connected false ist gilt der nicht. Das ist ein harter Fehler, der kann also nicht ignoriert werden. Deshalb:
- hier falscher Anwendungsfall
- ich ändere den Fehler nochmal in `invalid` damit der nicht an anderer Stelle versehentlich ignoriert wird (bei `api.ErrNotAvailable` machen wir das sonst) und besser zum Namen des Plugins passt.

Ob ein Circuit Fehler loggen sollte oder sie ignorieren wäre nochmal eine ganz andere Diskussion.

VolkerK62, 2025-06-12:
Ok