evcc-io/evcc #21797: Tariff data not showing up

tehFred, 2025-06-12:
### Describe the bug

Continuing from this discussion: https://github.com/evcc-io/evcc/discussions/21795

I can run this trace command and see a price response from the amber API, but the application still doesn't present me with any current prices.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
tariffs:
  currency: AUD
  grid:
    type: template
    template: amber
    token: ~
    siteid: ~
    channel: E1
  feedin:
    type: template
    template: amber
    token: ~
    siteid: ~
    channel: B1
```

### Log details

```text
➜  ~ docker exec -it addon_49686a9f_evcc sh
/app # evcc -c /config/evcc.yaml -l trace tariff
[main  ] INFO 2025/06/12 14:40:23 evcc 0.204.2
[main  ] INFO 2025/06/12 14:40:23 using config file: /config/evcc.yaml
[db    ] INFO 2025/06/12 14:40:23 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2025/06/12 14:40:23 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/06/12 14:40:23 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/06/12 14:40:23 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/06/12 14:40:23 SELECT * FROM `settings` 0 <nil>
[db    ] TRACE 2025/06/12 14:40:24 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2025/06/12 14:40:24 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/06/12 14:40:24 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[amber ] TRACE 2025/06/12 14:40:24 GET https://api.amber.com.au/v1/sites/~~~~/prices/current?resolution=30&next=96
[amber ] TRACE 2025/06/12 14:40:24 GET https://api.amber.com.au/v1/sites/~~~~/prices/current?resolution=30&next=96
[amber ] TRACE 2025/06/12 14:40:24 [{"type":"CurrentInterval","date":"2025-06-12","duration":30,"startTime":"2025-06-12T05:00:01Z","endTime":"2025-06-12T05:30:00Z","nemTime":"2025-06-12T15:30:00+10:00","perKwh":35.18919,"renewables":80.84700000000001,"spotPerKwh":23.50519,"channelType":"general","spikeStatus":"none","tariffInformation":{"period":"solarSponge"},"descriptor":"veryLow","estimate":true,"advancedPrice":{"low":30.32698,"predicted":33.95233,"high":49.96211}},{"type":"ForecastInterval","date":"2025-06-12","duration":30,"startTime":"2025-06-12T05:30:01Z","endTime":"2025-06-12T06:00:00Z","nemTime":"2025-06-12T16:00:00+10:00","perKwh":54.34655,"renewables":75.058,"spotPerKwh":25.9179,"channelType":"general","spikeStatus":"none","tariffInformation":{"period":"peak"},"descriptor":"neutral","advancedPrice":{"low":43.74835,"predicted":82.08654,"high":243.91255}},{"type":"ForecastInterval","date":"2025-06-12","duration":30,"startTime":"2025-06-12T06:00:01Z","endTime":"2025-06-12T06:30:00Z","nemTime":"2025-06-12T16:30:00+10:00","perKwh":57.22857,"renewables":56.067959909287154,"spotPerKwh":28.5131,"channelType":"general","spikeStatus":"none","tariffInformation":{"period":"peak"},"descriptor":"neutral","advancedPrice":{"low":44.36608,"predicted":178.19234,"high":492.42791}},{"type":"ForecastInterval","date":"2025-06-12","duration":30,"startTime":"2025-06-12T06:30:01Z","endTime":"2025-06-12T07:00:00Z","nemTime":"2025-06-12T17:00:00+10:00","perKwh":1190.0506,"renewables":35.76974455953544,"spotPerKwh":1048.59607,"channelType":"general","spikeStatus":"spike","tariffInformation":{"period":"peak"},"descriptor":"spike","advancedPrice":{"low":50.01477,"predicted":288.42337,"high":560.48903}},{"type":"ForecastInterval","date":"2025-06-12","duration":30,"startTime":"2025-06-12T07:00:01Z","endTime":"2025-06-12T07:30:00Z","nemTime":"2025-06-12T17:30:00+10:00","perKwh":1484.18818,"renewables":24.460888985820386,"spotPerKwh":1313.46092,"channelType":"general","spikeStatus":"spike","tariffInformation":{"period":"peak"},"descriptor":"spike","advancedPrice":{"low":58.92024,"predicted":367.74771,"high":590.41793}},{"type":"ForecastInterval","date":"2025-06-12","duration":30,"startTime":"2025-06-12T07:30:01Z","endTime":"2025-06-12T08:00:00Z","nemTime":"2025-06-12T18:00:00+10:00","perKwh":1644.80004,"renewables":15.392662305763116,"spotPerKwh":1458.0886,"channelType":"general","spikeStatus":"spike","tariffInformation":{"period":"peak"},"descriptor":"spike","advancedPrice":{"low":60.43573,"predicted":382.78736,"high":582.18219}},{"type":"ForecastInterval","date":"2025-06-12","duration":30,"startTime":"2025-06-12T08:00:01Z","endTime":"2025-06-12T08:30:00Z","nemTime":"2025-06-12T18:30:00+10:00","perKwh":2114.50515,"renewables":13.047458158793892,"spotPerKwh":1881.0484,"channelType":"general","spikeStatus":"spike","tariffInformation":{"period":"peak"},"descriptor":"spike","advancedPrice":{"low":60.1149,"predicted":376.60365,"high":562.55851}},{"type":"ForecastInterval","date":"2025-06-12","duration":30,"startTime":"2025-06-12T08:30:01Z","endTime":"2025-06-12T09:00:00Z","nemTime":"2025-06-12T19:00:00+10:00","perKwh":1990.37701,"renewables":11.768709082699978,"spotPerKwh":1769.27356,"channelType":"general","spikeStatus":"spike","tariffInformation":{"period":"peak"},"descriptor":"spike","advancedPrice":{"low":60.75096,"predicted":388.22607,"high":575.31915}},{"type":"ForecastInterval","date":"2025-06-12","duration":30,"startTime":"2025-06-12T09:00:01Z","endTime":"2025-06-12T09:30:00Z","nemTime":"2025-06-12T19:30:00+10:00","perKwh":1798.97578,"renewables":14.78825264100937,"spotPerKwh":1596.92068,"channelType":"general","spikeStatus":"spike","tariffInformation":{"period":"peak"},"descriptor":"spike","advancedPrice":{"low":60.19661,"predicted":403.48751,"high":596.02647}},{"type":"ForecastInterval","date":"2025-06-12","duration":30,"startTime":"2025-06-12T09:30:01Z","endTime":"2025-06-12T10:00:00Z","nemTime":"2025-06-12T20:00:00+10:00","perKwh":1780.3731,"renewables":14.320465634628532,"spotPerKwh":1580.16935,"channelType":"general","spikeStatus":"spike","tariffInformation":{"period":"peak"},"descriptor":"spike","advancedPrice":{"low":82.90718,"predicted":402.21515,"high":598.66745}},{"type":"ForecastInterval","date":"2025-06-12","duration":30,"startTime":"2025-06-12T10:00:01Z","endTime":"2025-06-12T10:30:00Z","nemTime":"2025-06-12T20:30:00+10:00","perKwh":1868.56675,"renewables":9.512568889888652,"spotPerKwh":1659.58592,"channelType":"general","spikeStatus":"spike","tariffInformation":{"period":"peak"},"descriptor":"spike","advancedPrice":{"low":83.18606,"predicted":370.39714,"high":595.63512}},{"type":"ForecastInterval","date":"2025-06-12","duration":30,"startTime":"2025-06-12T10:30:01Z","endTime":"2025-06-12T11:00:00Z","nemTime":"2025-06-12T21:00:00+10:00","perKwh":2056.88633,"renewables":9.255208036612924,"spotPerKwh":1829.16384,"channelType":"general","spikeStatus":"spike","tariffInformation":{"period":"peak"},"descriptor":"spike","advancedPrice":{"low":63.65984,"predicted":306.37164,"high":563.04844}},{"type":"ForecastInterval","date":"2025-06-12","duration":30,"startTime":"2025-06-12T11:00:01Z","endTime":"2025-06-12T11:30:00Z","nemTime":"2025-06-12T21:30:00+10:00","perKwh":1938.48773,"renewables":14.068655837296584,"spotPerKwh":1722.54833,"channelType":"general","spikeStatus":"spike","tariffInformation":{"period":"peak"},"descriptor":"spike","advancedPrice":{"low":52.49742,"predicted":228.80801,"high":532.15904}},{"type":"ForecastInterval","date":"2025-06-12","duration":30,"startTime":"2025-06-12T11:30:01Z","endTime":"2025-06-12T12:00:00Z","nemTime":"2025-06-12T22:00:00+10:00","perKwh":114.7236,"renewables":14.999392011024867,"spotPerKwh":80.28619,"channelType":"general","spikeStatus":"none","tariffInformation":{"period":"peak"},"descriptor":"high","advancedPrice":{"low":46.02471,"predicted":139.90896,"high":436.78382}},{"type":"ForecastInterval","date":"2025-06-12","duration":30,"startTime":"2025-06-12T12:00:01Z","endTime":"2025-06-12T12:30:00Z","nemTime":"2025-06-12T22:30:00+10:00","perKwh":85.13093,"renewables":16.070600649091112,"spotPerKwh":53.6386,"channelType":"general","spikeStatus":"none","tariffInformation":{"period":"peak"},"descriptor":"high","advancedPrice":{"low":43.67487,"predicted":77.16928,"high":232.31268}},{"type":"ForecastInterval","date":"2025-06-12","duration":30,"startTime":"2025-06-12T12:30:01Z","endTime":"2025-06-12T13:00:00Z","nemTime":"2025-06-12T23:00:00+10:00","perKwh":64.72809,"renewables":17.194970671606185,"spotPerKwh":35.26626,"channelType":"general","spikeStatus":"none" [... cut ...]
From    To    Price/Cost (AUD/kWh)

solar:
From    To    Yield (W)

grid:
From    To    Price/Cost (AUD/kWh)

/app #
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.204.2

tehFred, 2025-06-12:
I've been running evcc with these tarrif settings unsuccessfully since before my account transitioned to the 5 minute billing - I've only just gotten around to making a ticket. So I don't believe it's due to the resolution change.

ross-w, 2025-06-12:
Hi @tehFred I've just seen your config:
```
tariffs:
  currency: AUD
  grid:
    type: template
    template: amber
    token: ~
    siteid: ~
    channel: E1
  feedin:
    type: template
    template: amber
    token: ~
    siteid: ~
    channel: B1
```

`channel` doesn't look right. Try setting these to `general` for grid and `feedIn` for feed-in:
```
tariffs:
  currency: AUD
  grid:
    type: template
    template: amber
    token: ~
    siteid: ~
    channel: general
  feedin:
    type: template
    template: amber
    token: ~
    siteid: ~
    channel: feedIn
```

I think evcc is just ignoring all the data coming back from the API because the channels don't match your config

tehFred, 2025-06-12:
Amazing, that was the issue. I set the channel to those values based on the channel identifier when I request sites from the API.

It might be worth updating the docs to have channnel examples listed, i.e.

> tariffs:
>   grid:
>     type: template
>     template: amber
>     token: # optional
>     siteid: # optional
>     channel: # optional, i.e. general, feedIn

tehFred, 2025-06-13:
Thank you for sharing yours, that has solved my issue. As per the [docs](https://docs.evcc.io/en/docs/tariffs#amber-electric), I had the following:

```
tariffs:
  grid:
    type: template
    template: amber
```

I altered my config to be `type: amber` and removed the template line, and now it works.

Thanks again for the help!

andig, 2025-06-13:
Both should work identically