evcc-io/evcc #21805: Meter Titel als Tag in die InfluxDB schreiben.

cschlipf, 2025-06-12:
Es werden jetzt Title für PV, Battery, Aux und Ext Meter vergeben. Es wäre super, wenn diese als 'title' Tag auch mit in die Influx geschrieben wird. Das würde die Identifikation deutlich erleichtern und es würde auch beim Nutzen der Daten neue Möglichkeiten eröffnen, die sonst nur sehr umständlich zu erreichen sind.

Siehe auch dieser Feature Request für die Grafana EVCC Dashboards: https://github.com/ha-puzzles/evcc-grafana-dashboards/issues/123

> Ist es eigentlich geplant die 'titles' auch in die Influx DB zu schreiben?
> 
> Für meine PV habe ich noch keine Titles festgelegt (Custom ModBus Meter kann man ja über das Config UI nicht anlegen), aber falls doch, würde dann hier ein Title-Tag stehen?
> 
> ```
> select * from pvPower order by time desc limit 10
> name: pvPower
> time                 id value
> ----                 -- -----
> 2025-04-26T09:35:45Z 2  630
> 2025-04-26T09:35:45Z    2910
> 2025-04-26T09:35:45Z 1  2280
> 2025-04-26T09:35:15Z    3040
> 2025-04-26T09:35:15Z 2  690
> 2025-04-26T09:35:15Z 1  2350
> 2025-04-26T09:34:45Z 2  740
> 2025-04-26T09:34:45Z 1  2400
> 2025-04-26T09:34:45Z    3140
> 2025-04-26T09:34:15Z 2  780
> ```
> 
> Ist natürlich redundant zur ID, würde es beim Auslesen der InfluxDB aber deutlich einfacher machen, wenn man den bereits festgelegten Title direkt mit auslesen könnte. 

 _Originally posted by @cschlipf in [#6781](https://github.com/evcc-io/evcc/issues/6781#issuecomment-2831994801)_

andig, 2025-06-12:
Gerne PR. Evtl. genügt es schon in der influxdb.go zu schauen ob das Struct bereits ein `Title` Feld hat.

andig, 2025-07-02:
Implementierungsvorschlag steht oben. Es muss halt jemand die Zeit finden, das zu tun…

MartinRinas, 2025-10-24:
wie geht evcc denn mit den IDs um, ich hätte jetzt erwartet dass die quasi immutable sind, also werden einmal vergeben und dann für neue Geräte stets die nächst höhere verwendet? Wenn man jetzt natürlich ein ext/aux Zähler definiert und den Zweck und damit den Titel ändert ist das schwer nachzuvollziehen. also z.b. ein Shelly zuerst für die Waschmaschine, dann der Trocker, und dann wieder etwas anderes verwendet und nur den Titel aktualisiert.

cschlipf, 2025-10-24:
So wie ich das verstehe ist ID=1 das erst element in der YAML, ID=2 das zweite element in der YAML, u.s.w.

Als es noch keine Titel gab, gab es ja sonst nichts, das ein Element eindeutig referenziert hat.

andig, 2026-05-01:
Perfekt, danke!