evcc-io/evcc #21818: PV-Prognose basierte Priorisierung

VB162, 2025-06-12:
Liebes Entwicklerteam,

ich würde mich über ein Feature freuen, über welches es möglich ist, entsprechend der PV-Prognose (Resterzeugung), Verbraucher zu priorisieren.

**Anwendung:**
Ich bin in der Regel nicht während der Tageszeiten mit einer TOP-PV-Ausbeute zu Hause.
Wenn ich am Nachmittag nach Hause komme, soll das Fahrzeug automatisch priorisiert werden, unabhängig davon, ob der Heimspeicher voll ist. Gleiches gilt auch für den frühen Morgen. 
**--> Bedingung**: Priorisierung des Fahrzeuges, sofern die PV-Ausbeute des restlichen Tages die fehlende Speicherkapazität des Heimspeichers überschreitet (+selbst gewählter Sicherheitsfaktor). 

**Vorteil:** Das Fahrzeug kann z.B. am Nachmittag noch mit der erforderlichen Mindestleistung geladen werden. Die hohe verfügbare PV-Leistung wird nicht in den Speicher gesteckt, der selber keine Anforderungen an die Mindestladeleitung hat. Zu späterer Stunde nimmt die PV-Ausbeute ab, kann aber den Speicher noch befüllen--> Besserer Eigenverbrauch, Auto wird optimierter geladen.

Viele Grüße

VB162, 2025-06-13:
Hallo andig,

Ich hoffe, dass ich es hier noch einmal besser erklären kann... sorry für die Länge.

**Der Wunsch wäre:** 
Solange es laut Prognose möglich ist, den Speicher bis Tagesende noch zu 100% zu beladen, soll der PV-Überschuss in das Fahrzeug gehen.
Ein zusätzliches Entladen des Speichers ist nicht notwendig - Wäre aber natürlich noch das i-Tüpfelchen.

Vielleicht veranschaulicht das **Beispiel** noch einmal: 
**Problem:**
- Fahrzeug ist zu den besten PV-Zeiten nicht da
- Heimspeicher ist immer da und benötigt kein Pmin
- Dennoch wird der Heimspeicher priorisiert geladen, obwohl über den Tag ausreichend Überschuss zur Verfügung steht, um Fahrzeug (zumindest teilweise) und Heimspeicher zu laden

**Ausgangslage - Beispiel früher Morgen**:
- Der Speicher (10 kWh) ist über Nacht auf 0% gefallen. 
- Prognose: 50 kWh Tages-Restertrag 
- Ab 6 Uhr morgens liegt die PV Erzeugung-Verbrauch über Pmin des Fahrzeuges
- Zwischen 8-10 Uhr verlasse ich das Haus mit dem Fahrzeug (d.h. 2-4h Lademöglichkeit vor Abfahrt)

**IST-Verhalten "PV-Modus":**
- PV-Überschuss geht in den Heimspeicher, bis die 10 kWh aufgefüllt sind
- Fahrzeug wird bis Abfahrt nicht geladen
- Kurz nach Abfahrt ist der Heimspeicher voll + Netzeinspeisung beginnt.
- Persönliche Abhilfe: Sofern ich es mitbekomme, stelle ich manuell auf "Min + PV", damit nicht der ganze Überschuss in den Heimspeicher geht.

**Wunsch-Verhalten:**
- EVCC "priorisiert" den Ladepunkt, 
           - EVCC schaltet die Ladeleistung der Wallbox hoch, sobald die Batterie geladen wird und wartet nicht, bis die Netzeinspeisung beginnt
           -  Z.B. setzt EVCC die Ladeleistung am Ladepunkt automatisch P(Ladepunkt) = P (Erzeugung) - P (Verbrauch), Sofern P(Ladepunkt) > PMin

- Fahrzeug wird bis Abfahrt geladen
- Heimspeicher wird nach Abfahrt aufgeladen

andig, 2025-06-13:
> Dennoch wird der Heimspeicher priorisiert geladen

Das wird er nur wenn du das so konfiguriert hast- siehe Doku!

VB162, 2025-06-13:
Ok, es geht aber darum dass die Vorhersage einbezogen wird… das kann ich
nicht in der Doku finden.