evcc-io/evcc #21847: Delta AC Max Basic: UI Error Icon - "Connect again!"

TweetsOfNiklas, 2025-06-15:
### Describe the bug

Since a few updates the AC Max Delta Wallboxes always show an error in the UI. They function perfect, but when the car is not charging it shows the error "Vorgang abgebrochen. Erneut verbinden.". This happens everytime when the car is not charging. 

Already tested with different cars. The error appears since a few releases. I am able to test this further. So if you have a clue I could try reverting back to a specific version.

### Steps to reproduce

1. Plug your car in.
2. Error message shows up when car is not charging.
3. When the car is charging the error icon disappears. 
4. After the charge the icon shows up again.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: XXX

sponsortoken: XXX

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

mqtt:
  broker: 192.168.178.XXX:1883
  user: XXX
  password: XXX
  topic: evcc

meters:  
   - name: my_grid
     type: template
     template: homewizard
     usage: grid
     host: 192.168.178.XXX
   - name: HM800Garten
     type: custom
     power:
      source: mqtt
      topic: solar/XXX/0/power

circuits:
- name: main
  maxCurrent: 35
  meter: my_grid

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.178.XXX 
  name: wb-goe
- type: template
  template: delta
  modbus: tcpip
  id: 1
  port: 502 # Port
  host: 192.168.178.XXX
  name: wb-delta-garage
[... 263 lines cut ...]
[lp-3 ] DEBUG 2025/06/15 10:07:39 charger status: A
[site ] DEBUG 2025/06/15 10:07:49 ----
[lp-2 ] DEBUG 2025/06/15 10:07:49 charge power: 0W
[lp-3 ] DEBUG 2025/06/15 10:07:49 charge power: 0W
[lp-1 ] DEBUG 2025/06/15 10:07:49 charge power: 0W
[lp-1 ] DEBUG 2025/06/15 10:07:49 charge currents: [0 0 0]A
[circuit-main] DEBUG 2025/06/15 10:07:49 power: 968W
[circuit-main] DEBUG 2025/06/15 10:07:49 current: 2.7A
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.2

TweetsOfNiklas, 2025-06-15:
So that everybody knows what I'm talking about:

[image]

andig, 2025-06-15:
@naltatis: was ist das? Im Log ist nichts zu sehen.

@TweetsOfNiklas was sagt /api/state?

StefanSchoof, 2025-06-15:
From the timestamp there is a site grid power in your log every 10s. In your yaml is no Intervall. What interval do you use?

TweetsOfNiklas, 2025-06-15:
> [@TweetsOfNiklas](https://github.com/TweetsOfNiklas) was sagt /api/state?

[state.json](https://github.com/user-attachments/files/20744717/state.json)

> From the timestamp there is a site grid power in your log every 10s. In your yaml is no Intervall. What interval do you use?

I have no interval set. I'm using the Homewizard implementation to get grid data.

naltatis, 2025-06-15:
Improving per-device error communication is a general topic that we're planning to work on. We also have other chargers or meters where, in normal operation warnings or errors are triggered but self-recover over time. Currently we simply log errors/warnings, send them to the ui and dont react if the issue has recovered. We are planning to introduce a per-device health status. This topic does not only affect chargers but also tariffs, meters, vehicles, ... Therefore I would not implement a so specific solution as you proposed in the first run. The first step in this direction is issue https://github.com/evcc-io/evcc/issues/14496

In this issue lets focus on if the specific "Vorgang abgebrochen. Erneut verbinden." error can be avoided.
\cc @premultiply

andig, 2025-06-15:
> Improving per-device error communication

@naltatis there's no error in the log?

naltatis, 2025-06-15:
Ah sorry, misread the issue. Did not realize, that the screenshot above is something that we actually do right now 😂.

This UI state is triggered when `chargerStatusReason` has the value `disconnectrequired`.

andig, 2025-06-15:
Comes from https://github.com/evcc-io/evcc/pull/15659

> Hintergrund ist, es gibt insbesondere bei OCPP einen Zustand "Finishing" bei dem keine Transaktion mehr existiert und man die Box nur noch zum erneuten Laden überreden kann wenn man das Fahrzeug absteckt (und ggf. danach wieder ansteckt). Denn nur beim Anstecken wird eine neue Transaktion erzeugt und somit überhaupt eine Ladung ermöglicht.

@TweetsOfNiklas can you restart charging without disconnecting the car again? Then the charger's status doesn't match the situation :(

TweetsOfNiklas, 2025-06-15:
> Comes from [#15659](https://github.com/evcc-io/evcc/pull/15659)
> 
> > Hintergrund ist, es gibt insbesondere bei OCPP einen Zustand "Finishing" bei dem keine Transaktion mehr existiert und man die Box nur noch zum erneuten Laden überreden kann wenn man das Fahrzeug absteckt (und ggf. danach wieder ansteckt). Denn nur beim Anstecken wird eine neue Transaktion erzeugt und somit überhaupt eine Ladung ermöglicht.
> 
> [@TweetsOfNiklas](https://github.com/TweetsOfNiklas) can you restart charging without disconnecting the car again? Then the charger's status doesn't match the situation :(
> 
> /cc [@premultiply](https://github.com/premultiply)

Yes. The wallbox works without any issues. Charging starts immediately when Mode is changed to "Fast". There is no reconnect needed. Also planned charging starts as expected. During charging the icon disappears. After charging icon shows up again.

The error icon definetely didn't show up a few versions back. If required I can do some more investigation.

andig, 2025-06-15:
> The error icon definetely didn't show up a few versions back. If required I can do some more investigation.

As I wrote- comes from https://github.com/evcc-io/evcc/pull/15659. If it started happening later than that for you it might be charger firmware. From our point of view and understanding of the OCPP spec that's a bug. If charger doesn't require disconnect it MUST NOT return that status.

That said- nothing left to do here. Would be interesting to get vendor feedback.

andig, 2025-06-15:
I see. A trace log might help as there seem to be two flavors in the code:

    evcc charger --log trace

once you're in that state.

TweetsOfNiklas, 2025-06-15:
[evcc-20250615-130705-trace.log](https://github.com/user-attachments/files/20745220/evcc-20250615-130705-trace.log)

Planned charge started at 13:00 o'clock. I have the AC MAX Basic with Serial Connection and PE11 Modbus Adapter.

andig, 2025-06-15:
https://github.com/evcc-io/evcc/issues/21847#issuecomment-2973651555

TweetsOfNiklas, 2025-06-15:
> [#21847 (comment)](https://github.com/evcc-io/evcc/issues/21847#issuecomment-2973651555)

Just to be clear: Before 13.00 the wallbox was in that state. I wasn't sure what you're looking for exactly so I included the start of the charge as well.

TweetsOfNiklas, 2025-06-21:
Okay, I now put it in a docker container and were able to run charger --log trace:
```

/app # evcc charger --log trace
[main  ] INFO 2025/06/21 15:41:37 evcc 0.204.3
[main  ] INFO 2025/06/21 15:41:37 using config file: /etc/evcc.yaml
[db    ] INFO 2025/06/21 15:41:37 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2025/06/21 15:41:37 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/06/21 15:41:37 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/06/21 15:41:37 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/06/21 15:41:37 SELECT * FROM `settings` 6 <nil>
[mqtt  ] INFO 2025/06/21 15:41:37 connecting evcc-1537047792 at tcp://192.168.178.200:1883
[mqtt  ] DEBUG 2025/06/21 15:41:37 tcp://192.168.178.200:1883 connected
[db    ] TRACE 2025/06/21 15:41:37 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2025/06/21 15:41:37 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/06/21 15:41:37 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[go-e  ] TRACE 2025/06/21 15:41:37 GET http://192.168.178.207/api/status?filter=alw
[db    ] TRACE 2025/06/21 15:41:37 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
[delta ] TRACE 2025/06/21 15:41:37 modbus: send 00 01 00 00 00 06 01 04 03 e9 00 01
[delta ] TRACE 2025/06/21 15:41:37 modbus: send 00 01 00 00 00 06 01 04 03 e9 00 01
[go-e  ] TRACE 2025/06/21 15:41:37 {"alw":false}
[delta ] TRACE 2025/06/21 15:41:37 modbus: recv 00 01 00 00 00 05 01 04 02 00 00
[delta ] TRACE 2025/06/21 15:41:37 modbus: send 00 02 00 00 00 06 01 03 00 c9 00 01
[delta ] TRACE 2025/06/21 15:41:37 modbus: recv 00 01 00 00 00 05 01 04 02 00 07
[delta ] TRACE 2025/06/21 15:41:37 modbus: send 00 02 00 00 00 06 01 03 00 c9 00 01
[delta ] TRACE 2025/06/21 15:41:37 modbus: recv 00 02 00 00 00 05 01 03 02 00 00
[delta ] TRACE 2025/06/21 15:41:37 modbus: recv 00 02 00 00 00 05 01 03 02 00 00
wb-goe
------
[go-e  ] TRACE 2025/06/21 15:41:37 GET http://192.168.178.207/api/status?filter=alw,car,eto,nrg,wh,trx,cards
[go-e  ] TRACE 2025/06/21 15:41:37 {"alw":false,"eto":7862736,"car":1,"wh":77.75906,"trx":null,"nrg":[236,235,235,0,0,0,0,0,0,0,0,0,0,0,0,0],"cards":[{"name":"n/a","energy":0,"cardId":true},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false},{"name":"n/a","energy":0,"cardId":false}]}
Power:          0W
Energy:         7862.7kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 236V 235V 235V
Charge status:  A
Enabled:        false
Identifier:     <none>

wb-delta-carport
----------------
[delta ] TRACE 2025/06/21 15:41:37 modbus: send 00 03 00 00 00 06 01 04 03 ed 00 02
[delta ] TRACE 2025/06/21 15:41:37 modbus: recv 00 03 00 00 00 07 01 04 04 00 00 00 00
[delta ] TRACE 2025/06/21 15:41:37 modbus: send 00 04 00 00 00 06 01 04 03 e9 00 01
[delta ] TRACE 2025/06/21 15:41:37 modbus: recv 00 04 00 00 00 05 01 04 02 00 00
[delta ] TRACE 2025/06/21 15:41:37 modbus: send 00 05 00 00 00 06 01 04 03 e9 00 01
[delta ] TRACE 2025/06/21 15:41:37 modbus: recv 00 05 00 00 00 05 01 04 02 00 00
[delta ] TRACE 2025/06/21 15:41:37 modbus: send 00 06 00 00 00 06 01 03 06 40 00 02
[delta ] TRACE 2025/06/21 15:41:37 modbus: recv 00 06 00 00 00 07 01 03 04 00 00 00 00
[delta ] TRACE 2025/06/21 15:41:37 modbus: send 00 07 00 00 00 06 01 04 04 4c 00 14
[delta ] TRACE 2025/06/21 15:41:38 modbus: recv 00 07 00 00 00 2b 01 04 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Power:         0W
Charge status: A
Status reason: unknown
Enabled:       false
Identifier:    <none>

wb-delta-garage
---------------
[delta ] TRACE 2025/06/21 15:41:38 modbus: send 00 03 00 00 00 06 01 04 03 ed 00 02
[delta ] TRACE 2025/06/21 15:41:38 modbus: recv 00 03 00 00 00 07 01 04 04 00 00 00 00
[delta ] TRACE 2025/06/21 15:41:38 modbus: send 00 04 00 00 00 06 01 04 03 e9 00 01
[delta ] TRACE 2025/06/21 15:41:38 modbus: recv 00 04 00 00 00 05 01 04 02 00 07
[delta ] TRACE 2025/06/21 15:41:38 modbus: send 00 05 00 00 00 06 01 04 03 e9 00 01
[delta ] TRACE 2025/06/21 15:41:38 modbus: recv 00 05 00 00 00 05 01 04 02 00 07
[delta ] TRACE 2025/06/21 15:41:38 modbus: send 00 06 00 00 00 06 01 03 06 40 00 02
[delta ] TRACE 2025/06/21 15:41:38 modbus: recv 00 06 00 00 00 07 01 03 04 00 00 00 00
[delta ] TRACE 2025/06/21 15:41:38 modbus: send 00 07 00 00 00 06 01 04 04 4c 00 14
[delta ] TRACE 2025/06/21 15:41:38 modbus: recv 00 07 00 00 00 2b 01 04 28 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Power:         0W
Charge status: B
Status reason: disconnectrequired
Enabled:       false
Identifier:    <none>
```

I included all 3 charging stations. wb-delta-garage has a car connected and is showing "disconnectrequired".

closed 2025-06-23 by andig