evcc-io/evcc #21851: Authorize Zaptec (Pro) charger setup to avoid having to use an RFID tag or the app authorization

ttomasini, 2025-06-15:
**Is your feature request related to a problem? Please describe.**
Unfortunately, I cannot operate the charging station in free mode because the RFID/app authorization is hard-set by my load management operator. For this reason, every time I plug my vehicle into the charging station, I am forced to manually unlock the charging station using an RFID tag or the app. EVCC should do this automatically if a corresponding flag is set in the charging station yaml config (e.g. `authorize: true`).

**Describe the solution you'd like**
See https://github.com/evcc-io/evcc/issues/21745#issuecomment-2957238660 for more information as well as a working REST call.

**Describe alternatives you've considered**
The only alternative so far is to manually unlock/authorize the Zaptec charger every time I plug in vehicle.

andig, 2025-06-15:
Took a look. We have an interface for that:

```go
var _ api.Authorizer = (*OpenWB)(nil)

// Authorize implements the api.Authorizer interface
func (m *OpenWB) Authorize(key string) error {
	return m.authS(key)
}
```

...but apparently it's not being called anywhere. I couldn't find docs of `/api/chargers/{chargerId}/authorizecharge` either.

ttomasini, 2025-06-15:
> Took a look. We have an interface for that:
Yes, there has been only a implementation for openwb so far and it looks like there is an alternative implementation for easee chargers but this one is not utilizing the `api.Authorizer` interface.

> ...but apparently it's not being called anywhere. I couldn't find docs of `/api/chargers/{chargerId}/authorizecharge` either.

Yes, unfortunately it is not officially documented by Zaptec for some reasons. But it looks like it is stable (since it is used by the app) and it is pretty easy to use:

```bash
curl --request POST \
  --url https://api.zaptec.com/api/chargers/{chargerId}/authorizecharge \
  --header 'accept: */*' \
  --header 'Authorization: Bearer ey[...]'
```

premultiply, 2025-07-07:
As this requirement is very specific from my point of view, I do not want to do this.
evcc always works in such a way that a charging session is initiated and, if necessary, authenticated from the wallbox. This mode is often called autostart or free loading. The final charging authorisation is given via evcc.

Even if I can understand that the behaviour of the wallbox configuration, which in this case is unchangeable for the user, is somewhat annoying, it still corresponds exactly to the desired and expected behaviour from the point of view of the control system.

closed 2025-07-07 by premultiply