evcc-io/evcc #21902: Error with Daheimladen Wallbox

Alanin, 2025-06-18:
### Describe the bug

Hi,

after updating to latest version I have warning in the logs. the current is not calculated correctly. and therefore the limit is set to to high value consuming to much energy.

the problem does NOT exist in nightly build yet because the build is older than the latest release.

if testing is needed I can help

### Steps to reproduce

1. update to latest release
2. plug in car
3. watch the logs 

### Configuration details

```yaml
i am using the database configuration not the yaml one
```

### Log details

```text
[site ] DEBUG 2025/06/18 13:55:24 ----
[lp-1 ] DEBUG 2025/06/18 13:55:24 charge power: 6800W
[lp-1 ] DEBUG 2025/06/18 13:55:24 charge currents: [6.09e+05 6.23e+05 6.36e+05]A
[site ] DEBUG 2025/06/18 13:55:24 grid power: 2517W
[site ] DEBUG 2025/06/18 13:55:24 grid powers: [860 810 848]W
[site ] DEBUG 2025/06/18 13:55:24 grid currents: [3.95 3.74 3.77]A
[site ] DEBUG 2025/06/18 13:55:25 pv 1 power: 4878W
[site ] DEBUG 2025/06/18 13:55:25 site power: 2517W
[lp-1 ] DEBUG 2025/06/18 13:55:25 charge voltages: [242 242 237]V
[lp-1 ] DEBUG 2025/06/18 13:55:25 detected connected phases: 3p
[lp-1 ] DEBUG 2025/06/18 13:55:25 detected active phases: 3p
[lp-1 ] DEBUG 2025/06/18 13:55:25 charge total import: 1919.900kWh
[lp-1 ] DEBUG 2025/06/18 13:55:25 charger status: C
[lp-1 ] DEBUG 2025/06/18 13:55:25 soc estimated: 88.58% (vehicle: 88.00%)
[lp-1 ] DEBUG 2025/06/18 13:55:25 vehicle soc: 89%
[lp-1 ] DEBUG 2025/06/18 13:55:25 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/06/18 13:55:25 vehicle range: 437km
[lp-1 ] WARN 2025/06/18 13:55:25 charger logic error: current mismatch (got 6.36e+05A measured, expected 10A) - make sure your interval is at least 30s
[lp-1 ] DEBUG 2025/06/18 13:55:25 pv charge current: 6.36e+05A = 6.36e+05A + -3.65A (2517W @ 3p)
[lp-1 ] DEBUG 2025/06/18 13:55:25 set charge current limit: 10A
[site ] DEBUG 2025/06/18 13:55:54 ----
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.3

andig, 2025-06-18:
That's odd. What box do you have and could you please share

    evcc charger --log trace

for last and current version?

LutzDi, 2025-06-18:
kann es damit >https://github.com/evcc-io/evcc/commit/7a5474efe985e1c989a02631b863df2b487c6aa0 zusammenhängen?

andig, 2025-06-18:
> P.S.: ich bin mit der HA-Integration unterwegs... jfyi.

Ich kann zu HA nichts sagen. Erstellung von Logfiles ist dokumentiert. Ursache _dieses_ Fehlers hoffentlich gefunden.

LutzDi, 2025-06-18:
> Ich kann zu HA nichts sagen. Erstellung von Logfiles ist dokumentiert. Ursache dieses Fehlers hoffentlich gefunden.

Da das schon öfters Diskussionspunkt war, darf ich um einen Link zur Doku bitten? Ich habe viel gesucht und natürlich gefunden, wie ich den Loglevel einstelle, aber mir fehlt bis heute die Info, wo ich die Logs als Datei extrahieren kann. Natürlich habe ich den Webview auf das Log, aber irgendwo müssen die Daten ja auch liegen, oder?

Maschga, 2025-06-18:
Ich denke, dass das hier gemeint ist:
[wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen-?](https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen)

Alanin, 2025-06-18:
wow, vielen Dank für den schnellen fix