evcc-io/evcc #21917: use of Circuit to limit maxPower prevent startup

bekovan, 2025-06-18:
### Describe the bug

#circuits:
  - name: main # eindeutiger Name, wird als Referenz für andere Circuits verwendet
    title: main circuit # Anzeige in der UI (später)
#    maxCurrent: 63 # 63A (optional)
    maxPower: 2400 # 30kW (optional)
    meter: grid1 # optional
    parent: # no parent, this is the root circuit
stopped the boot of EVCC generating in error message
$FATAL decoding failed due to the following error(s): '' has invalid keys: circuit

### Steps to reproduce

1. no change in evcc.yaml
2.upgrade to 0.204.3
3.start EVCC
4. not starting
5. comment out all references to circuit
6. EVCC will start but without power limitation
...

### Configuration details

```yaml
#circuits:
#  - name: main # eindeutiger Name, wird als Referenz für andere Circuits verwendet
#    title: main circuit # Anzeige in der UI (später)
#    maxCurrent: 63 # 63A (optional)
#    maxPower: 2400 # 30kW (optional)
#    meter: grid1 # optional
#    parent: # no parent, this is the root circuit

was till now working
```

### Log details

```text
$FATAL decoding failed due to the following error(s): '' has invalid keys: circuit
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.3

bekovan, 2025-06-18:
ok, how to limit the max power from the net, I can not find this in the doc

mfuchs1984, 2025-06-18:
https://docs.evcc.io/en/docs/features/loadmanagement