evcc-io/evcc #21919: After update to 0.204.3 evcc is not starting anymore

micha444, 2025-06-18:
### Describe the bug

I made no changes at EVCC, just did the udpate. 

**error message:**

starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/06/18 22:38:27 evcc 0.204.3
[main  ] INFO 2025/06/18 22:38:27 using config file: /config/evcc.yaml
[mqtt  ] INFO 2025/06/18 22:38:27 connecting evcc-609741000 at tcp://localhost:1883
[main  ] INFO 2025/06/18 22:38:27 listening at :7070
[main  ] FATAL 2025/06/18 22:38:29 decoding failed due to the following error(s):
'' has invalid keys: maxgridsupplywhilebatterycharging
[main  ] FATAL 2025/06/18 22:38:29 will attempt restart in: 15m0s
Using config file: /config/evcc.yaml

As said, there were no changes the config.

Any ideas? :-)

Thanks for your support.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 192.168.178.254
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# trial token, valid until 2025-01-02
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoibWljaGE0NDQiLCJleHAiOjMwMzE5MDU2MDAsImlhdCI6MTczNTkwNTYwMCwic3JjIjoiZ2gifQ.OpcASAKyBR1SzUIgspjbNLDA7C6vvfAUBzTJQswMvCw

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
[... 260 lines cut ...]
[main  ] INFO 2025/06/18 22:38:27 evcc 0.204.3
[main  ] INFO 2025/06/18 22:38:27 using config file: /config/evcc.yaml
[mqtt  ] INFO 2025/06/18 22:38:27 connecting evcc-609741000 at tcp://localhost:1883
[main  ] INFO 2025/06/18 22:38:27 listening at :7070
[main  ] FATAL 2025/06/18 22:38:29 decoding failed due to the following error(s):
'' has invalid keys: maxgridsupplywhilebatterycharging
[main  ] FATAL 2025/06/18 22:38:29 will attempt restart in: 15m0s
Using config file: /config/evcc.yaml
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.3

andig, 2025-06-18:
Die Fehlermeldung sagt alles. Siehe Releasenotes.

micha444, 2025-06-18:
Ich bin leider Programmierer, :-( könntest du mir bitte kurz sagen was ich genau ändern muss?

Wäre sehr nett!

Lg

andig, 2025-06-18:
> Ich bin leider Programmierer

Wie hast Du denn evcc konfiguriert? Den Wert gibts nicht mehr und der muss raus, das hat mit Programmierung nix zu tun.

MichiGuti, 2025-06-19:
Das funktioniert leider bei mir nicht. Die Fehlermeldung bleibt bestehen - auch nach einem kompletten Homeassistant Neustart

andig, 2025-06-20:
Dann ist es noch in deiner Config.

andig, 2025-06-20:
Siehe https://github.com/evcc-io/evcc/issues/21925#issuecomment-2989691673