evcc-io/evcc #21920: Oauth refresh token expiring, no way to reauth

aguilaair, 2025-06-18:
### Describe the bug

I have the Volvo API integration and it had been wokring well until recently. 

I am now getting: `cannot create vehicle type 'template': cannot create vehicle type 'volvo-connected': cannot get vehicles: Get "https://api.volvocars.com/connected-vehicle/v2/vehicles": oauth2: "invalid_grant" "unknown, invalid, or expired refresh token"`

As far as I can tell there is no way to re-autheticate and get a new link. This should be detected and a new link should be printed to the log

### Steps to reproduce

1. Use oAuth to autheticate a vehicle 
2. Wait for the refresh_token to expire

### Configuration details

```yaml
N/A - UI configured
```

### Log details

```text
cannot create vehicle type 'template': cannot create vehicle type 'volvo-connected': cannot get vehicles: Get "https://api.volvocars.com/connected-vehicle/v2/vehicles": oauth2: "invalid_grant" "unknown, invalid, or expired refresh token"

TRACE LOGS:

httpd ] TRACE 2025/06/18 23:08:28 GET /api/config/devices/charger/Commander/status
[httpd ] TRACE 2025/06/18 23:08:31 POST /api/config/test/vehicle
[volvo-connected] TRACE 2025/06/18 23:08:31 POST https://volvoid.eu.volvocars.com/as/token.oauth2
[volvo-connected] TRACE 2025/06/18 23:08:31 grant_type=refresh_token&refresh_token=cQ6zGMH7zh.....QDTz -- {"error_description":"unknown, invalid, or expired refresh token","error":"invalid_grant"}
[db ] TRACE 2025/06/18 23:08:31 DELETE FROM `settings` WHERE `settings`.`key` = "oauth.dc-brhrch9....wp2cv" 1 <nil>
[httpd ] TRACE 2025/06/18 23:08:33 GET /api/system/log
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-06-18:
/cc @lehmanju

aguilaair, 2025-06-19:
where can I find it?

lehmanju, 2025-06-19:
Restart the instance and it should show up in log

andig, 2025-06-20:
@lehmanju does the RT also refresh and so we already store ist if updated? Or does the clientid expire here?

andig, 2025-06-21:
I've reached out to Volvo:

>we’ve meanwhile hit a real roadblock: https://github.com/evcc-io/evcc/issues/21920#issuecomment-2992215598
According to user feedback, the refresh token apparently expires (although being refreshed itself) after 7 days. At this stage, a fresh login is apparently required.
Could you please check/ confirm if that’s the expected behaviour?
We’ve also noticed that we can _only_ connect to localhost but for example not evcc.local or IP adress. Is that intended? For user with Homeassisstant and other setups not running on localhost it would be really helpful if these redirect urls could be allowed.

andig, 2025-06-21:
@lehmanju for token expiry, the UI-based login could make the process easier?

lehmanju, 2025-06-21:
Yes, the process would be easier with UI. However, since I have no experience with designing good UI and my time is currently quite scarce, the work on #21266 is stalled.

If there is somebody who is willing to implement the UI part (Vue), I can add the backend functionality.

andig, 2025-06-21:
> I have no experience with designing good U

@lehmanju the UI part is already there. The hamburger menu will list any device (not only vehicles) waiting for authentication. All that's missing is the necessary wiring to the device.

andig, 2025-06-24:
For time being (and I can think about it) it should be possible to add them to yaml if not ui. After restart, hamburger menu should work for login, wdyt?

MartijnvanGeffen13, 2025-06-25:
I have also update the modules https://github.com/[MartijnvanGeffen13/Volvo4evcc](https://github.com/MartijnvanGeffen13/Volvo4evcc) to use the new volvo OAUTH client flow and API V2 , 

I noticed that token lifetime has also drastically dropped from 30 min to 5 min,  and they added PKCE required now , and HTTP is no longer allowed on callback must be HTTPS  (This is now blocking your installation instructions ), Note that V1 API will be removed in next few months not sure if you allready on V2 seen your documentation suggests V1

andig, 2025-06-27:
@lehmanju should this be solved, too?

lehmanju, 2025-06-27:
Yes, this is basically a duplicate of the other Volvo issue anyway.

As for the API and OAuth stuff, that's already adapted for the latest version.

closed 2025-06-27 by andig