evcc-io/evcc #21925: 502 Bad Gateway seit letztem Update, maxgridsupplywhilebatterycharging

IloG78, 2025-06-19:
### Describe the bug

Hallo, ich nutze Evcc als add on und Integration in Home Assistant. Beides funktioniert seit dem gestrigen Update auf  204.3 nicht mehr. Laut Log wird folgendes Problem ausgegeben, an der evcc.config wurden zuvor keine Änderungen vorgenommen. Zuvor wurde Home Assistant core ebenfalls aktualisiert aber alle anderen Anwendungen funktionieren. Danke für eine Hilfe!

### Steps to reproduce

1. Evcc startet nicht 
2. Fehler 502 Bad Gateway erscheint 
3.
...

### Configuration details

```yaml
evcc

[main ] FATAL 2025/06/19

06:25:58 will attempt restart

in: 15m0s

Using config file:

/config/evcc.yaml

starting evcc:

'EVCC_DATABASE_DSN=/data/evcc

.db evcc --config

/config/evcc.yaml'

[main ] INFO 2025/06/19

06:29:03 evcc 0.204.3

[main ] INFO 2025/06/19

06:29:03 using config file:

/config/evcc.yaml

[main ] INFO 2025/06/19 06:29:04 listening at :7070

[main ] FATAL 2025/06/19

06:29:06 decoding failed due to the following error(s): has invalid keys:

maxgridsupplywhilebatterychar ging

[main ] FATAL 2025/06/19 06:29:06 will attempt restart

in: 15m0s

Live
```

### Log details

```text
evcc

[main ] FATAL 2025/06/19

06:25:58 will attempt restart

in: 15m0s

Using config file:

/config/evcc.yaml

starting evcc:

'EVCC_DATABASE_DSN=/data/evcc

.db evcc --config

/config/evcc.yaml'

[main ] INFO 2025/06/19

06:29:03 evcc 0.204.3

[main ] INFO 2025/06/19

06:29:03 using config file:

/config/evcc.yaml

[main ] INFO 2025/06/19 06:29:04 listening at :7070

[main ] FATAL 2025/06/19

06:29:06 decoding failed due to the following error(s): has invalid keys:

maxgridsupplywhilebatterychar ging

[main ] FATAL 2025/06/19 06:29:06 will attempt restart

in: 15m0s

Live

Informatio...

Dokument...

Konfigurati...

Protokoll
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

mfuchs1984, 2025-06-19:
Siehe Release Notes, maxgridsupplywhilebatterycharging ist schon sehr lange deprecated und würde jetzt entfernt.

mfuchs1984, 2025-06-19:
Ja.

closed 2025-06-19 by andig

MichiGuti, 2025-06-19:
Hallo zusammen,

ich habe das selbe Problem ebenfalls seit der Version 204.3.
Ich habe das maxgridsupplywhilebatterycharging bereits auskommentiert, komplett gelöscht oder durch den scheinbaren Nachfolger "maxACPower" ersetzt  alles ohne Erfolg. Ich bekomme jedes mal die Fehlermeldung: 
FATAL 2025/06/19 23:08:56 decoding failed due to the following error(s):
'' has invalid keys: maxgridsupplywhilebatterycharging.

Muss ich neben der config/evcc.yaml auch den Programmcode anpassen?

mfuchs1984, 2025-06-20:
Die Konfiguration liegt in Home Assistant in /addon_configs/49686a9f_evcc, siehe https://docs.evcc.io/docs/installation/home-assistant

Wer das File Editor Addon zum editieren benutzt, muss erst die Option "Enforce Basepath" abschalten, um in den Ordner zu kommen.

MichiGuti, 2025-06-20:
> Die Konfiguration liegt in Home Assistant in /addon_configs/49686a9f_evcc, siehe https://docs.evcc.io/docs/installation/home-assistant
> 
> Wer das File Editor Addon zum editieren benutzt, muss erst die Option "Enforce Basepath" abschalten, um in den Ordner zu kommen.

Vielen Dank!!
Das hat geholfen. Ich hatte ewig an der falschen yaml rum konfiguriert.

mfuchs1984, 2025-06-21:
@bennyrbg VSC meint Studio Code Server? Dann links das Menü mit den drei Strichen öffnen, File, Open Folder.