evcc-io/evcc #21935: EVCC doesn't start the charging when having to use 1P?

jaytonic, 2025-06-19:
### Describe the bug

Hi,

Currently I've only an excess of 2.6kw(evening), that implies using 1P charging.
My charger (Easee Home) does support it.

Currently, I plug the car(Audi q6), it briefly starts to charge at its maximum power(11kw), then it stay "charged interrupted". I check the easee home app, I see the charging session has been "paused", and on EVCC, I just see "Waiting for vehicle".

If I manually "unpause" the charge, it charge for 10-20secs at 11kw again, then get stopped as soon as EVCC realize it is charging(way much than its supposed to).

If I stop EVCC, the vehicule properly charges at the proper level.

I did see on the Easee charger app, that I cannot set less than 6A when being in 3P.
I'm not sure what should I set the 1P/3P on my charger? I can either set it to automatic(but then not controlled by EVCC, locked on 3Phase, locked on 1phase). For my tests it was "locked on 3 phases"

### Steps to reproduce

1. Have not enough solar for 3P charging
2. Have the Easee home charger set to "locked on 3 phase"
3. Plug the car

### Configuration details

```yaml
meters:
  - name: my_grid
    type: template
    template: enphase
    usage: grid
    host: 192.168.0.196 # IP address or hostname
    token: abc

  - name: my_pv
    type: template
    template: enphase
    usage: pv
    host: 192.168.0.196 # IP address or hostname
    token: abc

tariffs:
  currency: CHF
  grid:
    type: fixed
    price: 0.2917 # EUR/kWh (default)
    zones:
      - days: Mo-So
        hours: 21-0
        price: 0.1881 # EUR/kWh (weekdays 2-6 hours)
      - days: Mo-So
        hours: 0-7
        price: 0.1881 # EUR/kWh (weekdays 2-6 hours)
chargers:
  - name: easee_home
    type: template
    template: easee
    user: my@email.com
    password: AABBCC
    charger: AABBCC
    timeout: 30s # optional
    authorize: 

vehicles:
    - name: audi_q6
      type: template
      template: audi
      user: me@email.com
      password: AAABBBB
      title: Audi Q6
      capacity: 100

site:
  title: Home
  meters:
    grid: my_grid
    pv: my_pv

loadpoints:
  - title: Carport
    charger: easee_home
    mode: pv
    vehicle: audi_q6

sponsortoken: token
interval: 60s
```

### Log details

```text
[main ] INFO 2025/06/19 18:38:56 evcc 0.204.3
[main ] INFO 2025/06/19 18:38:56 using config file: /root/.evcc/evcc.yaml
[db ] INFO 2025/06/19 18:38:56 using sqlite database: /root/.evcc/evcc.db
[main ] INFO 2025/06/19 18:38:56 listening at :7070
[lp-1 ] DEBUG 2025/06/19 18:38:59 charge total import: 5.592kWh
[site ] INFO 2025/06/19 18:38:59 site config:
[site ] INFO 2025/06/19 18:38:59 meters: grid ✓ pv ✓ battery ✗
[site ] INFO 2025/06/19 18:38:59 grid: power ✓ energy ✗ currents ✓
[site ] INFO 2025/06/19 18:38:59 pv 1: power ✓ energy ✓ currents ✓
[site ] INFO 2025/06/19 18:38:59 vehicles:
[site ] INFO 2025/06/19 18:38:59 vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site ] INFO 2025/06/19 18:38:59 tariffs:
[site ] INFO 2025/06/19 18:38:59 grid: priceforecast
[site ] INFO 2025/06/19 18:38:59 feed-in: ✗
[site ] INFO 2025/06/19 18:38:59 co2: ✗
[site ] INFO 2025/06/19 18:38:59 solar: ✓
[lp-1 ] INFO 2025/06/19 18:38:59 loadpoint 1:
[lp-1 ] INFO 2025/06/19 18:38:59 mode: pv
[lp-1 ] INFO 2025/06/19 18:38:59 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1 ] INFO 2025/06/19 18:38:59 meters: charge ✓
[lp-1 ] INFO 2025/06/19 18:38:59 charge: power ✓ energy ✓ currents ✓
[lp-1 ] DEBUG 2025/06/19 18:38:59 set smart cost limit: 0.2
[lp-1 ] DEBUG 2025/06/19 18:38:59 phase timer inactive
[lp-1 ] DEBUG 2025/06/19 18:38:59 pv timer inactive
[lp-1 ] INFO 2025/06/19 18:38:59 vehicle updated: unknown -> Audi Q6
[site ] DEBUG 2025/06/19 18:38:59 ----
[lp-1 ] DEBUG 2025/06/19 18:38:59 charge power: 4220W
[lp-1 ] DEBUG 2025/06/19 18:38:59 charge currents: [5.92 5.93 5.84]A
[site ] DEBUG 2025/06/19 18:39:00 pv 1 power: 2924W
[site ] DEBUG 2025/06/19 18:39:01 grid power: 1775W
[site ] DEBUG 2025/06/19 18:39:05 grid currents: [2.64 2.51 3.29]A
[site ] DEBUG 2025/06/19 18:39:05 site power: 1775W
[lp-1 ] DEBUG 2025/06/19 18:39:05 vehicle odometer: 158km
[lp-1 ] DEBUG 2025/06/19 18:39:05 charge total import: 5.592kWh
[lp-1 ] DEBUG 2025/06/19 18:39:05 charger status: C
[lp-1 ] INFO 2025/06/19 18:39:05 car connected
[lp-1 ] DEBUG 2025/06/19 18:39:05 pv timer elapse
[lp-1 ] DEBUG 2025/06/19 18:39:05 pv timer inactive
[lp-1 ] DEBUG 2025/06/19 18:39:05 charge total import: 5.592kWh
[lp-1 ] INFO 2025/06/19 18:39:05 start charging ->
[lp-1 ] DEBUG 2025/06/19 18:39:05 wake-up timer: stop
[lp-1 ] DEBUG 2025/06/19 18:39:05 vehicle status: C
[lp-1 ] DEBUG 2025/06/19 18:39:05 vehicle soc: 73%
[lp-1 ] DEBUG 2025/06/19 18:39:05 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/06/19 18:39:05 vehicle range: 396km
[lp-1 ] WARN 2025/06/19 18:39:05 charger logic error: disabled but charging
[lp-1 ] DEBUG 2025/06/19 18:39:07 charger enable
[lp-1 ] DEBUG 2025/06/19 18:39:07 pv charge current: 0A = 0A + -2.57A (1775W @ 3p)
[lp-1 ] DEBUG 2025/06/19 18:39:07 projected site power 1775W >= 200W disable threshold
[lp-1 ] DEBUG 2025/06/19 18:39:07 pv disable in 0s
[lp-1 ] DEBUG 2025/06/19 18:39:07 pv disable timer elapsed
[lp-1 ] DEBUG 2025/06/19 18:39:07 pv timer reset
[lp-1 ] DEBUG 2025/06/19 18:39:07 pv timer inactive
[lp-1 ] DEBUG 2025/06/19 18:39:10 charger disable
[lp-1 ] DEBUG 2025/06/19 18:39:10 wake-up timer: stop
[site ] DEBUG 2025/06/19 18:39:10 ----
[lp-1 ] DEBUG 2025/06/19 18:39:10 charge power: 0W
[lp-1 ] DEBUG 2025/06/19 18:39:10 charge currents: [0.548 0.184 0.184]A
[site ] DEBUG 2025/06/19 18:39:10 grid power: -190W
[site ] DEBUG 2025/06/19 18:39:11 pv 1 power: 2918W
[site ] DEBUG 2025/06/19 18:39:16 grid currents: [-3.67 -3.65 -2.98]A
[site ] DEBUG 2025/06/19 18:39:16 site power: -190W
[lp-1 ] DEBUG 2025/06/19 18:39:16 vehicle odometer: 158km
[lp-1 ] DEBUG 2025/06/19 18:39:16 charge total import: 5.592kWh
[lp-1 ] DEBUG 2025/06/19 18:39:16 charger status: B
[lp-1 ] INFO 2025/06/19 18:39:16 stop charging <-
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.3

mfuchs1984, 2025-06-19:
Tried using `authorize: true`? Afaik this prevents the charger from starting on its own, so evcc does not have to stop it, which most likely causes issues since currently, Volkswagen cars including Audi have issues with stopping and resuming charging and phase switching.

jaytonic, 2025-06-19:
> Tried using `authorize: true`? Afaik this prevents the charger from starting on its own, so evcc does not have to stop it, which most likely causes issues since currently, Volkswagen cars including Audi have issues with stopping and resuming charging and phase switching.

I will try (not enough solar at all right now). I tried to set the charger to "locked single phase" and it worked, I really have the impression that the issue is around phase switching

jaytonic, 2025-06-19:
> If you do, remember to put the charger in "private" mode as well (see easee app). In my setup the phases are 1p or 3p at minimum 6A (I read about 7A required for 3p in other threads shortly). My car has issues with phase switching sometimes but normally it works well. See sometimes op-code 5 errors which require the cable to be disconnected - but that's another story ;-)
> 
> Couldn't you set to auto-detect (phases: 0) as well? Up to 3.6kW is possible using 1p, lowering to 6A/1p ~1.3 minimum - or try using 7A.

Where should I set the phase to autodetect?

mfuchs1984, 2025-06-19:
Selecting Auto / 1p / 3p is done in the UI now.

jaytonic, 2025-06-19:
> Selecting Auto / 1p / 3p is done in the UI now.

I'm in auto mode, but the minimum was set to 1A, but the easee home has a minimum of 6a, so I probably misconfigured something. Maybe because of this it was trying to set 3A and failed.

mfuchs1984, 2025-06-19:
Yes, 6 is the minimum for AC charging.

jaytonic, 2025-06-19:
> mincurrent is configured at vehicle level

Maybe it's also possible on the vehicule, but in the UI, you can set the min current for the loadpoints

jaytonic, 2025-06-19:
> Things change, here it's still at the vehicle (UI), depends on the template. Searched it in the documentation but did not find actually. Therefore I referenced my config. Try it, hopefully it solves the issue

Yeah, I hope it will work, I will wait to see the sun tomorrow :)