evcc-io/evcc #21958: Inkonsistente Anzeige des Ladestands für Fahrzeug (Kona) - Startseite vs. Konfiguration/Fahrzeug-App

David62215, 2025-06-21:
### Describe the bug

Der angezeigte Ladestand meines Elektrofahrzeugs (Hyundai Kona) ist auf der Startseite inkonsistent mit dem Wert, der in den Konfigurationseinstellungen und der Fahrzeug-App angezeigt wird. Dies führt zu Verwirrung und potenziell falschen Annahmen über den aktuellen Ladezustand des Fahrzeugs.

### Steps to reproduce

I don't know

### Configuration details

```yaml
network:
  port: 7070

mqtt:
  broker: 192.168.10.45:1883
  topic: evcc
  user: XXXXXXXXX
  password: XXXXXXXXX

interval: 15s

meters:
  - name: grid
    type: template
    template: sungrow-hybrid
    usage: grid

    modbus: tcpip
    id: 1
    host: 192.168.13.2
    port: 502
    timeout: 10s

  - name: pv
    type: template
    template: sungrow-hybrid
    usage: pv

    modbus: tcpip
    id: 1
    host: 192.168.13.2
    port: 502
    timeout: 10s

  - name: ecoflow
    type: custom
    power: 
      source: mqtt
      topic: "homeassistant/ecoflow/powerstream/solar"

  - name: battery1
    type: template
    template: sungrow-hybrid
    usage: battery

    modbus: tcpip
    id: 1
    host: 192.168.13.2
    port: 502
    timeout: 10s

chargers:
  - name: wallbox
    type: template
    template: abb

    modbus: tcpip
    id: 1
    host: 192.168.11.3
    port: 502

vehicles:
  - name: kona
    type: template
    template: hyundai
    user: 'XXXXXXXXXXXXX'  # Replace with your actual user email or username
    password: 'XXXXXXXXXXXXX'       # Replace with your actual password
    vin: XXXXXXXXXXX
    title: Kona                 # Optional, display name for the vehicle in the UI
    language: de                    # Optional, language (de = German)
    icon: car                       # Optional, icon for the vehicle in the UI
    phases: 3                       # Optional, maximum number of phases for charging
    minCurrent: 1                   # Optional, minimum amperage per connected phase
    maxCurrent: 16                  # Optional, maximum amperage per connected phase
    cache: 120m                      # Optional, cache interval for reloading vehicle data
    capacity: 39

loadpoints:
  - title: Garage
    charger: wallbox
    vehicle: kona
    mode: minpv

    soc:
      poll: 
        mode: charging
        interval: 120m
      estimate: true
    
    enable:
      threshold: 3000
      delay: 1m

    disable:
      threshold: 3000
      delay: 1m

    phases: 0
    mincurrent: 6
    maxcurrent: 16

site:
  title: Zuhause
  meters:
    grid: grid
    pv: 
     - pv
     - ecoflow
    battery: battery1
  residualPower: 100 # additional household usage margin

tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    type: fixed
    price: 0.3079 # EUR/kWh
  feedin:
    type: fixed
    price: 0.08 # EUR/kWh
  solar:
    - type: template
      template: forecast-solar
      lat: XX.XXXX
      lon: XX.XXXX
      dec: 35 # 0 = horizontal, 90 = vertical
      kwp: 8.1
      az: 180
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/06/16 21:52:38 evcc 0.204.2
[main  ] INFO 2025/06/16 21:52:38 using config file: /config/evcc.yaml
[db    ] INFO 2025/06/16 21:52:38 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2025/06/16 21:52:38 connecting evcc-1705631178 at tcp://192.168.10.45:1883
[main  ] INFO 2025/06/16 21:52:38 listening at :7070
[lp-1  ] WARN 2025/06/16 21:52:39 PV mode enable threshold 3000W > 0 will start PV charging on grid power consumption. Did you mean -3000?
[site  ] INFO 2025/06/16 21:52:40 site config:
[site  ] INFO 2025/06/16 21:52:40   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/06/16 21:52:40     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2025/06/16 21:52:40     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/06/16 21:52:40     pv 2:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/06/16 21:52:40     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2025/06/16 21:52:40   vehicles:
[site  ] INFO 2025/06/16 21:52:40     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2025/06/16 21:52:40   tariffs:
[site  ] INFO 2025/06/16 21:52:40     grid:      pricestatic
[site  ] INFO 2025/06/16 21:52:40     feed-in:   pricestatic
[site  ] INFO 2025/06/16 21:52:40     co2:       ✗
[site  ] INFO 2025/06/16 21:52:40     solar:     ✓
[lp-1  ] INFO 2025/06/16 21:52:40 loadpoint 1:
[lp-1  ] INFO 2025/06/16 21:52:40   mode:        minpv
[lp-1  ] INFO 2025/06/16 21:52:40   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2025/06/16 21:52:40   meters:      charge ✓
[lp-1  ] INFO 2025/06/16 21:52:40     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] WARN 2025/06/16 21:52:40 ignoring deprecated minCurrent: 6.000000. please configure via UI
[lp-1  ] WARN 2025/06/16 21:52:40 ignoring deprecated maxCurrent: 16.000000. please configure via UI
[lp-1  ] INFO 2025/06/16 21:52:40 vehicle updated: unknown -> Kona
[site  ] WARN 2025/06/16 21:52:40 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[lp-1  ] INFO 2025/06/16 21:52:41 car connected
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.2

andig, 2025-06-21:
Welcher Wert stimmt denn?

andig, 2025-06-21:
evcc liest den Soc bei Verbindung aus. Warum das bei Dir nicht klappen sollte ist mangels Debuglog nicht erkennbar.

closed 2025-06-21 by andig

David62215, 2025-06-21:
@andig 
Ich habe das Auto am 19.06 angesteckt, da hatte es 26% dann habe ich es bis nachts voll Geladen auf 100% das wurde auch angezeigt als ich um 23:00Uhr am 19.06 drauf geschaut habe. Dann Gestern am 20.06 hab ich so gegen 12:00Uhr drauf geschaut, und es ist bei 19%...
 
Hier sind weiter Logs vom 18.06 - 21.06:
[main  ] INFO 2025/06/18 15:52:40 new version available: 0.204.3
[mqtt  ] ERROR 2025/06/18 21:42:24 tcp://192.168.10.45:1883 connection lost: read tcp 192.168.10.45:47354->192.168.10.45:1883: read: connection reset by peer
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/06/18 21:44:27 evcc 0.204.2
[main  ] INFO 2025/06/18 21:44:27 using config file: /config/evcc.yaml
[db    ] INFO 2025/06/18 21:44:27 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2025/06/18 21:44:27 connecting evcc-1263877267 at tcp://192.168.10.45:1883
[main  ] INFO 2025/06/18 21:44:27 listening at :7070
[lp-1  ] WARN 2025/06/18 21:44:28 PV mode enable threshold 3000W > 0 will start PV charging on grid power consumption. Did you mean -3000?
[site  ] INFO 2025/06/18 21:44:29 site config:
[site  ] INFO 2025/06/18 21:44:29   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/06/18 21:44:29     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2025/06/18 21:44:29     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/06/18 21:44:29     pv 2:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/06/18 21:44:29     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2025/06/18 21:44:29   vehicles:
[site  ] INFO 2025/06/18 21:44:29     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2025/06/18 21:44:29   tariffs:
[site  ] INFO 2025/06/18 21:44:29     grid:      pricestatic
[site  ] INFO 2025/06/18 21:44:29     feed-in:   pricestatic
[site  ] INFO 2025/06/18 21:44:29     co2:       ✗
[site  ] INFO 2025/06/18 21:44:29     solar:     ✓
[lp-1  ] INFO 2025/06/18 21:44:29 loadpoint 1:
[lp-1  ] INFO 2025/06/18 21:44:29   mode:        minpv
[lp-1  ] INFO 2025/06/18 21:44:29   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2025/06/18 21:44:29   meters:      charge ✓
[lp-1  ] INFO 2025/06/18 21:44:29     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] WARN 2025/06/18 21:44:29 ignoring deprecated minCurrent: 6.000000. please configure via UI
[lp-1  ] WARN 2025/06/18 21:44:29 ignoring deprecated maxCurrent: 16.000000. please configure via UI
[lp-1  ] INFO 2025/06/18 21:44:29 vehicle updated: unknown -> Kona
[main  ] INFO 2025/06/18 21:44:29 new version available: 0.204.3
[site  ] WARN 2025/06/18 21:44:29 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[lp-1  ] INFO 2025/06/18 21:44:30 car disconnected
[main  ] INFO 2025/06/19 03:44:29 new version available: 0.204.3
[lp-1  ] INFO 2025/06/19 08:20:00 car connected
[lp-1  ] INFO 2025/06/19 08:20:16 start charging ->
[main  ] INFO 2025/06/19 09:44:29 new version available: 0.204.3
[lp-1  ] INFO 2025/06/19 15:34:15 stop charging <-
[main  ] INFO 2025/06/19 15:44:29 new version available: 0.204.3
[main  ] INFO 2025/06/19 21:44:29 new version available: 0.204.3
[main  ] INFO 2025/06/20 03:44:29 new version available: 0.204.3
[main  ] INFO 2025/06/20 09:44:29 new version available: 0.204.3
[main  ] INFO 2025/06/20 15:44:29 new version available: 0.204.3
[main  ] INFO 2025/06/20 21:44:29 new version available: 0.204.3
[main  ] INFO 2025/06/21 03:44:29 new version available: 0.204.3
[main  ] INFO 2025/06/21 09:44:29 new version available: 0.204.3

andig, 2025-06-21:
> Dann Gestern am 20.06 hab ich so gegen 12:00Uhr drauf geschaut, und es ist bei 19%...

Das mag alles sein, ist ohne Logfile aber nicht nachvollziehbar.

David62215, 2025-06-21:
Wo bekomme ich die logfile her?