evcc-io/evcc #21970: BELGIUM Follow the peak

hyperbart, 2025-06-21:
**Is your feature request related to a problem? Please describe.**
In Belgium we are confronted with the well known "peak tariff", meaning they want to stabilize the grid by having users pay if their per 15 min average power usage exceeds 2.5kW and this peak usage is written down for that month as "the month peak"
Hence it is important to keep this number under control in Belgian House holds.
The P1 (DSMR) provides the current peak of this month in the telegrams.

**Describe the solution you'd like**
A boolean slider enabled yes/no which is called follow the peak.
This enables a function which you could call "follow the peak" and it reads your max power peak of this month (if you generated a peak usage by cooking or using the oven and the stove at the same time) and uses this as a mechanism to follow this peak as you are already penalized for that month.
At the beginning of the month this value should be reset to a value initially defined by the user.

**Describe alternatives you've considered**
I set this manually now by following this up but this is tedious and if I forget it at the end of the month to set it for next month I am penalized.

**Additional context**

Below you can see a telegram from our national meters (DSMR spec, Zuidwijk reader)
In telegram number 1-0:1.6.0 you can see via value (250606130000S) i generated a peak of (02.672*kW) at 25-06-06 at 13H00.

```
/FLU5\xxxxxxxxxxxxx

0-0:96.1.4(xxxxx)
0-0:96.1.1(xxxxxxx)
0-0:1.0.0(xxxxxxxx)
1-0:1.8.1(xxxxxx.652*kWh)
1-0:1.8.2(xxxxxx.775*kWh)
1-0:2.8.1(xxxxxx.960*kWh)
1-0:2.8.2(xxxxxx.471*kWh)
0-0:96.14.0(0002)
1-0:1.4.0(00.050*kW)
1-0:1.6.0(250606130000S)(02.672*kW)
0-0:98.1.0(13)(1-0:1.6.0)(1-0:1.6.0)(xxxxxx)(xxxxxxx)(xxxx*kW)(xxxx)(xxxxx)(x*kW)(xxx)(xxxxxx)(xxxxx*kW)(xxxxxxx)(xxxxxxx)(xxxxx*kW)(xxxxxxxxxx)(xxxxxxxx)(xxxx*kW)(xxxxxxxx)(xxxxxxxx)(xxxxx*kW)(xxxxxx)(xxxxxxxx)(xxxxxx*kW)(xxxxxxxxxx)(xxxxxxxx)(xxxxx*kW)(xxxxxxx)(xxxxxx)(xxxx*kW)(xxxxxx)(xxxxxxx)(xxxxx*kW)(xxxxxxxx)(xxxxxxx)(xxxx*kW)(xxxxxx)(xxxxxx)(xxxx*kW)(xxxxxxx)(xxxxxx)(xxxx*kW)
1-0:1.7.0(00.065*kW)
1-0:2.7.0(00.000*kW)
1-0:21.7.0(01.119*kW)
1-0:41.7.0(00.000*kW)
1-0:61.7.0(00.000*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.990*kW)
1-0:62.7.0(00.063*kW)
1-0:32.7.0(248.0*V)
1-0:52.7.0(240.3*V)
1-0:72.7.0(245.8*V)
1-0:31.7.0(005.79*A)
1-0:51.7.0(004.16*A)
1-0:71.7.0(001.42*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
0-1:24.1.0(003)
0-1:96.1.1(xxxxxxx)
0-1:24.4.0(1)
0-1:24.2.3(xxxxxx)(00927.766*m3)
!D995
/FLU5\xxxxxxxx
```

hyperbart, 2025-06-21:
> Let me shortly summarize how I solved this.
> 
> You need some external controller (Homeassistant here).
> 
> * I defined a 'political' target I want to achieve (let's say 5 kW).
> * put all deferrable loads under evcc
> * configured load management in evcc: the maxPower is dynamically controlled via plugin following the target value provided by HA
> * in HA: monitor the measured peak. if measured peak is higher than target, adapt target to higher value until end of month
> * on 1st of month reset dynamic target to 'political' target.
> 
> And, somewhat defying the purpose of stabilizing the grid you can squeeze it to the limits: At minutes 9, 10, 11, 12, 13 of every quarter hour I look if there is 'power' left for this quarter hour w/o hurting the peak target. If yes, I can scale up the admissible maxPower for the remaining minutes of the quarter hour (and re-check every minute of course).
> 
> (To frame it for the German audience: For every kW peak power above 2.5kW, the Netzentgelt increases by roughly 4 EUR for the given month.)

Euh yes, don’t get me wrong. Thanks for presenting your solution with HA. I am sure it serves you well but this is EVCC, I am not going to run HA if EVCC does everything so fantastically well.

EVCC is together with Homebridge the 2 single pieces of software which are critical in our home.

Needless to say I love EVCC 💚 the progress I have seen and been able to benefit from.

andig, 2025-06-22:
It seems there is a very valid external solution together with load management. Seems to make more sense for us to focus on capabilities (maybe lie zero feed-in) that are otherwise not possible at all atm.

In that sense: wontfix (now).

/cc @naltatis

hyperbart, 2025-06-23:
No issue, this is due to its national use case (Belgium) indeed quite limited so I fully understand the wontfix now.
Is it possible to put it on the backlog or as a follow up to the bigger load management story?

andig, 2025-06-23:
Done

hyperbart, 2025-06-24:
@andig apologies for revisiting this issue.

I found it a bit unfortunate that @RenatusRo mentioned his solution, because that would serve as an argument to non prioritise the issue as there are external controls possible. @RenatusRo I hope you step on Lego 😜

However, he got me thinking and I wanted to explore the API of EVCC and noticed that the circuit maxpower parameter is not accessible via the API.
Would it be OK if I log a feat request/extension of API too for this? Then I can work around the issue easily myself.

RenatusRo, 2025-06-24:
> [@andig](https://github.com/andig) apologies for revisiting this issue.
>
> I found it a bit unfortunate that [@RenatusRo](https://github.com/RenatusRo) mentioned his solution, because that would serve [...]

Yes, felt already miserable for ondermining your FR.
This is the MaxPower Thingie (not an api but a plugin):

https://github.com/evcc-io/evcc/discussions/16784#discussioncomment-11041954

```
circuits:
  - name: main
    maxPower: 6500
    GetMaxPower: 
      source: mqtt
      topic: sim/circuit/maxpower
    meter: grid
    parent:

```

andig, 2026-10-14:
@hyperbart seems this would fall under https://github.com/evcc-io/evcc/issues/34840 Now?

hyperbart, 2026-10-14:
Or the other way around?
I think it would have been better if we had reopened the original discussion: https://github.com/evcc-io/evcc/issues/21970

I remember that the issue I originally created got closed and moved to the epic for Load Management, AT principle is the same as Belgium. Issue for Belgium Peak Tariff was first so I would assume it gets merged in the original one. We close, we group and now we ungroup for this in a new one.

andig, 2026-10-14:
Other issue has more problem context, hence I chose that.

hyperbart, 2026-10-14:
Fully understand, no harm!