evcc-io/evcc #21986: Individual View of Power Consumers in Overview

unstressable, 2025-06-22:
Would be nice to be able to add these devices:

[image]

here:

[image]

In the same way as it is at the solar systems view.

unstressable, 2025-06-22:
> similar idea [#7235](https://github.com/evcc-io/evcc/discussions/7235)

Nice, looks great.
My one would be easy to implement, the other one much prettier.

Maybe a mix of both would be nice.

unstressable, 2025-06-24:
Thanks for clarifying that aux-meters are not intended for my purpose.
I've now integrated my devices as ext-meters, evcc starts well, but I don't see any of them in the graphical user interface.
It would be great if my suggestion above could be considered.

Here's how I did:

meters:
- name: grid1
  type: custom
  power:
    source: mqtt
    topic: evcc_meters/grid1_power

- name: Waschmaschine_links
  type: custom
  power:
    source: mqtt
    topic: evcc_meters/Waschmaschine_links

- name: Waschmaschine_rechts
  type: custom
  power:
    source: mqtt
    topic: evcc_meters/Waschmaschine_rechts

site:
  title: Home
  meters:
    grid: grid1
    ext:
      - Waschmaschine_links
      - Waschmaschine_rechts

naltatis, 2025-06-25:
> It would be great if my suggestion above could be considered.

@unstressable yes, it is considered as a thing we want to do (see labels). My aux-comment was just a hint for you, that you don't run into unexpected behaviors.

michiproep, 2025-08-31:
this is a nice feature. I hope the ext meters will also feed back the HEMS so you get detailed statistics on your hems, too