evcc-io/evcc #21990: DaheimLaden Min+PV funktioniert nicht

Loos-Fuxsteiner, 2025-06-23:
### Describe the bug

Seit letztem Update funktioniert das Min + PV bei Daheimladen Ladestationen nicht mehr.

Fehlermeldung:
```
charger logic error: current mismatch (got 1.02e+06A measured, expected 16A) - make sure your interval is at least 30s
```

"Nigthly Version" kann ich nicht testen, da ich dieses System nicht parallel betreiben kann.

### Steps to reproduce

1. Aktivieren von Min + PV
2.
3.
...

### Configuration details

```yaml
Über UI konfiguriert
```

### Log details

```text
charger logic error: current mismatch (got 1.02e+06A measured, expected 16A) - make sure your interval is at least 30s
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

Loos-Fuxsteiner, 2025-06-23:
Es wird immer mit max. Ladeleistung geladen.

marcelGoerentz, 2025-06-23:
Wie wird der Netzanschluss gemessen?
Ich glaube dort liegt die Ursache des Problems

StefanSchoof, 2025-06-23:
Bei Daheimladen hab gab es einen fix im nightly

closed 2025-06-23 by Loos-Fuxsteiner