evcc-io/evcc #21991: SAX: battery charging power is changed when battery is put into hold

juergen-weber, 2025-06-23:
### Describe the bug

If 'Fast' is selected as the charging mode, a Modbus command is triggered to deactivate the discharge of the battery. In the current template, ‘encoding: uint32’ is used in the batterymode section. Using a 32-bit integer overwrites the adjacent register with ‘0’ - with strange consequences for the operation of the battery. Change to ‘encoding: uint32’ resolves the issue.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
[main  ] INFO 2025/06/23 08:45:10 using config file: /etc/evcc/evcc.yaml
[mqtt  ] INFO 2025/06/23 08:45:10 connecting evcc-2144234481 at tcp://10.0.0.20:1883
[eebus ] ERROR 2025/06/23 08:45:10 websocket server error: listen tcp :4712: bind: address already in use
[main  ] ERROR 2025/06/23 08:45:11 creating vehicle car1 failed: cannot create vehicle type 'template': cannot create vehicle type 'vw': login.error.throttled

Konfiguration (/etc/evcc/evcc.yaml):

network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: 10.0.0.23
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
database:
  type: sqlite
  dsn: /config/evcc/evcc.db

# sponsor token: *****
# sponsortoken: *****

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  cache: error
  db: error
  eebus: error

plant: *****

circuits:
  - name: main # eindeutiger Name, wird als Referenz für andere Circuits verwendet
    title: Hauptstromkreis # Anzeige in der UI (später)
[... 342 lines cut ...]
[circuit-garage-steckdose] DEBUG 2025/06/17 09:32:00 current: 0A
[battery] TRACE 2025/06/17 09:32:00 modbus: send 00 37 00 00 00 06 40 03 00 2f 00 01
[battery] TRACE 2025/06/17 09:32:00 modbus: recv 00 37 00 00 00 05 40 03 02 3f 3d
[battery] TRACE 2025/06/17 09:32:00 modbus: send 00 38 00 00 00 06 40 03 00 2e 00 01
[battery] TRACE 2025/06/17 09:32:00 modbus: recv 00 38 00 00 00 05 40 03 02 00 27
[circuit-garage-steckdose] TRACE 2025/06/17 09:32:00 validate current: 0A + (0A -> 0A) <= 10A ok
[circuit-garage-steckdose] DEBUG 2025/06/17 09:32:30 power: 0W
[circuit-garage-steckdose] DEBUG 2025/06/17 09:32:30 current: 0A
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.3 (a6925c61a)

y-pinky, 2025-07-11:
just to make more sense...

I think the original description is wrong. the last sentence
  "Change to ‘encoding: uint32’ resolves the issue."
should be changed to;
  "Change to ‘encoding: uint16' resolves the issue."

... just to make more sense...

juergen-weber, 2025-07-11:
Ups, apologies. uint16 is the value to be used for writemultiple.