evcc-io/evcc #22008: Option „Verhindere Entladung im Schnellmodus und bei geplantem Laden“ in Verbindung mit E3DC

ArnoD15, 2025-06-23:
Hallo zusammen,

ich verwende EVCC mit einem E3DC S10 E PRO und einer E3DC Wallbox.
Wenn die Option „Verhindere Entladung im Schnellmodus und bei geplantem Laden“ ausgewählt wird, wird beim E3DC Hauskraftwerk die maximale Entladeleistung statisch auf 0 W gesetzt.
Wenn das Schnellladen nun beendet wird, wird die maximale Entladeleistung statisch wieder auf den Standardwert von 12.000 W gesetzt, ebenso wie die untere Lade-/Entladeschwelle, die ebenfalls statisch auf den Standardwert von 95 W gesetzt wird.  

Drei Gründe, warum man das beim E3DC nicht auf diese Weise lösen sollte:
1.	Jede Änderung dieser statischen Einstellungen führt zu einem Schreibzugriff auf die interne SSD des E3DC. E3DC rät davon ab, diese Einstellungen für externe Regelungen zu verwenden, da es dafür bereits eine bessere Lösung gibt.
2.	Wenn EVCC aus irgendeinem Grund die Verbindung zum E3DC verliert, bleiben diese Einstellungen bestehen und führen im Worst Case dazu, dass die Batterie nicht mehr entladen werden kann, bis der Fehler bemerkt wird.
3.	Durch die Verwendung der Standardwerte können keine individuellen Einstellungen der unteren Lade-/Entladeschwelle mehr gesetzt werden, da diese von EVCC überschrieben werden.  

Ist es möglich, das so abzuändern, dass über die RSCP-Schnittstelle die beiden Tags „TAG_EMS_REQ_SET_POWER“ und „TAG_EMS_REQ_SET_POWER_MODE“ für eine dynamische Regelung verwendet werden?
Diese sind nämlich dafür da, die Lade-/Entladeleistung beim E3DC von extern zu steuern, ohne dass dies zu den oben genannten Nachteilen führt.
Um das Entladen der Batterie zu verhindern, muss über die Schnittstelle **SET_POWER** mindestens alle 4 Sekunden der Wert 0 gesendet werden, und **SET_POWER_MODE** muss auf 1 für „idle“ gesetzt werden.
Wenn nun die Verbindung unterbrochen wird, übernimmt das E3DC nach 10 Sekunden wieder die Regelung, und an den Benutzereinstellungen wird nichts geändert.

Viele Grüße 
Arno

andig, 2025-06-23:
Dafür brauchts ne vollständige Beschreibung der neuen Parameter bzw. ihrer Optionen. Die API von e3dc ist leider kryptisch. Am Besten: Du änderst es selbst und testest auch gleich, denn ohne wirds mit Sicherheit falsch...

andig, 2025-06-23:
Siehe e3dc.go

ArnoD15, 2025-06-23:
Hier überschätzt du meine Fähigkeiten.
Ich kann gerade mal Javascript und würde mich auch da nicht als Profi bezeichnen. :-)

0x3d13f, 2025-06-25:
Nicht jeder hat die E3/DC Wallbox im Einsatz.
Gibt es denn irgendwo eine verwertbare Übersicht über alle RSCP Befehle?

ArnoD15, 2025-06-25:
Es gibt die Tagliste von E3DC, diese enthält zumindest die meisten RSCP Befehle.

[RSCP_GUI_TAGS_DE_1.1.pdf](https://github.com/user-attachments/files/20906678/RSCP_GUI_TAGS_DE_1.1.pdf)

[Remote-Storage-Control-Protocol_2015_04_30_DE.pdf](https://github.com/user-attachments/files/20906688/Remote-Storage-Control-Protocol_2015_04_30_DE.pdf)

[RSCP-Tags-Official.pdf](https://github.com/user-attachments/files/20906750/RSCP-Tags-Official.pdf)

[RSCP-Tags-Official.pdf](https://github.com/user-attachments/files/20906786/RSCP-Tags-Official.pdf)