evcc-io/evcc #22033: at sunset charging continues with minimum without surplus in pv mode

derflucher, 2025-06-24:
### Describe the bug

charge mode PV and car connected with enough surplus to start charging in PV mode
at sunset the house-battery takes over and charging continues with set minimum (8A)

### Steps to reproduce

1.set mode PV
2.connect EV with enough surplus to start charging
3. wait for sunset :) 
4. charging continues with minimum set power
5. after restart of evcc the logic seems to be correct again (timer starts to stop the active charging session)
...

### Configuration details

```yaml
most of configuration done via UI - dump only includes pv and battery

Charging Points
go-e Charger V3 [db:1]
default Solar
maximum solar
priority 1
type 11kW
auto phases

my-PV AC ELWA 2 [db:3]
max solar
priority 0
type other
min 2A, max 16A
1-phase

Vehicle
Renault
52kWh
default mode Solar
max phases 3-phases
min current 8A, max 16A
Priority 0

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.hof92 # .local suffix announces the hostname on MDNS
  port: 7070

log: warn
levels:
  site: warn
  cache: error
  renault: debug

# unique installation id
plant: *****

interval: 30s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: victron-energy 
[... 745 lines cut ...]
[lp-1  ] DEBUG 2025/06/24 21:09:01 charge voltages: [233 236 238]V
[lp-1  ] DEBUG 2025/06/24 21:09:01 detected active phases: 1p
[lp-1  ] DEBUG 2025/06/24 21:09:01 charge total import: 5265.244kWh
[lp-1  ] DEBUG 2025/06/24 21:09:01 charger status: C
[lp-1  ] DEBUG 2025/06/24 21:09:01 soc estimated: 78.32% (vehicle: 78.00%)
[lp-1  ] DEBUG 2025/06/24 21:09:01 vehicle soc: 78%
[lp-1  ] DEBUG 2025/06/24 21:09:01 vehicle range: 267km
[lp-1  ] DEBUG 2025/06/24 21:09:01 pv charge current: min 8A > 0A (2923W @ 1p, battery: true)
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.2

andig, 2025-06-24:
Bitte alles trace und db aus dem Log raus damits lesbar wird.

derflucher, 2025-06-25:
> How are your home battery limits configured?

Battery-supported vehicle charging
when home battery is above 70%.
Start automatically only with enough surplus.

Prioritize vehicle charging
when home battery is above 50%.

Prioritize home battery charging
until it reaches 50%.

VolkerK62, 2025-06-25:
Du hast Batterieunterstützung bis 70% erlaubt.
Deine Batterie hat 88%. Deshalb wird weiter geladen.

closed 2025-06-25 by derflucher