evcc-io/evcc #22034: Elli Charger 2  Pro manuelle Phasenumschaltung

Patten310, 2025-06-24:
### Describe the bug

Da die phadenumschaltung über EEBUS nicht funktioniert schalten ich diese jetzt manuell über die Webseite des Chargers. Danach funktioniert die Darstellung in evcc nicht mehr richtig, das Fahrzeug wird nicht mehr als Ladend angezeigt. 

### Steps to reproduce

1. Laden starten 3 Phasen 
2. umschalten auf der Elli Charger Webseite auf 1 Phase 
3.nach kurzer seit lädt der Charger 1 Phasig ohne weitere anzeigen 

[image]

### Configuration details

```yaml
interval: 30s # control cycle interval

meters:
- type: template
  template: siemens-pac2200
  usage: grid
  modbus: tcpip
  id: 1
  host: 192.168.178.26
  port: 502
  name: grid1
- type: template
  template: senec-home
  usage: pv
  host: 192.168.178.60
  schema: https
  name: pv2
- type: template
  template: senec-home
  usage: battery
  host: 192.168.178.60
  schema: https
  name: battery3
chargers:
- type: template
  template: elli-2
  ski: 
  ip: 192.168.178.37
  name: wallbox5

vehicles:
- type: template
  template: vw
  user: 
  password: 
  vin: WVW
  title: ID7 Tourer
  capacity: 77
  timeout: 10s
  phases: 3
  name: ev4
  welcomecharge:

loadpoints:
- title: Carport
  charger: wallbox5
  mode: pv

site:
  title: Home
  meters:
    grid: grid1
    pv:
    - pv2
battery:
[... 192 lines cut ...]
[site ] DEBUG 2025/06/24 21:45:00 pv 1 power: -0W
[site ] DEBUG 2025/06/24 21:45:00 battery 1 power: 1578W
[site ] DEBUG 2025/06/24 21:45:00 grid currents: [-6.33 6.33 0.544]A
[site ] DEBUG 2025/06/24 21:45:00 battery 1 soc: 68%
[site ] DEBUG 2025/06/24 21:45:00 site power: 1566W
[lp-1 ] DEBUG 2025/06/24 21:45:00 charger status: B
[lp-1 ] DEBUG 2025/06/24 21:45:00 pv charge current: min 6A > 0A (1566W @ 3p, battery: false)
[eebus ] DEBUG 2025/06/24 21:45:01 Recv: d:_i:3210_Elli-00004316:Measurement to Measurement notify 24328 MeasurementListData
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.3

Patten310, 2025-06-25:
> So phase switching actually work for you? I always get following error when I try it
> 
> [image]
> 
> Would you mind sending me your device information tab? Mine looks like this where only 31MB of ROM is used which seems weird
> 
> [image]

Phase switching works, but the car must be loading. You can not switch the Phase when no car is connected. 
The ROM usage is the same

andig, 2025-06-25:
Duplicate of https://github.com/evcc-io/evcc/issues/21658

Patten310, 2025-06-25:
> Duplicate of [#21658](https://github.com/evcc-io/evcc/issues/21658)

No i think this is something diffrent. The Phase switching is working without eebus but the evcc screen dont show it

Patten310, 2025-07-06:
> [@Patten310](https://github.com/Patten310) your screenshot shows OCPP. We support phase switching over OCPP just fine.
Thanks for your hint. I just have to use a ocpp backend for my work. I think the manual switch on the Elli Website is the only solution