evcc-io/evcc #22041: heating: Temperaturlimit wird bei "AUS" immer auf 100 gesetzt

LeonSol, 2025-06-25:
### Describe the bug

Betr. "heating"

Moin zusammen,

vielen Dank für die schnelle Lösung #22014 - hierbei kommt es noch zu einem kleinen Fehler:

Wenn ich ein Temperaturlimit setze (egal in welchem Modus) wird dieses bei betätigen von "AUS"  wieder auf 100 gesetzt nach dem ganz kurz "**Geladen** mit 0,0 kWh und **Ladelimit** mit keins" angezeigt werden und erst dann wieder auf "Temperatur" und "Temperaturlimit" gewechselt wird, letzteres aber halt mit 100 und nicht den vorher eingestellten 60 Grad.
 
Sonnige Grüße

Leon

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Not needed
```

### Log details

```text
Not needed
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

RenatusRo, 2025-06-25:
die 100 ist ein Klassiker:

heating attribut rausnemen
neustart
im loappoint,planner, ankunftstab default limit auf 60
heating attribut wierder  rein
neustart

LeonSol, 2025-06-25:
Naja, und wenn ich dann das Limit z.Bsp. auf 56 Grad setzen möchte, dann knallt es mir bei "AUS" immer 60 Grad rein, oder?

LeonSol, 2025-06-25:
Dann ist das keine Lösung, da es ja vorher mit dem Limit geklappt!

andig, 2025-06-25:
Was für ein Limit? Logfile?

erikgieseler, 2025-06-25:
Hab genau das gleiche Problem mit lp-4.
Der schaltet von 60 zu 100 wenn `Aus` und bleibt bei 100 wenn ich wieder auf `Schnell` schalte.
Dann muss ich manuell wieder auf 60 umschalten, siehe Log.
[evcc-20250625-170439-trace.log](https://github.com/user-attachments/files/20906393/evcc-20250625-170439-trace.log)

andig, 2025-06-25:
Da beisst sich die Katze jetzt in den Schwanz. Das Limit ist per Session und die beenden wir jetzt per brute force. Root cause ist egtl. die "falsche" Erwartung dass es überhaupt sowas wie Ladesessions bei Heating gibt.

Wir können die beiden Änderung reverten oder das so akzeptieren. 

@naltatis hast du eine bessere Idee?

LeonSol, 2025-06-25:
Och, ich würde eine Warmwasserladung schon als Ladesession sehen und es geht ja darum, PV-Überschuss im Haus zu lassen.

Aber laienhaft gesprochen: wie wäre es statt des "harten" Beenden ein weiches Verfahren zu wählen, in dem man bei "AUS" den vehicleSoc über den limitSoc hebt? Den Rest sollte EVCC dann doch automatisch machen, oder?

VolkerK62, 2025-06-26:
Evtl sollte man https://github.com/evcc-io/evcc/issues/13050 in den Fokus nehmen.
Dann hätte sich das ganze Thema doch erledigt?

VolkerK62, 2025-06-26:
Du willst doch einen Ladelog-Eintrag erhalten. Dazu reicht es nicht, den Ladevorgang zu beenden. Dazu muss die Ladesession beendet werden und dazu muss aktuell "das Fahrzeug abgesteckt" werden. Das passiert mit "Aus".
mögliche Alternativen:
Den Ladelogeintrag i.V.m. heating zeitlich gesteuert vornehmen (das wäre #13050) 
Oder jeder Ladevorgang erzeugt einen Logeintrag. Dann braucht es den Hack mit "Aus" nicht.

docolli, 2025-06-26:
Grund für das Ändern des Limits bei AUS ist doch das limitsoc des vehicle. https://github.com/evcc-io/evcc/issues/13679
Dies kann als Workaround auch einfach per API auf einen passenden Wert gesetzt werden: 
`/vehicles/{name}/limitsoc/{soc}` Set SoC limit
So muss man nicht das "heating" aus der Konfig raus nehmen, evcc neu starten, den Wert setzen, und das Ganze wieder zurück bauen mit nochmaligem evcc Neustart.

@LeonSol : Bei einem "heating" loadpoint ist vehicleSoc ja die Temperatur. Wenn man diese nun im charger-code für SOC() auf einen Wert über limitSoc hebt, falls der LP Status = "Aus" ist, dann verliert man die Anzeige der tatsächlichen Temperatur wenn der LP aus "Aus" ist.

LeonSol, 2025-06-26:
@andig das wäre absolut ausreichend und bis auf den nicht beendeten Ladevorgang lief vor dem update auch alles rund.