evcc-io/evcc #22045: Battery Config Page Shows "Dynamic Price Not Configured" After 0.204.5 Update (Web UI Only)

galadril, 2025-06-25:
### Describe the bug

On the battery config page, it shows on dynamic price configured after updating to 0.204.5:
[image]

but it is configured:
[image]

config:
[image]

i dont see this issue on the mobile app (android)

### Steps to reproduce

- Update to 0.204.5
- Click on the home battery menu item
- Go to the second tab

### Configuration details

```yaml
Im using experimental ui so in db
```

### Log details

```text
- no error in logs
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.5

galadril, 2025-06-25:
@andig How do i get the config, i have the experimental mode turned on, so db mode
I cant read the second line, english? :)

andig, 2025-06-26:
evcc config

Should do the trick

andig, 2025-06-26:
It should show db config, too. Is it hitting the right DB file?

/cc @naltatis brauchts hier noch einen Check für Service-DB?