evcc-io/evcc #22061: Integrated Device Misreporting Charging State

nriel, 2025-06-26:
### Describe the bug

Warmwasser charger shows "Ladebereit" despite active 3 kW load

Hello evcc team,

I'm using evcc version v0.204.5 with a template-based charger setup for my heat pump water heater ("Warmwasser"). Even though the connected power meter (WPmeter) correctly shows a consumption of 3.0 kW, the charger state remains stuck on “Ladebereit. Warte auf Fahrzeug …” instead of transitioning to “Ladevorgang läuft”.

From what I understand, integrateddevice: true may cause evcc to rely only on the charger's internal state and ignore the meter, but this makes little sense for heat pumps. The power draw is real and above standbypower, so evcc should detect this as active charging.

Expected behavior:
When the meter shows sustained power > 3.0 kW, evcc should indicate active charging ("Ladevorgang läuft").

Observed behavior:
evcc reports only "Ladebereit", even with 3.0 kW load.

Let me know if I should remove integrateddevice: true, or if this is a bug in how template chargers interact with meters.

Thanks in advance!

Best regards,
Nils

### Steps to reproduce

1. Set up a Shelly switch-based charger using the template type with integrateddevice: true and standbypower set:
`- name: WP_SG
  type: template
  integrateddevice: true
  template: shelly
  host: 192.168.1.189
  icon: heater
  standbypower: 50
`
2. Configure a custom meter to monitor the same load (e.g., Shelly EM or similar), providing both power and energy:
```
- name: WPmeter
  type: custom
  power:
    source: http
    uri: http://192.168.1.194/emeter/0
    jq: .power
  energy:
    source: http
    uri: http://192.168.1.194/emeter/0
    jq: .total / 1000
```

3.Link both the charger and meter in a site config:
```
- title: Warmwasser
  charger: WP_SG
  meter: WPmeter
  mode: pv
  priority: 3
  enable:
    threshold: -2300
    delay: 1m
  disable:
    threshold: 100
    delay: 5m
```
4. Ensure the heat pump is actively drawing power (confirmed via meter at ~3.0 kW).
5. Open the evcc UI and observe the status of the Warmwasser charger.

### Configuration details

```yaml
- name: WPmeter
  type: custom
[... 647 lines cut ...]
[site ] DEBUG 2025/06/26 11:38:19 battery 1 soc: 74%
[site ] DEBUG 2025/06/26 11:38:19 battery 2 soc: 74%
[site ] DEBUG 2025/06/26 11:38:19 battery power: -2231W
[site ] DEBUG 2025/06/26 11:38:19 battery soc: 74%
[site ] DEBUG 2025/06/26 11:38:19 battery has priority at soc 74% (< 90%)
[site ] DEBUG 2025/06/26 11:38:19 site power: 73W
[lp-1 ] DEBUG 2025/06/26 11:38:19 charge voltages: [5.3 0 5.2]V
[lp-1 ] DEBUG 2025/06/26 11:38:19 charger status: A
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.204.5

andig, 2025-06-26:
Lass einfach den extra Zähler weg.

andig, 2025-06-28:
> Wie sehe ich dann, ob geladen wird oder nicht?

Siehst Du nicht.

> PV-ready deaktiviert: Wärmepumpe arbeitet, da sie selbst entscheidet, dass Warmwasser produziert werden muss (ca. 400-3000W Verbrauch - mit oder ohne Heizstab)

Genau das beschreibt dein Issue. Da wir aber nur auf den Boostmodus schauen (="charging") ignorieren wir das. Anderefalls würde der Ladepunkt "out of sync" sagen da er nicht aktiv eingeschaltet ist...