evcc-io/evcc #22101: Bei "min+pv" wird Min-Current Limit aus der Weboberfläche wird nicht beachtet.

sipiyou, 2025-06-28:
### Describe the bug

Wenn die Option MIn+PV verwendet wird und Min-Current z.B. auf 10A gesetzt wird, lädt evcc trotzdem nicht mit 10A.

In diesem Beispiel ist es mit 1P, das Problem lässt sich auch mit 3P auf der gleichen Weise reproduzieren.

### Steps to reproduce

1. EVCC_Webseite öffnen, auf die Optionen klicken, Min-Current aus der Dropbox auswählen (z.B. 10A)
2. Dann auch MIN+PV klicken
3. EVCC lädt mit 6A min.

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

#log: warn
log: debug
levels:
  cache: error

# unique installation id
plant: xxx

interval: 30s # control cycle interval

circuits:
  - name: main # eindeutiger Name, wird als Referenz für andere Circuits verwendet
    title: Hauptstromkreis # Anzeige in der UI (später)
#    maxCurrent: 63 # 63A (optional)
    maxPower: 8000 # 8kW (optional)
#    meter: grid1 # optional

tariffs:
  grid: # price using energy from the grid
    type: fixed
    price: 0.23 # EUR/kWh

  feedin: # price for feeding solar energy to the grid
    type: fixed
    price: 0.0663 # EUR/kWh

  solar:
     type: custom
     interval: 30m     
     forecast:
       source: http
       uri: http://192.168.1.101/verbrauch/solcast.php

meters:
- type: template
  template: fronius-solarapi-v1 
  usage: grid  
  host: 192.168.1.180  
  user: xxx
  password: xxx
  name: grid1
- type: template
  template: fronius-solarapi-v1 
  usage: pv  
  host: 192.168.1.180  
  user: xxx
  password: xxx
  name: pv2
- type: template
  template: fronius-solarapi-v1 
  usage: battery  
  host: 192.168.1.180  
  user: xxx
  password: xxx
  name: battery3
  capacity: 7.7 # in kWh
  
chargers:
  - name: wallbox2
    type: template
    template: openwb-pro
    host: 192.168.1.26 # Hostname

loadpoints:
- title: Garage
  charger: wallbox2
  mode: pv
  phases: 0
  circuit: main
#  minCurrent: 6
#  maxCurrent: 10

vehicles:
- name: nima_eqs
  type: template
  template: mercedes
  title: xxx
  capacity: 107.8 # Battery capacity in kWh
  user: xxxx
  region: EMEA
  accessToken: xxx
  refreshToken: xxx
  vin: xxx
- name: xxxx
  type: template
  template: mercedes
  title: xxxx
  capacity: 67 # Battery capacity in kWh
  user: xxx
  region: EMEA
  accessToken: xxx
  refreshToken: xxx
  vin: xxx

site:
  title: zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### Log details

```text
[lp-1 ] INFO 2025/06/28 08:40:50 stop charging <-
[lp-1 ] DEBUG 2025/06/28 08:40:50 charge total import: 204.300kWh
[lp-1 ] DEBUG 2025/06/28 08:40:50 vehicle status: B
[lp-1 ] DEBUG 2025/06/28 08:40:50 vehicle soc: 66%
[lp-1 ] DEBUG 2025/06/28 08:40:50 vehicle soc limit: 80%
[lp-1 ] DEBUG 2025/06/28 08:40:50 vehicle range: 379km
[site ] DEBUG 2025/06/28 08:40:50 solar forecast: accumulated 23.636kWh, produced 28.097kWh, scale 1.189
[lp-1 ] DEBUG 2025/06/28 08:41:04 set min current: 10
[lp-1 ] DEBUG 2025/06/28 08:41:10 set phases: 1
[site ] DEBUG 2025/06/28 08:41:10 ----
[lp-1 ] DEBUG 2025/06/28 08:41:10 charge power: 0W
[lp-1 ] DEBUG 2025/06/28 08:41:10 charge currents: [0 0 0]A
[circuit-main] DEBUG 2025/06/28 08:41:10 power: 0W
[circuit-main] DEBUG 2025/06/28 08:41:10 current: 0A
[site ] DEBUG 2025/06/28 08:41:10 grid power: 19W
[site ] DEBUG 2025/06/28 08:41:10 pv 1 power: 2081W
[site ] DEBUG 2025/06/28 08:41:10 battery 1 power: -1690W
[site ] DEBUG 2025/06/28 08:41:10 battery 1 soc: 50%
[site ] DEBUG 2025/06/28 08:41:10 site power: -1671W
[lp-1 ] DEBUG 2025/06/28 08:41:10 charge voltages: [231 233 232]V
[lp-1 ] DEBUG 2025/06/28 08:41:10 charge total import: 204.300kWh
[lp-1 ] DEBUG 2025/06/28 08:41:10 charger status: B
[site ] DEBUG 2025/06/28 08:41:10 solar forecast: accumulated 23.637kWh, produced 28.097kWh, scale 1.189
[lp-1 ] DEBUG 2025/06/28 08:41:17 set charge mode: minpv
[site ] DEBUG 2025/06/28 08:41:17 ----
[lp-1 ] DEBUG 2025/06/28 08:41:17 charge power: 0W
[lp-1 ] DEBUG 2025/06/28 08:41:17 charge currents: [0 0 0]A
[circuit-main] DEBUG 2025/06/28 08:41:17 power: 0W
[circuit-main] DEBUG 2025/06/28 08:41:17 current: 0A
[site ] DEBUG 2025/06/28 08:41:17 grid power: -6W
[site ] DEBUG 2025/06/28 08:41:17 pv 1 power: 2070W
[site ] DEBUG 2025/06/28 08:41:17 battery 1 power: -1673W
[site ] DEBUG 2025/06/28 08:41:17 battery 1 soc: 50%
[site ] DEBUG 2025/06/28 08:41:17 site power: -1680W
[lp-1 ] DEBUG 2025/06/28 08:41:17 charge voltages: [231 233 233]V
[lp-1 ] DEBUG 2025/06/28 08:41:17 charge total import: 204.300kWh
[lp-1 ] DEBUG 2025/06/28 08:41:17 charger status: B
[lp-1 ] DEBUG 2025/06/28 08:41:17 pv charge current: min 10A > 7.3A (-1680W @ 1p, battery: false)
[lp-1 ] DEBUG 2025/06/28 08:41:17 set charge current limit: 10A
[lp-1 ] DEBUG 2025/06/28 08:41:17 charger enable
[lp-1 ] DEBUG 2025/06/28 08:41:17 wake-up timer: start
[site ] DEBUG 2025/06/28 08:41:17 solar forecast: accumulated 23.638kWh, produced 28.097kWh, scale 1.189
[site ] DEBUG 2025/06/28 08:41:19 ----
[lp-1 ] DEBUG 2025/06/28 08:41:19 charge power: 0W
[lp-1 ] DEBUG 2025/06/28 08:41:19 charge currents: [0 0 0]A
[circuit-main] DEBUG 2025/06/28 08:41:19 power: 0W
[circuit-main] DEBUG 2025/06/28 08:41:19 current: 0A
[site ] DEBUG 2025/06/28 08:41:19 grid power: -10W
[site ] DEBUG 2025/06/28 08:41:20 pv 1 power: 2067W
[site ] DEBUG 2025/06/28 08:41:20 battery 1 power: -1662W
[site ] DEBUG 2025/06/28 08:41:20 battery 1 soc: 50%
[site ] DEBUG 2025/06/28 08:41:20 site power: -1673W
[lp-1 ] DEBUG 2025/06/28 08:41:20 charge voltages: [231 233 232]V
[lp-1 ] DEBUG 2025/06/28 08:41:20 charge total import: 204.300kWh
[lp-1 ] DEBUG 2025/06/28 08:41:20 charger status: B
[lp-1 ] DEBUG 2025/06/28 08:41:20 pv charge current: min 10A > 7.27A (-1673W @ 1p, battery: false)
[site ] DEBUG 2025/06/28 08:41:20 solar forecast: accumulated 23.638kWh, produced 28.097kWh, scale 1.189
[site ] DEBUG 2025/06/28 08:41:49 ----
[lp-1 ] DEBUG 2025/06/28 08:41:49 charge power: 1812W
[lp-1 ] DEBUG 2025/06/28 08:41:49 charge currents: [7.94 0 0]A
[circuit-main] DEBUG 2025/06/28 08:41:49 power: 1812W
[circuit-main] DEBUG 2025/06/28 08:41:49 current: 7.94A
[site ] DEBUG 2025/06/28 08:41:49 grid power: 24W
[site ] DEBUG 2025/06/28 08:41:50 pv 1 power: 2097W
[site ] DEBUG 2025/06/28 08:41:50 battery 1 power: 151W
[site ] DEBUG 2025/06/28 08:41:50 battery 1 soc: 50%
[site ] DEBUG 2025/06/28 08:41:50 site power: 175W
[lp-1 ] DEBUG 2025/06/28 08:41:50 charge voltages: [230 235 230]V
[lp-1 ] DEBUG 2025/06/28 08:41:50 charge total import: 204.300kWh
[lp-1 ] DEBUG 2025/06/28 08:41:50 charger status: C
[lp-1 ] INFO 2025/06/28 08:41:50 start charging ->
[lp-1 ] DEBUG 2025/06/28 08:41:50 wake-up timer: stop
[lp-1 ] DEBUG 2025/06/28 08:41:50 soc estimated: 66.01% (vehicle: 66.00%)
[lp-1 ] DEBUG 2025/06/28 08:41:50 vehicle soc: 66%
[lp-1 ] DEBUG 2025/06/28 08:41:50 vehicle soc limit: 80%
[lp-1 ] DEBUG 2025/06/28 08:41:50 vehicle range: 379km
[lp-1 ] DEBUG 2025/06/28 08:41:50 pv charge current: min 10A > 9.18A (175W @ 1p, battery: false)
[site ] DEBUG 2025/06/28 08:41:50 solar forecast: accumulated 23.640kWh, produced 28.097kWh, scale 1.189
[site ] DEBUG 2025/06/28 08:42:19 ----
[lp-1 ] DEBUG 2025/06/28 08:42:19 charge power: 1810W
[lp-1 ] DEBUG 2025/06/28 08:42:19 charge currents: [7.94 0 0]A
[circuit-main] DEBUG 2025/06/28 08:42:19 power: 1810W
[circuit-main] DEBUG 2025/06/28 08:42:19 current: 7.94A
[site ] DEBUG 2025/06/28 08:42:19 grid power: -12W
[site ] DEBUG 2025/06/28 08:42:20 pv 1 power: 2103W
[site ] DEBUG 2025/06/28 08:42:20 battery 1 power: 121W
[site ] DEBUG 2025/06/28 08:42:20 battery 1 soc: 50%
[site ] DEBUG 2025/06/28 08:42:20 site power: 109W
[lp-1 ] DEBUG 2025/06/28 08:42:20 charge voltages: [229 237 230]V
[lp-1 ] DEBUG 2025/06/28 08:42:20 detected active phases: 1p
[lp-1 ] DEBUG 2025/06/28 08:42:20 charge total import: 204.300kWh
[lp-1 ] DEBUG 2025/06/28 08:42:20 charger status: C
[lp-1 ] DEBUG 2025/06/28 08:42:20 soc estimated: 66.02% (vehicle: 66.00%)
[lp-1 ] DEBUG 2025/06/28 08:42:20 vehicle soc: 66%
[lp-1 ] DEBUG 2025/06/28 08:42:20 vehicle soc limit: 80%
[lp-1 ] DEBUG 2025/06/28 08:42:20 vehicle range: 379km
[lp-1 ] DEBUG 2025/06/28 08:42:20 pv charge current: min 10A > 9.47A (109W @ 1p, battery: false)
[site ] DEBUG 2025/06/28 08:42:20 solar forecast: accumulated 23.643kWh, produced 28.097kWh, scale 1.188
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.5 (b872dbf2)

andig, 2025-06-28:
> EVCC lädt mit 6A min.

Wo sehe ich das im Log?

marcelGoerentz, 2025-06-28:
Nach start Charge sind es doch  9,7A, da passt es doch

sipiyou, 2025-06-28:
Was soll ich einstellen, damit es in der Log sichtbar ist ? Ich hab schon auf Debug.

Das Problem kann man ganz einfach wie oben beschrieben reproduzieren.

VolkerK62, 2025-06-28:
Es werden 10A freigegeben und mit 8A geladen
```
[lp-1 ] DEBUG 2025/06/28 08:41:17 set charge current limit: 10A
[lp-1 ] DEBUG 2025/06/28 08:42:19 charge currents: [7.94 0 0]A
```

mfuchs1984, 2025-06-28:
Evcc gibt, soweit ich in Log sehen kann, 10 A frei, das Auto lädt dann mit knapp 8.

andig, 2025-06-28:
Also wie erwartet.