evcc-io/evcc #22119: SOC Victron after Update v3.62

mmanderscheid, 2025-06-29:
### Describe the bug

SOC, Power,….read faild

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
SOC, Power,….read faild
```

### Log details

```text
SOC, Power,….read faild
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

VolkerK62, 2025-06-29:
echt jetzt???
mit dem "weniger als nichts" an Info, erwartest du Hilfe?

closed 2025-06-29 by andig

mmanderscheid, 2025-06-29:
Hallo Danke,für die nette Rückmeldung.Ich kenne mich bei dem Gith Hub Portal nicht aus. Und habe versucht nur ob es geht hier ein Thema zu positionieren. Aber danke nochmal benötige keine Hilfe mehr beim mir geht es und der Rest ist mir dann auch egal.Unterstützt habe ich euch das war mir Wichtig!Grüße MathiasVon meinem iPhone gesendet