evcc-io/evcc #22141: Phase switching with Zaptec Pro

helijunky, 2025-06-30:
### Describe the bug

Hi,
wie ich herausgefunden habe, funktioniert die Phasenumschaltung bei Zaptec Pro nur, wenn im Anschluss der Strom verändert wird (mehr oder weniger Ampere). Wenn also nur von 1 auf 3 Phasen (oder umgekehrt) umgestellt wird passiert erstmal nichts. Erst nachdem der Strom (max. current) verändert wurde, wird der Ladevorgang kurz gestoppt und mit der neuen Phasenanzahl und Strom erneut gestartet. Leider kommt dies in der Praxis ab und zu vor, was zu "phase mismatch" Fehlern führt.

Das Verhalten ist offenbar so gewollt und wurde vom Zaptec Support bestätigt:
"We're fully aware of this behavior and should be considered when integrating with us. (if planned to set phases on charger level) ... there should be always a current change with a phase change."

Frage: wie wurde bei evcc die Phasenumschaltung für Zaptec implementiert? Gibt es jeweils im Anschluss eine Stromanpassung?

Vorschlag (falls nicht schon ähnlich implementiert): nach jeder Phasenumschaltung den max. Strom verändern (+/- 1 Ampere). Dadurch wird gewährleistet, dass die Phasenumschaltung sofort und in jedem Fall erfolgt.

### Steps to reproduce

1. systemctl stop evcc.service
2. evcc charger -p 1 -i 6 -e # Ladevorgang mit 1 Phase@6A starten
3. evcc charger # status prüfen
4. evcc charger -p 3 # auf 3 Phasen umschalten --> hier passiert noch nichts
5. evcc charger # status prüfen
6. evcc charger -i 7 # Strom ändern (egal welcher Wert ausser 6) --> wechselt sofort auf 3 Phasen@7A
7. evcc charger # status prüfen

### Configuration details

```yaml
Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

database:
  type: sqlite
  dsn: /var/lib/evcc/evcc.db

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 30s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- name: fronius_symo-7-0-3-M
  type: template
  template: fronius-gen24
  usage: pv
  host: 192.168.20.12
  port: 502
  maxacpower: 7000

- name: fronius_smartmeter
  type: template
  template: sunspec-inverter
  id: 240 
  host: 192.168.20.12
  port: 502
  usage: grid
  modbus: tcpip

chargers:
- name: zaptec_pro_PP22
  type: template
  template: zaptec 
  id: 032a0fb6-8fa2-44df-8125-*****
  #user: *****
  #password: *****
  user: *****
  password: *****

loadpoints:
- title: PP 22
  charger: zaptec_pro_PP22
  mode: pv
  vehicle: kia_ev3
  enable:
    threshold: 0
    delay: 1m
  disable:
    threshold: 500
    delay: 3m

vehicles:
- name: kia_ev3
  type: template
  template: kia
  user: *****
  password: *****
  title: 'Kia EV3'
  icon: car
  capacity: 81.4
  phases: 3
  minCurrent: 6
  maxCurrent: 16
  cache: 15m
  language: de

site:
  title: ****
  meters:
    grid: fronius_smartmeter
    pv: fronius_symo-7-0-3-M
```

### Log details

```text
evcc charger -p 1 -i 6 -e && sleep 30 && evcc charger
Power:          1269W
Current L1..L3: 5.44A 0.152A 0.17A
Charge status:  C
Enabled:        false
Charged:        0.0kWh
Identifier:     <none>

evcc charger -p 3 && sleep 30 && evcc charger
Power:          1269W
Current L1..L3: 5.44A 0.152A 0.17A
Charge status:  C
Enabled:        false
Charged:        0.1kWh
Identifier:     <none>

evcc charger -i 7 && sleep 30 && evcc charger
Power:          4769W
Current L1..L3: 6.71A 6.76A 6.75A
Charge status:  C
Enabled:        false
Charged:        0.1kWh
Identifier:     <none>
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.5 (57f59ca7)

andig, 2025-07-02:
@tbaehler kannst Du das nachvollziehen? Es wird ja schon `AvailableCurrentPhase` gesetzt, aber nicht `MaxChargeCurrent`?

helijunky, 2025-07-03:
@tbaehler es muss zwingend ein anderer Stromwert gesetzt werden, damit der Phasenwechsel übernommen wird. Beispiel: wenn vorher X gesetzt war, könnte man auf X+1 setzen und gleich wieder zurück auf X. Dann wäre sichergestellt, dass der Wert verändert wurde. Wenn ein fixer Wert gesetzt wird besteht immer die Gefahr, dass der Wert vorher schon so war.

andig, 2025-07-06:
> bisher hatte ich das Problem nicht.

@tbaehler könntest du es evtl. mal mit evcc "aus" gezielt testen?

> man könnte beim switch von 1 zu 3 auf 4.1kw und beim 3:1 switch auf 3.7 schalten. das sollte reichen rnicht?

Ändern sollten wir es nur wenn es nachvollziehbar ist.

tbaehler, 2025-07-07:
Ich habe eine Zaptec2 da funktioniert der phase switch über das installations api. 
Aber ich kann mal einen workaround vorbereiten. testen kann ich ihn aber nicht

tbaehler, 2025-07-07:
Ok I made a linux build here 
https://www.dropbox.com/scl/fi/obcw9rcjtezuuicgdqeyo/evcc?rlkey=0qpls3lycyoa3x590a6is9c0r&st=a1w3hwc5&dl=0
it reads the CurrentPower sets the MaxCurrent to 0 then back to CurrentPower. 
But I am total blind here. Can someone test this build before I create a Pull Request? If you see some errors, please post the logs here...

helijunky, 2025-07-07:
@tbaehler I just tested your build. It looks quite well except that the current set at the end is wrong (see last line of the log below). It seems you are setting it to the stateID 513 (1477 in the log below). However, you should set it to **stateID 510** which corresponds to the maxChargeCurrent.
I am pretty sure it will work if you could fix this. Many thanks!

```

./evcc.phasefix1 charger -p 3 --log trace

[main
] INFO 2025/07/07 17: 19: 56 evcc 8ec34b15 (8ec34b15)
[main
] INFO 2025/07/07 17: 19: 56 using config file: /etc/evcc.yaml
[db
] INFO 2025/07/07 17: 19: 56 using sqlite database: /var/lib/evcc/evcc.db
[db
] TRACE 2025/07/07 17: 19: 56 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings"-1 <nil>
[db
] TRACE 2025/07/07 17: 19: 56 SELECT sql FROM sqlite_master WHERE type IN ("table",
"index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db
] TRACE 2025/07/07 17: 19: 56 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db
] TRACE 2025/07/07 17: 19: 56 SELECT * FROM `settings` 30 <nil>
[db
] TRACE 2025/07/07 17: 19: 56 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs"-1 <nil>
[db
] TRACE 2025/07/07 17: 19: 56 SELECT sql FROM sqlite_master WHERE type IN ("table",
"index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db
] TRACE 2025/07/07 17: 19: 56 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db
] TRACE 2025/07/07 17: 19: 56 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
[zaptec
] TRACE 2025/07/07 17: 19: 56 POST https: //api.zaptec.com/oauth/token
[zaptec
] TRACE 2025/07/07 17: 19: 56 grant_type=password&password=***&scope=openid+offline_access&username=***
--
{
    "access_token": "***",
    "token_type": "Bearer",
    "expires_in": 86399,
    "scope": "offline_access",
    "refresh_token": "***"
}
[zaptec
] TRACE 2025/07/07 17: 19: 56 GET https: //api.zaptec.com/api/chargers
[zaptec
] TRACE 2025/07/07 17: 19: 56 {
    "Pages": 1,
    "Data": [ ***
    ]
}
[zaptec
] TRACE 2025/07/07 17: 19: 56 GET https: //api.zaptec.com/api/chargers/032****3f7/state
[zaptec
] TRACE 2025/07/07 17: 19: 57 [
    {
        "ChargerId": "032****3f7",
        "StateId": -2,
        "Timestamp": "2025-07-01T19:27:38.637",
        "ValueAsString": "1"
    },
    {
        "ChargerId": "032****3f7",
        "StateId": -1,
        "Timestamp": "2025-07-07T15:19:56.247"
    },
    {
        "ChargerId": "032****3f7",
        "StateId": 1,
        "Timestamp": "2025-03-14T07:20:27.87",
[... 622 lines cut ...]
                "StateId": 721,
                "Timestamp": "2025-07-07T14:31:24.003",
                "ValueAsString": "***"
            },
            {
                "ChargerId": "032****3f7",
                "StateId": 722,
                "Timestamp": "2025-07-07T14:31:17.59"
            },
            {
                "ChargerId": "032****3f7",
                "StateId": 723,
                "Timestamp": "2025-07-07T14:31:21.057",
                "ValueAsString":"{\"SessionId\":\"***\",\"Energy\":19.072,\"StartDateTime\":\"2025-07-06T16:18:57.626976Z\",\"EndDateTime\":\"2025-07-07T14:31:17.574076Z\",\"ReliableClock\":true,\"StoppedByRFID\":false,\"AuthenticationCode\":null,\"SignedSession\":\"OCMF|{\\\"FV\\\":\\\"1.0\\\",\\\"GI\\\":\\\"Zaptec Pro MID\\\",\\\"GS\\\":\\\"ZPR283071\\\",\\\"GV\\\":\\\"5.4.1.0\\\",\\\"PG\\\":\\\"T1\\\",\\\"MF\\\":\\\"v1.4.6\\\",\\\"IS\\\":false,\\\"RD\\\":[{\\\"TM\\\":\\\"2025-07-06T16:18:57,606+00:00 R\\\",\\\"TX\\\":\\\"B\\\",\\\"RV\\\":679.583,\\\"RI\\\":\\\"1-0:1.8.0\\\",\\\"RU\\\":\\\"kWh\\\",\\\"ST\\\":\\\"G\\\"},{\\\"TM\\\":\\\"2025-07-06T16:30:00,875+00:00 R\\\",\\\"TX\\\":\\\"T\\\",\\\"RV\\\":680.661,\\\"RI\\\":\\\"1-0:1.8.0\\\",\\\"ST\\\":\\\"G\\\"},{\\\"TM\\\":\\\"2025-07-06T16:45:00,807+00:00 R\\\",\\\"TX\\\":\\\"T\\\",\\\"RV\\\":682.757,\\\"RI\\\":\\\"1-0:1.8.0\\\",\\\"ST\\\":\\\"G\\\"},{\\\"TM\\\":\\\"2025-07-06T17:00:00,928+00:00 R\\\",\\\"TX\\\":\\\"T\\\",\\\"RV\\\":684.86,\\\"RI\\\":\\\"1-0:1.8.0\\\",\\\"ST\\\":\\\"G\\\"},{\\\"TM\\\":\\\"2025-07-06T17:15:01,107+00:00 R\\\",\\\"TX\\\":\\\"T\\\",\\\"RV\\\":686.961,\\\"RI\\\":\\\"1-0:1.8.0\\\",\\\"ST\\\":\\\"G\\\"},{\\\"TM\\\":\\\"2025-07-06T17:30:01,137+00:00 R\\\",\\\"TX\\\":\\\"T\\\",\\\"RV\\\":687.687,\\\"RI\\\":\\\"1-0:1.8.0\\\",\\\"ST\\\":\\\"G\\\"},{\\\"TM\\\":\\\"2025-07-07T06:15:[zaptec
                ] TRACE 2025/07/07 17: 19: 57 POST https: //api.zaptec.com/api/chargers/032****3f7/update
                [zaptec
                ] TRACE 2025/07/07 17: 19: 57 {
                    "maxChargeCurrent": 0
                }
[zaptec
                ] TRACE 2025/07/07 17: 19: 57 POST https: //api.zaptec.com/api/chargers/032****3f7/update
                [zaptec
                ] TRACE 2025/07/07 17: 19: 57 {
                    "maxChargeCurrent": 1477
                }

tbaehler, 2025-07-07:
Oh, my fault....

the 1477 is in Watt and corresponds to the Current Power, that is of course not correct the maxChargeCurrent is of course in Ampere.. and therefore I read now the Currents. 

Can you try this build again. 

	a1, a2, a3, err := c.Currents()
	if err != nil {
		return err
	}

	err = c.MaxCurrent(0)
	if err != nil {
		return err
	}

	return c.MaxCurrent(int64(a1 + a2 + a3))

https://www.dropbox.com/scl/fi/x03vgsy77htfpoqqn04lf/evcc?rlkey=vf7sfmv4mu1nm63a7uyp5y2gd&st=azxpc2sq&dl=0

helijunky, 2025-07-07:
@tbaehler I tested it, unfortunately, the maxChargeCurrent is still not right. From what I see you are reading the actual currents. Instead you should get the desired maxChargeCurrent which is stored under stateID 510. This is a single number (integer), you don't have to build the sum since it is one value applicable to all phases.

The code should probably look something like this:
```
a, err := c.ObservationID(510) # I don't know how to do this but you need to get the desired current which is currently set as maxChargeCurrent
if err != nil {
	return err
}

err = c.MaxCurrent(0)
if err != nil {
	return err
}

return c.MaxCurrent(a)
```

Thanks!

helijunky, 2025-07-07:
wow, you are very fast :-) now the currents are set correctly as I would expect. However, it seems that the fast switch to 0 and back is too fast for the Zaptec portal to react.
Could you please add a short delay of 1 second between setting the currents to see if this would fix the issue?

```
res, err := c.statusG.Get()
if err != nil {
	return err
}

maxCur, err := res.ObservationByID(zaptec.ChargerMaxCurrent).Float64() <-- this is 510 
if err != nil {
	return err
}

err = c.MaxCurrent(0)
if err != nil {
	return err
}

time.Sleep(time.Second)  // allow the Zaptec portal to react before changing to the previous current

return c.MaxCurrent(int64(maxCur))

andig, 2025-07-07:
@tbaehler solche Zeitverhalten sind ziemlich unglücklich. Stattdessen sollten wir egtl. evcc die Möglichkeit geben, den Strom wieder richtig zu setzen. Die Abweichung kann es identifizieren:

```go
var _ api.CurrentGetter = (*Pulsatrix)(nil)

// GetMaxCurrent implements the api.CurrentGetter interface
func (c *Pulsatrix) GetMaxCurrent() (float64, error) {
	res, err := c.data.Get()
	return res.AmperageLimit, err
}
```

Leider erzeugt das dann einen Logikfehler. Für eine schöne Lösung habe ich gerade keine Idee.

lenusch, 2025-08-27:
Hi, danke dir. Hast du auch zufaellig ein BUILD / etwas einfaches und ich vermute dann habe ich die Updates von evcc direkt und @andig nicht mehr?

lenusch, 2025-08-27:
Cool danke. Sollte für Ubuntu gehen oder?

helijunky, 2025-08-27:
> Cool danke. Sollte für Ubuntu gehen oder?

dann brauchst du wahrscheinlich [dieses Binary für AMD64](https://github.com/helijunky/evcc/releases/download/878b3fb5/evcc.x86-64)