evcc-io/evcc #22160: Carwings broken - URL fix inside

docolli, 2025-07-01:
### Describe the bug

Nissan has changed the URL for carwings on July 2025.
The new URL is https://gdcportalgw.its-mo.com/api_v250205_NE/gdc/

I will test the new URL and if it works I will contact the upstream maintainer of carwings go library.

### Steps to reproduce

1. start evcc after 07/2025
2. Use carwings API to fetch data for Nissan EV
3.
...

### Configuration details

```yaml
<not necessary>
```

### Log details

```text
<not necessary>
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.5

andig, 2025-07-01:
Usually @joeshaw is quite quick to respond to PRs.

closed 2025-07-04 by andig

docolli, 2025-07-09:
Yes, this was exactly the error message I have got.

Fix is already there, will be released soon.

Zouz777, 2025-07-18:
I still get: "cannot create vehicle type 'carwings': login failed: invalid character '<'"
Is the fix in place?

PR #22207 (2025-07-04): carwings: upgrade to latest commit with new API URL
## Summary
- Upgrades carwings module to commit 1708e349f36c from joeshaw/carwings repository
- Includes update to new Carwings URL and API with new backend URL and encryption algorithm
- Fixes Nissan's July 2025 API changes

## Test plan
- [x] Build completes successfully
- [x] Go linting passes
- [x] Module dependencies resolve correctly

Fixes #22160

🤖 Generated with [Claude Code](https://claude.ai/code)