evcc-io/evcc #22176: UI Eingabe Tarife - PV Solar Vorhersage wird nicht mehr angezeigt

depeche8888, 2025-07-02:
### Describe the bug

ich hatte unter tariffs einen festen Preis definiert und Solar Vorhersage mit Solcast. Alles in evcc.yaml definiert und funktioniert.
Jetzt habe ich auf Dynamischen strom Tarif ostrom umgestellt. jetzt wird nur noch unter Vorhersage der Preis angezeigt.Kein Solar Vorhersage mehr. Dann habe ich die Tarife über die UI konfiguriert. Selbe Ergebnis. Keine Solar Vorhersage mehr. Option ist ausgegraut. Der Preis Voehersage wird angezeigt. kann ich zurück auf evcc.yaml ohne Tarife über die UI ? 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
currency: EUR

grid: # price using energy from the grid
  #type: fixed
  #price: 0.33 # EUR/kWh
  type: template
  template: ostrom
  clientid: XXX # Client ID
  clientsecret: XXX # Client Secret
  

feedin: # price for feeding solar energy to the grid
  type: fixed
  price: 0.08 # EUR/kWh

#co2: # carbon intensity forecast
#  type: template
#  template: grünstromindex
#  zip: XXXXX

solar: # list of pv generation forecast (additive)
- type: template
  template: solcast
  site: XXX # Ressource ID deiner Anlage
  token: XXX # Solcast API Token
  from: 6 # Startzeit der Datenabrufe, Angabe in vollen Stunden, z.B "6" (optional)
  to: 20 # Endzeit der Datenabrufe, Angabe in vollen Stunden, z.B "20" (optional)
  interval: 3h # optional
- type: template
  template: solcast
  site: XXX # Ressource ID deiner Anlage
  token: XXX # Solcast API Token
  from: 6 # Startzeit der Datenabrufe, Angabe in vollen Stunden, z.B "6" (optional)
  to: 20 # Endzeit der Datenabrufe, Angabe in vollen Stunden, z.B "20" (optional)
  interval: 3h # optional
```

### Log details

```text
[site ] DEBUG 2025/07/02 20:40:00 ----
[lp-1 ] DEBUG 2025/07/02 20:40:00 charge power: 0W
[lp-1 ] DEBUG 2025/07/02 20:40:00 charge currents: [0 0 0]A
[site ] DEBUG 2025/07/02 20:40:00 grid power: -411W
[site ] DEBUG 2025/07/02 20:40:00 grid powers: [-168 -88 -156]W
[site ] DEBUG 2025/07/02 20:40:00 grid currents: [-0.786 -1.22 -1.99]A
[site ] DEBUG 2025/07/02 20:40:00 pv 2 power: 65W
[site ] DEBUG 2025/07/02 20:40:00 battery 1 power: 0W
[site ] DEBUG 2025/07/02 20:40:00 pv 1 power: 729W
[site ] DEBUG 2025/07/02 20:40:00 battery 1 soc: 89%
[site ] DEBUG 2025/07/02 20:40:01 pv power: 794W
[site ] DEBUG 2025/07/02 20:40:01 site power: -261W
[lp-1 ] DEBUG 2025/07/02 20:40:01 charge voltages: [238 236 237]V
[lp-1 ] DEBUG 2025/07/02 20:40:01 charge total import: 204.110kWh
[lp-1 ] DEBUG 2025/07/02 20:40:01 charger status: B
[... 345 lines cut ...]
[site ] DEBUG 2025/07/02 20:45:30 pv 1 power: 546W
[site ] DEBUG 2025/07/02 20:45:31 battery 1 soc: 89%
[site ] DEBUG 2025/07/02 20:45:31 pv power: 596W
[site ] DEBUG 2025/07/02 20:45:31 site power: -91W
[lp-1 ] DEBUG 2025/07/02 20:45:31 charge voltages: [238 236 237]V
[lp-1 ] DEBUG 2025/07/02 20:45:31 charge total import: 204.110kWh
[lp-1 ] DEBUG 2025/07/02 20:45:31 charger status: B
[lp-1 ] DEBUG 2025/07/02 20:45:31 limitSoc reached: 90.0% > 90%
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.5

naltatis, 2025-07-04:
@depeche8888 Mach mal im Config UI das Eingabefeld ganz leer und starte neu. Dann werden wieder die Tarife aus der evcc.yaml gezogen. Falls dein Problem immer noch besteht poste hier mal bitte die Ausgabe von `evcc tariff`.

humi208, 2025-07-05:
Ich habe auch ein Problem mit solcast. Startet man evcc neu zu einem Zeitpunkt, wenn solcast nicht abgefragt werden darf, dann wird auch später oder am nächsten Tag kein Vorhersage angezeigt. Man muss evcc während des Zeitabschnitt neustarten, wenn es solcast auch abgefragt werden darf. Dann funktioniert es. Evtl. ist das auch hier das Problem.

depeche8888, 2025-07-06:
> [@depeche8888](https://github.com/depeche8888) Mach mal im Config UI das Eingabefeld ganz leer und starte neu. Dann werden wieder die Tarife aus der evcc.yaml gezogen. Falls dein Problem immer noch besteht poste hier mal bitte die Ausgabe von `evcc tariff`.

ich habe die Traiffs über Config Ui komplett geleert. Alle Einstellungen werden jetzt über die evcc.yaml gezogen.
was meinst du mit Ausgabe von evcc tariff ?

depeche8888, 2025-07-06:
> Ich habe auch ein Problem mit solcast. Startet man evcc neu zu einem Zeitpunkt, wenn solcast nicht abgefragt werden darf, dann wird auch später oder am nächsten Tag keine Vorhersage angezeigt oder abgefragt. Man muss evcc während des Zeitabschnitt neustarten, wenn solcast auch abgefragt werden darf. Dann funktioniert es. Evtl. ist das auch hier das Problem.

ok ich habe mal die Einschränkungen wann abegfragt werden darf 
      from: 6 # Startzeit der Datenabrufe, Angabe in vollen Stunden, z.B "6" (optional)
      to: 20 # Endzeit der Datenabrufe, Angabe in vollen Stunden, z.B "20" (optional)
      interval: 3h # optional

auskommentiert # 
evcc neu gestartet. immer noch keine PV Vorhersage.
Werde es morgen mal in dem Zeitfenster 6-20 Uhr nochmal testen (vorherige Konfiguration).

andig, 2025-07-07:
Was sagt

    evcc config

andig, 2025-07-07:
> es wird nach wie vor in der Vorhersage nur der Preis angezeigt. (kein PV Vorhersage)

andig, 2025-07-19:
Das sollte helfen: https://github.com/evcc-io/evcc/pull/22446

hybe2022, 2025-08-03:
Leider hilft dies nicht. Die Prognose ist nur sichtbar, wenn evcc gestartet wird in dem Fenster, wenn es auch die Daten abrufen darf und wenn noch freie Zugriffe verfügbar sind. 
Tritt beim Starten von evcc ein Fehler auf, weil der Start außerhalb des Zeitfenster stattfindet, dann wird die Solar Prognose nie abgerufen, bis evcc wieder im definierten Zeitfenster neugestartet wird.

humi208, 2025-08-03:
Zum Glück kommen hier immer diese tollen Rückmeldungen. Das macht Freude dem Projekt Feedback zu geben und es zu sponsoren. Man hat ja auch immer direkt das Log verfügbar, wenn man ein Problem bemerkt... Aber zum Glück werden auch Tickets geschlossen ohne diese jemals getestet zu haben.

andig, 2025-08-07:
> Ich gehe davon aus, dass evcc bei jedem Start von solcast liest und somit die 10 maximalen Abrufe pro Tag aufbraucht, statt nur im vorgegebenen Intervall zwischen den angegeben Zeiten zu lesen (Start 03:00, Ende 20:00, Intervall 5h).

Dafür gibts das bereits verlinkte https://github.com/evcc-io/evcc/pull/22446, sollte also nicht so sein.

hybe2022, 2025-08-07:
PR https://github.com/evcc-io/evcc/pull/22446 wurde schon in 0.206.0 geliefert. Das Problem schient also noch nicht gelöst zu sein. 

Ein weiteres Problem scheint zu sein, dass die Startzeit und das Intervall nicht eingehalten werden. Siehe:

> Vor allem frage ich mich aber auch, warum um 10:00 und 15:00 abgefragt wurde. Bei der Einstellung Start 03:00, Ende 20:00, und Intervall 5h hätte ich ein Abfrage um 3:00, 8:00, 13:00, 18:00... erwartet. Es sieht so aus als wird 5:00, 10:00, 15:00...

andig, 2025-08-07:
> Das Problem schient also noch nicht gelöst zu sein

Ah- Du hast die aktuelle Version? Das ist im Log leider nicht zu sehen. Vollständige Informationen helfen immer bei der Problemanalyse! Mit dem Log lässt sich nichts anfangen weil es keine Tarifaufrufe zeigt. Helfen würde mal

```
evcc cache get
evcc tariff --log trace,db:error <problematischer tarif>
```

hybe2022, 2025-08-07:
Die Version ist im Log sichtbar:

Aug 04 09:17:09 raspberrypi evcc[13297]: [main  ] INFO 2025/08/04 09:17:09 evcc 0.207.1

Auch sind vollumfänglich alle tariff aufrufe enthalten. Verheimlicht wird nichts:

	Line  21599: Aug 04 03:42:21 raspberrypi evcc[3759]: [site  ] INFO 2025/08/04 03:42:21   tariffs:
	Line  25220: Aug 04 04:09:53 raspberrypi evcc[3861]: [tariff] DEBUG 2025/08/04 04:09:53 using cache: electricitymaps-free- (start: 2025-08-04 04:00:00 +0200 CEST, end: 2025-08-06 04:00:00 +0200 CEST)
	Line  25221: Aug 04 04:09:53 raspberrypi evcc[3861]: [tariff] DEBUG 2025/08/04 04:09:53 using cache: solcast- (start: 2025-08-04 04:00:00 +0200 CEST, end: 2025-08-06 05:00:00 +0200 CEST)
	Line  25222: Aug 04 04:09:53 raspberrypi evcc[3861]: [tariff] DEBUG 2025/08/04 04:09:53 using cache: solcast- (start: 2025-08-04 04:00:00 +0200 CEST, end: 2025-08-06 05:00:00 +0200 CEST)
	Line  25229: Aug 04 04:09:53 raspberrypi evcc[3861]: [site  ] INFO 2025/08/04 04:09:53   tariffs:
	Line  25283: Aug 04 04:10:23 raspberrypi evcc[3885]: [tariff] DEBUG 2025/08/04 04:10:23 using cache: electricitymaps-free- (start: 2025-08-04 04:00:00 +0200 CEST, end: 2025-08-06 04:00:00 +0200 CEST)
	Line  25284: Aug 04 04:10:23 raspberrypi evcc[3885]: [tariff] DEBUG 2025/08/04 04:10:23 using cache: solcast- (start: 2025-08-04 04:00:00 +0200 CEST, end: 2025-08-06 05:00:00 +0200 CEST)
	Line  25285: Aug 04 04:10:23 raspberrypi evcc[3885]: [tariff] DEBUG 2025/08/04 04:10:23 using cache: solcast- (start: 2025-08-04 04:00:00 +0200 CEST, end: 2025-08-06 05:00:00 +0200 CEST)
	Line  25292: Aug 04 04:10:23 raspberrypi evcc[3885]: [site  ] INFO 2025/08/04 04:10:23   tariffs:
	Line  25353: Aug 04 04:10:55 raspberrypi evcc[3907]: [tariff] DEBUG 2025/08/04 04:10:55 using cache: electricitymaps-free- (start: 2025-08-04 04:00:00 +0200 CEST, end: 2025-08-06 04:00:00 +0200 CEST)
	Line  25354: Aug 04 04:10:55 raspberrypi evcc[3907]: [tariff] DEBUG 2025/08/04 04:10:55 using cache: solcast- (start: 2025-08-04 04:00:00 +0200 CEST, end: 2025-08-06 05:00:00 +0200 CEST)
	Line  25355: Aug 04 04:10:55 raspberrypi evcc[3907]: [tariff] DEBUG 2025/08/04 04:10:55 using cache: solcast- (start: 2025-08-04 04:00:00 +0200 CEST, end: 2025-08-06 05:00:00 +0200 CEST)
	Line  25362: Aug 04 04:10:55 raspberrypi evcc[3907]: [site  ] INFO 2025/08/04 04:10:55   tariffs:
	Line  25416: Aug 04 04:11:23 raspberrypi evcc[3921]: [tariff] DEBUG 2025/08/04 04:11:23 using cache: solcast- (start: 2025-08-04 04:00:00 +0200 CEST, end: 2025-08-06 05:00:00 +0200 CEST)
	Line  25417: Aug 04 04:11:23 raspberrypi evcc[3921]: [tariff] DEBUG 2025/08/04 04:11:23 using cache: solcast- (start: 2025-08-04 04:00:00 +0200 CEST, end: 2025-08-06 05:00:00 +0200 CEST)
	Line  25418: Aug 04 04:11:23 raspberrypi evcc[3921]: [tariff] DEBUG 2025/08/04 04:11:23 using cache: electricitymaps-free- (start: 2025-08-04 04:00:00 +0200 CEST, end: 2025-08-06 04:00:00 +0200 CEST)
	Line  25425: Aug 04 04:11:23 raspberrypi evcc[3921]: [site  ] INFO 2025/08/04 04:11:23   tariffs:
	Line  25479: Aug 04 04:11:52 raspberrypi evcc[3935]: [tariff] DEBUG 2025/08/04 04:11:52 using cache: solcast- (start: 2025-08-04 04:00:00 +0200 CEST, end: 2025-08-06 05:00:00 +0200 CEST)
	Line  25480: Aug 04 04:11:52 raspberrypi evcc[3935]: [tariff] DEBUG 2025/08/04 04:11:52 using cache: electricitymaps-free- (start: 2025-08-04 04:00:00 +0200 CEST, end: 2025-08-06 04:00:00 +0200 CEST)
	Line  25481: Aug 04 04:11:52 raspberrypi evcc[3935]: [tariff] DEBUG 2025/08/04 04:11:52 using cache: solcast- (start: 2025-08-04 04:00:00 +0200 CEST, end: 2025-08-06 05:00:00 +0200 CEST)
	Line  25488: Aug 04 04:11:52 raspberrypi evcc[3935]: [site  ] INFO 2025/08/04 04:11:52   tariffs:
	Line  25553: Aug 04 04:12:26 raspberrypi evcc[3953]: [tariff] DEBUG 2025/08/04 04:12:26 using cache: electricitymaps-free- (start: 2025-08-04 04:00:00 +0200 CEST, end: 2025-08-06 04:00:00 +0200 CEST)
	Line  25554: Aug 04 04:12:26 raspberrypi evcc[3953]: [tariff] DEBUG 2025/08/04 04:12:26 using cache: solcast- (start: 2025-08-04 04:00:00 +0200 CEST, end: 2025-08-06 05:00:00 +0200 CEST)
	Line  25555: Aug 04 04:12:26 raspberrypi evcc[3953]: [tariff] DEBUG 2025/08/04 04:12:26 using cache: solcast- (start: 2025-08-04 04:00:00 +0200 CEST, end: 2025-08-06 05:00:00 +0200 CEST)
	Line  25562: Aug 04 04:12:26 raspberrypi evcc[3953]: [site  ] INFO 2025/08/04 04:12:26   tariffs:
	Line  61195: Aug 04 08:42:14 raspberrypi evcc[10703]: [tariff] DEBUG 2025/08/04 08:42:14 using cache: electricitymaps-free- (start: 2025-08-04 07:00:00 +0200 CEST, end: 2025-08-06 04:00:00 +0200 CEST)
	Line  61196: Aug 04 08:42:14 raspberrypi evcc[10703]: [tariff] DEBUG 2025/08/04 08:42:14 using cache: solcast- (start: 2025-08-04 07:00:00 +0200 CEST, end: 2025-08-06 05:00:00 +0200 CEST)
	Line  61197: Aug 04 08:42:14 raspberrypi evcc[10703]: [tariff] DEBUG 2025/08/04 08:42:14 using cache: solcast- (start: 2025-08-04 07:00:00 +0200 CEST, end: 2025-08-06 05:00:00 +0200 CEST)
	Line  61204: Aug 04 08:42:14 raspberrypi evcc[10703]: [site  ] INFO 2025/08/04 08:42:14   tariffs:
	Line  65824: Aug 04 09:17:11 raspberrypi evcc[13297]: [tariff] DEBUG 2025/08/04 09:17:11 using cache: electricitymaps-free- (start: 2025-08-04 08:00:00 +0200 CEST, end: 2025-08-06 04:00:00 +0200 CEST)
	Line  65825: Aug 04 09:17:11 raspberrypi evcc[13297]: [tariff] DEBUG 2025/08/04 09:17:11 using cache: solcast- (start: 2025-08-04 08:00:00 +0200 CEST, end: 2025-08-06 05:00:00 +0200 CEST)
	Line  65826: Aug 04 09:17:11 raspberrypi evcc[13297]: [tariff] DEBUG 2025/08/04 09:17:11 using cache: solcast- (start: 2025-08-04 08:00:00 +0200 CEST, end: 2025-08-06 05:00:00 +0200 CEST)
	Line  65833: Aug 04 09:17:12 raspberrypi evcc[13297]: [site  ] INFO 2025/08/04 09:17:12   tariffs:

hybe2022, 2025-08-07:
Wie gesagt nach einem Fehler kommen keine Aufrufe mehr. Das war ja initial bereits meine Aussage nach den logs zu urteilen... Wird das Limit erreicht und führt zum Fehler kommen bis zu einem Neustart keine Abfragen mehr...