evcc-io/evcc #22190: Use last selected vehicle

gcastelain, 2025-07-03:
**Is your feature request related to a problem? Please describe.**
I have 2 vehicles with one loadpoint. None of them has auto detection capabilities. So I'm forced to pick one car as the "default" car. Both cars are almost used equally. So 1 out of 2, the wrong car get selected automatically.

**Describe the solution you'd like**
Can you add an option in the "default vehicle" dropdown, which says: "keep the selected vehicle"? FYI: this option already exists for the "mode" setting (the 3 dashes in the dropdown)

**Describe alternatives you've considered**
None

**Additional context**
The real problem is my SMA EV Charger, which starts charging as soon as the vehicle is connected, giving the error message "charger out of sync: expected disabled, got enabled". But stops after a few seconds because EVCC tells it to. (e.g. charging is set to off, or PV and there is not enough sun)

What I'm worry about is that every time I connect my car, it gets 3 seconds juice and then stops. Which may not be good for the battery/car I guess?

So the only (easy) option I see is: before connection my car, I selected the correct car and also the mode PV or Fast. So when plugging in, the car starts charging and it keeps charging because EVCC will tell it to. 

With the current EVCC version, when connecting the charger, the default car is always selected which makes it the wrong car in 1 out of 2...

gcastelain, 2025-07-03:
I use EVCC v0.204.5 in a Docker

gcastelain, 2025-07-04:
Yes, I confirm it works.

Thanks for the help and the fast reply!

I guess there is nothing to do about the fact that my SMA charger starts directly with charging when plugged in?

andig, 2025-07-04:
No :( except asking SMA for better controllability…